linuxwacom/input-wacom #107: Tablet logs me out when changing screens

cohen990, 2019-08-15:
Apologies if this is the wrong place.

I'm using a Wacom Intuos Pro M (PTH-651/K) with Ubuntu 18.04.3 LTS

I have one of my buttons mapped to "Switch Monitor" and if I press it too many times (twice or more) it logs me out of Ubuntu and I have to log back in.

I'm not sure where to find the relevant logs and configuration files but happy to post them if you can show me where to look

cohen990, 2019-08-15:
I found the following system logs which happen at the time when it logs me out 

```Aug 15 12:10:39 danbuntu guake[8054]: guake: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Power.desktop' exited with code 1
Aug 15 12:10:39 danbuntu at-spi-bus-launcher[7327]: XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
Aug 15 12:10:39 danbuntu at-spi-bus-launcher[7327]:       after 82 requests (82 known processed) with 0 events remaining.
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Power.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Wacom.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Clipboard.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gsd-color[7990]: gsd-color: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Wacom.desktop' exited with code 1
Aug 15 12:10:39 danbuntu nautilus-deskto[8062]: nautilus-desktop: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Clipboard.desktop' exited with code 1
Aug 15 12:10:39 danbuntu [7976]: gsd-clipboard: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu gsd-media-keys[8001]: gsd-media-keys: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu gsd-keyboard[7992]: gsd-keyboard: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu gnome-shell[7883]: gnome-shell: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu gsd-xsettings[7962]: gsd-xsettings: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu [7950]: gsd-power: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu [7971]: gsd-wacom: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Color.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Color.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Keyboard.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.Keyboard.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.MediaKeys.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.MediaKeys.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.XSettings.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.SettingsDaemon.XSettings.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gsd-color[2827]: failed to connect to device: Failed to connect to missing device /org/freedesktop/ColorManager/devices/xrandr_AOC_2460G4_62469_dan_1000
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.Shell.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.Shell.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-shell[8174]: Failed to create backend: Unable to open display ':0'
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.Shell.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: WARNING: App 'org.gnome.Shell.desktop' respawning too quickly
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.Shell.desktop' exited with code 1
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: WARNING: App 'org.gnome.Shell.desktop' respawning too quickly
Aug 15 12:10:39 danbuntu gnome-session[7789]: gnome-session-binary[7789]: CRITICAL: We failed, but the fail whale is dead. Sorry....
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: Unrecoverable failure in required component org.gnome.Shell.desktop
Aug 15 12:10:39 danbuntu gnome-session-binary[7789]: CRITICAL: We failed, but the fail whale is dead. Sorry....
Aug 15 12:10:39 danbuntu kernel: [10348.591789] rfkill: input handler enabled
Aug 15 12:10:39 danbuntu kernel: [10348.634300] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=2) Notifying crtc state: 3
Aug 15 12:10:40 danbuntu systemd[1]: Stopping User Manager for UID 1000...
Aug 15 12:10:40 danbuntu systemd[7180]: Stopping Virtual filesystem service - Media Transfer Protocol monitor...
Aug 15 12:10:40 danbuntu systemd[7180]: Stopping D-Bus User Message Bus...
Aug 15 12:10:40 danbuntu systemd[7180]: Stopping Virtual filesystem service - Apple File Conduit monitor...
[... cut ...]
Aug 15 12:10:40 danbuntu systemd[7180]: Reached target Shutdown.
Aug 15 12:10:40 danbuntu systemd[7180]: Starting Exit the Session...
Aug 15 12:10:40 danbuntu systemd[7180]: Stopped target Timers.
Aug 15 12:10:40 danbuntu systemd[7180]: Received SIGRTMIN+24 from PID 8183 (kill).
Aug 15 12:10:40 danbuntu systemd[1]: Stopped User Manager for UID 1000.
Aug 15 12:10:40 danbuntu systemd[1]: Removed slice User Slice of dan.
Aug 15 12:10:40 danbuntu gsd-color[2827]: could not find device: property match 'XRANDR_name'='DP-1' does not exist```

cohen990, 2019-08-15:
It only happens when I switch to one of my monitors (I have a dual monitor setup) and only when the cursor comes into proximity with the tablet while that screen has just been switched to

skomra, 2019-08-15:
Hi, I was not immediately able to reproduce this with the Intuos Pro on 18.04.03 with two monitors. From your log it looks like there may be a problem with gsd-color. Unfortunately I don't have a lot of experience with gsd-color but I think you can turn it off by:

$sudo apt install dconf-editor
$dconf-editor

then in the GUI:
org > gnome > settings-daemon > plugins > color (switch to off)

Then reboot and see if you can still reproduce the issue.

jigpu, 2019-08-16:
You may also want to run `coredumpctl` to see if there are any crashes logged within a few seconds of the journalctl output. If there are, you try running `coredumpctl info <path>` with the path to the process that crashed and post the output here.

cohen990, 2019-08-19:
The crashes are now happening less predictably.

```
Aug 19 12:03:42 danbuntu xournalpp[11918]: xournalpp: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: #
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: # A fatal error has been detected by the Java Runtime Environment:
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: #
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: #  SIGSEGV (0xb) at pc=0x00007f3ef1f47ff6, pid=10454, tid=0x00007f3dcd346700
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: #
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: # JRE version: OpenJDK Runtime Environment (8.0_212-b04) (build 1.8.0_212-release-1586-b4)
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: # Java VM: OpenJDK 64-Bit Server VM (25.212-b4 mixed mode linux-amd64 compressed oops)
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: # Problematic frame:
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: # C  [libc.so.6+0x42ff6]
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: #
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: # Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: #
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: # An error report file with more information is saved as:
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: # /home/dan/java_error_in_RIDER_10454.log
Aug 19 12:03:42 danbuntu jetbrains-rider.desktop[10385]: pure virtual method called
Aug 19 12:03:42 danbuntu org.gnome.Shell.desktop[5192]: [6261:6261:0819/120342.777253:ERROR:chrome_browser_main_extra_parts_x11.cc(62)] X IO error received (X server probably went away)
Aug 19 12:03:42 danbuntu guake[5401]: guake: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu update-notifier[5977]: update-notifier: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu nautilus-deskto[5406]: nautilus-desktop: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu gsd-keyboard[5306]: gsd-keyboard: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu gsd-color[5304]: gsd-color: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu gsd-xsettings[5276]: gsd-xsettings: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu gsd-wacom[5282]: gsd-wacom: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu at-spi-bus-launcher[953]: XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
Aug 19 12:03:42 danbuntu at-spi-bus-launcher[953]:       after 9149 requests (9149 known processed) with 0 events remaining.
Aug 19 12:03:42 danbuntu gsd-clipboard[5289]: gsd-clipboard: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu gsd-power[5262]: gsd-power: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu gsd-media-keys[5322]: gsd-media-keys: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu gnome-shell[5192]: gnome-shell: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 19 12:03:42 danbuntu gnome-session[5097]: gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Wacom.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Wacom.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session[5097]: gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Power.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session[5097]: gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Color.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session[5097]: gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Keyboard.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session[5097]: gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.MediaKeys.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Power.desktop' exited with code 1
Aug 19 12:03:42 danbuntu org.gnome.Shell.desktop[5192]: [6314:6314:0819/120342.783673:ERROR:x11_util.cc(109)] X IO error received (X server probably went away)
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Color.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Keyboard.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.MediaKeys.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Clipboard.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session[5097]: gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.Clipboard.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session[5097]: gnome-session-binary[5097]: WARNING: App 'org.gnome.SettingsDaemon.XSettings.desktop' exited with code 1
[... cut ...]
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: WARNING: App 'org.gnome.Shell.desktop' exited with code 1
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: WARNING: App 'org.gnome.Shell.desktop' respawning too quickly
Aug 19 12:03:42 danbuntu gnome-session-binary[5097]: CRITICAL: We failed, but the fail whale is dead. Sorry....
Aug 19 12:03:42 danbuntu kernel: [10833.182369] rfkill: input handler enabled
Aug 19 12:03:43 danbuntu kernel: [10833.465339] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=2) Notifying crtc state: 3
```

cohen990, 2019-08-19:
I did turn off the colors as instructed but the problem still occurs

cohen990, 2019-08-19:
here is the result of coredump

```coredumpctl info /usr/lib/xorg/Xorg                                                                                                                                                               dan@danbuntu
           PID: 13189 (Xorg)
           UID: 1000 (dan)
           GID: 1000 (dan)
        Signal: 11 (SEGV)
     Timestamp: Mon 2019-08-19 12:15:30 BST (1min 9s ago)
  Command Line: /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
    Executable: /usr/lib/xorg/Xorg
 Control Group: /user.slice/user-1000.slice/session-9.scope
          Unit: session-9.scope
         Slice: user-1000.slice
       Session: 9
     Owner UID: 1000 (dan)
       Boot ID: e04f2cac24bf4c42b9ac54f12f0607cd
    Machine ID: a8c84753764447e7b1bf3bd00476dcd8
      Hostname: danbuntu
       Storage: /var/lib/systemd/coredump/core.Xorg.1000.e04f2cac24bf4c42b9ac54f12f0607cd.13189.1566213330000000.lz4
       Message: Process 13189 (Xorg) of user 1000 dumped core.
                
                Stack trace of thread 13204:
                #0  0x00007f4ed6af0ea6 n/a (modesetting_drv.so)
                #1  0x00005571298e50f3 n/a (Xorg)
                #2  0x00005571298effa2 n/a (Xorg)
                #3  0x00005571299bc53b n/a (Xorg)
                #4  0x00005571299bd309 miPointerSetPosition (Xorg)
                #5  0x00005571298815de n/a (Xorg)
                #6  0x0000557129881d80 n/a (Xorg)
                #7  0x00005571298835bf GetPointerEvents (Xorg)
                #8  0x0000557129883b60 QueuePointerEvents (Xorg)
                #9  0x00005571298bbbd9 xf86PostMotionEventP (Xorg)
                #10 0x00007f4ec6a63ecc n/a (wacom_drv.so)
                #11 0x00007f4ec6a65233 n/a (wacom_drv.so)
                #12 0x00007f4ec6a6a89d n/a (wacom_drv.so)
                #13 0x00007f4ec6a62677 n/a (wacom_drv.so)
                #14 0x00007f4ec6a628d3 n/a (wacom_drv.so)
                #15 0x00005571299d2c33 n/a (Xorg)
                #16 0x00005571299d5331 n/a (Xorg)
                #17 0x00005571299d2a7e n/a (Xorg)
                #18 0x00007f4ed9e526db start_thread (libpthread.so.0)
                #19 0x00007f4ed9b7b88f __clone (libc.so.6)
                
                Stack trace of thread 13189:
                #0  0x00007f4ed9b7bbb7 epoll_wait (libc.so.6)
                #1  0x00005571299d52e7 n/a (Xorg)
                #2  0x00005571299ce32b WaitForSomething (Xorg)
                #3  0x000055712986ac43 n/a (Xorg)
                #4  0x000055712986eee0 n/a (Xorg)
                #5  0x00007f4ed9a7bb97 __libc_start_main (libc.so.6)
                #6  0x0000557129858b8a _start (Xorg)
                
                Stack trace of thread 13190:
                #0  0x00007f4ed9e589f3 futex_wait_cancelable (libpthread.so.0)
                #1  0x00007f4ed45be31a n/a (i965_dri.so)
                #2  0x00007f4ed45be047 n/a (i965_dri.so)
                #3  0x00007f4ed9e526db start_thread (libpthread.so.0)
                #4  0x00007f4ed9b7b88f __clone (libc.so.6)
                
                Stack trace of thread 13199:
                #0  0x00007f4ed9e589f3 futex_wait_cancelable (libpthread.so.0)
                #1  0x00007f4ed0fa80aa n/a (nouveau_dri.so)
                #2  0x00007f4ed0fa7dd7 n/a (nouveau_dri.so)
                #3  0x00007f4ed9e526db start_thread (libpthread.so.0)
                #4  0x00007f4ed9b7b88f __clone (libc.so.6)
```

cohen990, 2019-08-19:
Looks like something is tripping up xorg

skomra, 2019-08-19:
Thanks for the core dump, it looks like the modesetting driver may be to blame, but let's try and narrow it down. Please use apt to install the xorg debug symbols appropriate for your graphics driver as well as the evdev debug symbols (see below). Then catch another core dump after the problem occurs.
```

xserver-xorg-video-amdgpu-dbg - X.Org X server -- AMDGPU display driver (debugging symbols)
xserver-xorg-video-ati-dbg - X.Org X server -- AMD/ATI display driver wrapper (debugging symbols)
xserver-xorg-video-intel-dbg - X.Org X server -- Intel i8xx, i9xx display driver (debug symbols)
xserver-xorg-video-radeon-dbg - X.Org X server -- AMD/ATI Radeon display driver (debugging symbols)

xserver-xorg-input-evdev-dbg - X.Org X server -- evdev input driver (debug symbols)
```

cohen990, 2019-08-20:
I tried installing xserver-xorg-video-intel-dbg - which I think I the correct one for my driver but the core dump looks the same

This is the command I used to check which driver I'm using
```
~ » egrep -i " connected|card detect|primary dev|Setting driver" /var/log/Xorg.0.log
[    62.478] (II) intel(G0): Using Kernel Mode Setting driver: i915, version 1.6.0 20151010
[    86.406] (II) modeset(0): Output DP-1 connected
```

jigpu, 2019-08-20:
Weird. I wonder if asking gdb directly would work any better. Install gdb and then run `coredumpctl debug <path>`. At the "(gdb)" prompt that follows, type in `bt full` and press enter. Afterwards press CTRL+D to quit. Hopefully that'll do the trick and print out the additional details that we need to understand what's going on.

cohen990, 2019-08-22:
@jigpu unfortunately looks like we are missing some debug symbols

```
#0  0x00007f6b80d9bea6 in ?? () from /usr/lib/xorg/modules/drivers/modesetting_drv.so
No symbol table info available.
#1  0x000055bb62c8b0f3 in ?? ()
No symbol table info available.
#2  0x000055bb62c95fa2 in ?? ()
No symbol table info available.
#3  0x000055bb62d6253b in ?? ()
No symbol table info available.
#4  0x000055bb62d63309 in miPointerSetPosition ()
No symbol table info available.
#5  0x000055bb62c275de in ?? ()
No symbol table info available.
#6  0x000055bb62c27d80 in ?? ()
No symbol table info available.
#7  0x000055bb62c295bf in GetPointerEvents ()
No symbol table info available.
#8  0x000055bb62c29b60 in QueuePointerEvents ()
No symbol table info available.
#9  0x00007f6b71ce4fb5 in ?? () from /usr/lib/xorg/modules/input/libinput_drv.so
No symbol table info available.
#10 0x00007f6b71ce5b40 in ?? () from /usr/lib/xorg/modules/input/libinput_drv.so
No symbol table info available.
#11 0x000055bb62d78c33 in ?? ()
No symbol table info available.
#12 0x000055bb62d7b331 in ?? ()
No symbol table info available.
#13 0x000055bb62d78a7e in ?? ()
No symbol table info available.
#14 0x00007f6b840fd6db in start_thread (arg=0x7f6b6affd700) at pthread_create.c:463
        pd = 0x7f6b6affd700
        now = <optimised out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140099333379840, 2214544212643191700, 140099333377216, 0, 0, 140734716615744, -2275037141702632556, -2275254433581720684}, mask_was_saved = 0}}, priv = {
            pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimised out>
#15 0x00007f6b83e2688f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
No locals.
(gdb) bt full
#0  0x00007f6b80d9bea6 in ?? () from /usr/lib/xorg/modules/drivers/modesetting_drv.so
No symbol table info available.
#1  0x000055bb62c8b0f3 in ?? ()
No symbol table info available.
#2  0x000055bb62c95fa2 in ?? ()
No symbol table info available.
#3  0x000055bb62d6253b in ?? ()
No symbol table info available.
#4  0x000055bb62d63309 in miPointerSetPosition ()
No symbol table info available.
#5  0x000055bb62c275de in ?? ()
No symbol table info available.
#6  0x000055bb62c27d80 in ?? ()
No symbol table info available.
#7  0x000055bb62c295bf in GetPointerEvents ()
No symbol table info available.
#8  0x000055bb62c29b60 in QueuePointerEvents ()
No symbol table info available.
#9  0x00007f6b71ce4fb5 in ?? () from /usr/lib/xorg/modules/input/libinput_drv.so
No symbol table info available.
#10 0x00007f6b71ce5b40 in ?? () from /usr/lib/xorg/modules/input/libinput_drv.so
No symbol table info available.
#11 0x000055bb62d78c33 in ?? ()
No symbol table info available.
#12 0x000055bb62d7b331 in ?? ()
No symbol table info available.
#13 0x000055bb62d78a7e in ?? ()
No symbol table info available.
#14 0x00007f6b840fd6db in start_thread (arg=0x7f6b6affd700) at pthread_create.c:463
        pd = 0x7f6b6affd700
        now = <optimised out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140099333379840, 2214544212643191700, 140099333377216, 0, 0, 140734716615744, -2275037141702632556, -2275254433581720684}, mask_was_saved = 0}}, priv = {
            pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimised out>
#15 0x00007f6b83e2688f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
No locals.
(gdb) bt full
#0  0x00007f6b80d9bea6 in ?? () from /usr/lib/xorg/modules/drivers/modesetting_drv.so
No symbol table info available.
#1  0x000055bb62c8b0f3 in ?? ()
No symbol table info available.
#2  0x000055bb62c95fa2 in ?? ()
No symbol table info available.
#3  0x000055bb62d6253b in ?? ()
No symbol table info available.
#4  0x000055bb62d63309 in miPointerSetPosition ()
No symbol table info available.
#5  0x000055bb62c275de in ?? ()
No symbol table info available.
#6  0x000055bb62c27d80 in ?? ()
No symbol table info available.
#7  0x000055bb62c295bf in GetPointerEvents ()
No symbol table info available.
#8  0x000055bb62c29b60 in QueuePointerEvents ()
No symbol table info available.
#9  0x00007f6b71ce4fb5 in ?? () from /usr/lib/xorg/modules/input/libinput_drv.so
No symbol table info available.
#10 0x00007f6b71ce5b40 in ?? () from /usr/lib/xorg/modules/input/libinput_drv.so
No symbol table info available.
#11 0x000055bb62d78c33 in ?? ()
No symbol table info available.
#12 0x000055bb62d7b331 in ?? ()
No symbol table info available.
#13 0x000055bb62d78a7e in ?? ()
No symbol table info available.
#14 0x00007f6b840fd6db in start_thread (arg=0x7f6b6affd700) at pthread_create.c:463
        pd = 0x7f6b6affd700
        now = <optimised out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140099333379840, 2214544212643191700, 140099333377216, 0, 0, 140734716615744, -2275037141702632556, -2275254433581720684}, mask_was_saved = 0}}, priv = {
            pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimised out>
#15 0x00007f6b83e2688f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
No locals.

```

jigpu, 2019-08-22:
Looks like the symbols might be in another package. Please try following the instructions at https://wiki.ubuntu.com/Debug%20Symbol%20Packages#Getting_-dbgsym.ddeb_packages to add the "ddebs" repository and then run `sudo apt-get install xserver-xorg-core-dbgsym`. Afterwards, try the coredumpctl command again and see if you get more information.

cohen990, 2019-09-03:
sorry for the slow response. Just got back from holiday

adding the ddebs repository seems to have changed things a bit

cohen990, 2019-09-03:
```
~ » coredumpctl gdb /usr/lib/xorg/Xorg                                                                                                                                                                dan@danbuntu
           PID: 3582 (Xorg)
           UID: 1000 (dan)
           GID: 1000 (dan)
        Signal: 11 (SEGV)
     Timestamp: Mon 2019-09-02 16:05:37 BST (17h ago)
  Command Line: /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
    Executable: /usr/lib/xorg/Xorg
 Control Group: /user.slice/user-1000.slice/session-13.scope
          Unit: session-13.scope
         Slice: user-1000.slice
       Session: 13
     Owner UID: 1000 (dan)
       Boot ID: 0ebf621cbd864e09a8063a969f0a19d9
    Machine ID: a8c84753764447e7b1bf3bd00476dcd8
      Hostname: danbuntu
       Storage: /var/lib/systemd/coredump/core.Xorg.1000.0ebf621cbd864e09a8063a969f0a19d9.3582.1567436737000000.lz4
       Message: Process 3582 (Xorg) of user 1000 dumped core.
                
                Stack trace of thread 3597:
                #0  0x00007fe3294d9ea6 n/a (modesetting_drv.so)
                #1  0x000055e78d9170f3 n/a (Xorg)
                #2  0x000055e78d921fa2 n/a (Xorg)
                #3  0x000055e78d9ee53b n/a (Xorg)
                #4  0x000055e78d9ef309 miPointerSetPosition (Xorg)
                #5  0x000055e78d8b35de n/a (Xorg)
                #6  0x000055e78d8b3d80 n/a (Xorg)
                #7  0x000055e78d8b55bf GetPointerEvents (Xorg)
                #8  0x000055e78d8b5b60 QueuePointerEvents (Xorg)
                #9  0x000055e78d8edbd9 xf86PostMotionEventP (Xorg)
                #10 0x00007fe31041eecc n/a (wacom_drv.so)
                #11 0x00007fe310420233 n/a (wacom_drv.so)
                #12 0x00007fe31042589d n/a (wacom_drv.so)
                #13 0x00007fe31041d677 n/a (wacom_drv.so)
                #14 0x00007fe31041d8d3 n/a (wacom_drv.so)
                #15 0x000055e78da04c33 n/a (Xorg)
                #16 0x000055e78da07331 n/a (Xorg)
                #17 0x000055e78da04a7e n/a (Xorg)
                #18 0x00007fe32c83b6db start_thread (libpthread.so.0)
                #19 0x00007fe32c56488f __clone (libc.so.6)
                
                Stack trace of thread 3582:
                #0  0x00007fe32c5596e7 __GI___writev (libc.so.6)
                #1  0x000055e78da0b16e n/a (Xorg)
                #2  0x000055e78da05bfd n/a (Xorg)
                #3  0x000055e78da061ce WriteToClient (Xorg)
                #4  0x000055e78d8a8432 WriteEventsToClient (Xorg)
                #5  0x000055e78d8a85a4 TryClientEvents (Xorg)
                #6  0x000055e78d8abe52 n/a (Xorg)
                #7  0x000055e78d8abf8e DeliverEventsToWindow (Xorg)
                #8  0x000055e78d98f4f2 n/a (Xorg)
                #9  0x000055e78d99ada1 n/a (Xorg)
                #10 0x000055e78d99b6ed XIChangeDeviceProperty (Xorg)
                #11 0x000055e78d99c5f0 n/a (Xorg)
                #12 0x000055e78d89ce98 n/a (Xorg)
                #13 0x000055e78d8a0ee0 n/a (Xorg)
                #14 0x00007fe32c464b97 __libc_start_main (libc.so.6)
                #15 0x000055e78d88ab8a _start (Xorg)
                
                Stack trace of thread 3592:
                #0  0x00007fe32c8419f3 futex_wait_cancelable (libpthread.so.0)
                #1  0x00007fe31f6c923a n/a (nouveau_dri.so)
                #2  0x00007fe31f6c8f67 n/a (nouveau_dri.so)
                #3  0x00007fe32c83b6db start_thread (libpthread.so.0)
                #4  0x00007fe32c56488f __clone (libc.so.6)
                
                Stack trace of thread 3583:
                #0  0x00007fe32c8419f3 futex_wait_cancelable (libpthread.so.0)
                #1  0x00007fe326fa70aa n/a (i965_dri.so)
[... 168 lines cut ...]
            events = 4294967295, data = {ptr = 0x1, fd = 1, u32 = 1, u64 = 1}}, {events = 4, data = {ptr = 0x31, fd = 49, u32 = 49, u64 = 49}}, {events = 0, data = {ptr = 0x0, fd = 0, u32 = 0, u64 = 0}}, {
            events = 0, data = {ptr = 0x7c00000000, fd = 0, u32 = 0, u64 = 532575944704}}, {events = 119, data = {ptr = 0x5b0000006e, fd = 110, u32 = 110, u64 = 390842024046}}, {events = 4093640736, data = {
              ptr = 0x3000007fe2, fd = 32738, u32 = 32738, u64 = 206158462946}}, {events = 0, data = {ptr = 0x17bc0, fd = 97216, u32 = 97216, u64 = 97216}}, {events = 48, data = {ptr = 0x2c8338f800000000, 
              fd = 0, u32 = 0, u64 = 3207469997420969984}}, {events = 32739, data = {ptr = 0x7fe2f4009440, fd = -201288640, u32 = 4093678656, u64 = 140612733015104}}, {events = 4093640736, data = {
              ptr = 0x90e5542000007fe2, fd = 32738, u32 = 32738, u64 = 10440843807544082402}}, {events = 21991, data = {ptr = 0x7fe32c4dd0b1 <__libc_calloc+129>, fd = 743297201, u32 = 743297201, 
              u64 = 140613677600945}}, {events = 2378809600, data = {ptr = 0x55e7, fd = 21991, u32 = 21991, u64 = 21991}}, {events = 0, data = {ptr = 0x7fe32c83f7df <__GI___pthread_mutex_unlock+63>, 
              fd = 746846175, u32 = 746846175, u64 = 140613681149919}}, {events = 1, data = {ptr = 0x8da04c1000000000, fd = 0, u32 = 0, u64 = 10205240387224731648}}, {events = 21991, data = {
              ptr = 0x55e78da07058 <ospoll_add+408>, fd = -1918865320, u32 = 2376101976, u64 = 94453001908312}}, {events = 4093678672, data = {ptr = 0x2c8452cc00007fe2, fd = 32738, u32 = 32738, 
[... cut ...]
        pd = 0x7fe313fff700
        now = <optimised out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140613269845760, 4356930679972880053, 140613269843136, 0, 0, 140736612012496, -4345750846319305035, -4345775341780017483}, mask_was_saved = 0}}, priv = {
            pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimised out>
#22 0x00007fe32c56488f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
No locals.
(gdb) 
(gdb) quit
```

jigpu, 2019-09-03:
This looks like some kind of issue with either the Xorg server or its modesetting driver. The event our driver sends seems perfectly sensible, and the screen coordinates the server transforms them to seems fairly reasonable as well. For some reason, however, there isn't a cursor to be displayed and the modesetting driver crashes as a result.

I would file an issue at https://gitlab.freedesktop.org/xorg/xserver/issues with information about your system and the dump you posted above. It would also be good to include a link back to this issue for reference. The developers over there should be better able to track down the cause of the issue.

cohen990, 2019-09-06:
issue filed. Thanks for your help so far!

jigpu, 2019-10-28:
Ref to upstream bug: https://gitlab.freedesktop.org/xorg/xserver/issues/883

closed 2019-10-28 by jigpu