linuxwacom/input-wacom #110: Wacom Touchscreen Not Working After Suspend

cm3z4, 2019-08-16:
Hello,

I'm having a hard time understanding why my (ThinkPad - Yoga 12) laptop's touchscreen stops working after a suspended state.

Basically, the touchscreen works just fine with a fresh installation of Ubuntu/Gnome 18.04 or 19.04. However, after the computer suspends, the touchscreen no longer works at all. This happens both in Xorg and Wayland. A reboot doesn't fix the issue.

Things I've already tried...
1. Updated bios.
2. modprobe -r wacom / modprobe wacom doesn't fix the issue.
3. Reboot.

See below system logs.

**uname -a**

Linux yoda-12 5.0.0-25-generic #26~18.04.1-Ubuntu SMP Thu Aug 1 13:51:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

**dmesg**

[    2.975207] wacom 0003:056A:501B.0005: hidraw3: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-5/input0
[    3.094681] wacom 0003:056A:501B.0006: hidraw4: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-5/input1
[    9.018255] wacom 0003:056A:501B.0005: wacom_set_report: ran out of retries (last error = -110)
[ 2728.305357] wacom 0003:056A:501B.000B: hidraw3: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-5/input0
[ 2728.438077] wacom 0003:056A:501B.000C: hidraw4: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-5/input1
[ 2734.332379] wacom 0003:056A:501B.000B: wacom_set_report: ran out of retries (last error = -110)
[ 2814.309342] wacom 0003:056A:501B.0011: hidraw3: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-5/input0
[ 2814.470232] wacom 0003:056A:501B.0012: hidraw4: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-5/input1
[ 2820.348589] wacom 0003:056A:501B.0011: wacom_set_report: ran out of retries (last error = -110)
[ 2994.901374] wacom 0003:056A:501B.0017: hidraw3: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-5/input0
[ 2995.065018] wacom 0003:056A:501B.0018: hidraw4: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-5/input1
[ 3001.084367] wacom 0003:056A:501B.0017: wacom_set_report: ran out of retries (last error = -110)
[30568.762817] wacom 0003:056A:501B.0017: wacom_set_report: ran out of retries (last error = -110)

**lsusb**

Bus 001 Device 003: ID 0483:91d1 STMicroelectronics Sensor Hub
Bus 001 Device 002: ID 8087:8001 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 04f2:b46a Chicony Electronics Co., Ltd 
Bus 002 Device 004: ID 056a:501b Wacom Co., Ltd 
Bus 002 Device 002: ID 1b1c:1b65 Corsair 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**lsmod**

wacom                 110592  0
usbhid                 53248  1 wacom
hid                   126976  4 wacom,usbhid,hid_sensor_hub,hid_generic

**xinput**

⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer:15                     	id=6	[slave  pointer  (2)]
⎜   ↳ xwayland-relative-pointer:15            	id=7	[slave  pointer  (2)]
⎜   ↳ xwayland-stylus:15                      	id=10	[slave  pointer  (2)]
⎜   ↳ xwayland-eraser:15                      	id=11	[slave  pointer  (2)]
⎜   ↳ xwayland-cursor:15                      	id=12	[slave  pointer  (2)]
⎜   ↳ xwayland-touch:15                       	id=9	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ xwayland-keyboard:15

**modinfo**

version:        v2.00
srcversion:     6669036FD12FD0BF16F93C5

Thanks in advance for your time and effort.

jigpu, 2019-08-16:
A couple of clarifications, if you don't mind:

* You didn't mention this, but I assume a cold boot fixes the issue (despite a reboot not working)? 
* Does the device work normally after a reboot as long as you didn't previously suspend?
* I see the device being detected multiple times with delays of a minute-or-so in between -- did you suspend your system at these points in time?
* If you have Windows installed, can you check that the device behaves normally there?

Issues that persist across modprobe and reboot are pretty rare and usually indicate that the device (or something upstream from it) is being misconfigured somehow. This could be an ACPI issue or the hardware not responding to our driver's configuration commands in an expected way.

cm3z4, 2019-08-16:
> A couple of clarifications, if you don't mind:
>
> * You didn't mention this, but I assume a cold boot fixes the issue (despite a reboot not working)?
> * Does the device work normally after a reboot as [...]

- A cold boot nor a reboot fixes the issue.
- After a fresh install, the touchscreen will continue to work after reboots, until it's suspended. After that, it stops working.
- I can't remember, possiblely so.
- Yes, the touchscreen works just fine on Windows.

The only way to get the screen working again is to perform a clean install of Ubuntu/Gnome. It will continue to work just fine until it suspends. It stops working after that and the errors occur.

jigpu, 2019-08-16:
Thanks. This sounds reminiscent of issue #59, which involves a Yoga 12 touchscreen working within the Manjaro installer, but not after rebooting into the installed OS. Where things are a little different is that you are able to get your device working inside Ubuntu 18.04 while that didn't seem to be possible in the other issue.

You might try reading the "Trouble Booting" section of https://wiki.ubuntu.com/DebuggingACPI and seeing if any of the suggested boot options make a difference in the touchscreen behavior.

If none of that works, can I have you install and run [hid-recorder](http://bentiss.github.io/hid-replay-docs/) in both the broken and working states? In each case, once hid-recorder is running make a few single-finger swipes with your finger and then press CTRL+C. You can paste the output from each run here for review.

cm3z4, 2019-08-17:
> Thanks. This sounds reminiscent of issue #59, which involves a Yoga 12 touchscreen working within the Manjaro installer, but not after rebooting into the installed OS. Where things are a little [...]

I tried the ACPI grub changes and none of the options worked. However, there was one case after a cold boot is started to work. So, I disabled all Ubuntu suspend features and it survived a reboot and cold boot. Then, like a real dummy, I tried to see if it would survive the hibernate state. And, after that, I haven't been able to reenable the touchscreen.

I installed hid-tools with pip3 and echoed the output to different files, accordingly.

These are the files for the touchscreen in a nonworking state.

**bin sudo ./hid-recorder**
Available devices:
/dev/hidraw0:	Corsair CORSAIR HARPOON RGB WIRELESS Gaming Dongle
/dev/hidraw1:	Corsair CORSAIR HARPOON RGB WIRELESS Gaming Dongle
/dev/hidraw2:	Corsair CORSAIR HARPOON RGB WIRELESS Gaming Dongle
/dev/hidraw3:	Wacom Co.,Ltd. Pen and multitouch sensor
/dev/hidraw4:	Wacom Co.,Ltd. Pen and multitouch sensor
Select the device event number [0-4]:

[wacom_3.txt](https://github.com/linuxwacom/input-wacom/files/3511535/wacom_3.txt)
[wacom_4.txt](https://github.com/linuxwacom/input-wacom/files/3511536/wacom_4.txt)

I hope I using hid-recorder correctly???
If so, I'll get you the logs once I can get the screen to work again.
Thanks.

cm3z4, 2019-08-17:
I was able to get the touchscreen working.
Here is how:
I installed the Xfce DE for testing and shut down the laptop. When I reached the login screen, it was working again. I've attached the hid-recorder logs with the touchscreen working. Hope this helps.

Thanks.

[wacom_3W.txt](https://github.com/linuxwacom/input-wacom/files/3511570/wacom_3W.txt)
[wacom_4W.txt](https://github.com/linuxwacom/input-wacom/files/3511571/wacom_4W.txt)

cm3z4, 2019-08-19:
I've found somewhat of a solution to my problem...

First, I installed Xfce. I then logged out of Gnome and into Xfce. I then closed my eyes and prayed to the computer gods. I then logged out of Xfce and then back into Gnome. Bam! The touchscreen worked again. I hope this helps anyone else having similar issues.

I must repeat those steps if the computer suspends for any reseason.

cm3z4, 2019-08-21:
Any update to as if this is a DE or driver issue? My quick fix has stopped working...

jigpu, 2019-08-21:
Sorry, I'm still not totally sure what's going on. The first pair of logs didn't have any events in them, which means that the hardware isn't sending anything for our driver to convert. Your ACPI checks make me more suspicious of our kernel driver mis-configuring the hardware, but its really hard to guess exactly how (especially since it _sometimes_ works okay).

The sensor is connected over an (internal) USB link, so maybe its possible to use something like Wireshark within Windows to capture the USB traffic that occurs when the system enters and leaves suspend. I'm not sure that would actually work, however, since Windows may very well suspend Wireshark before the hardware, and resume the hardware before Wireshark.

cm3z4, 2019-08-21:
I highly doubt the Wireshark method would work, as you stated the computer would most likely suspend Wireshark before the hardware. Is there any other data I can gather or monitor to help out this issue? I'd like to solve this, I enjoy using my touchscreen when it works...

jigpu, 2019-09-03:
No progress to report, unfortunately. I can't think of anything else to try at the moment...

As for the dmesg log, that's simply warning you that your module hasn't been signed with a trusted key. The kernel still loaded the module, but marked itself as "tainted" by this untrusted code. See https://github.com/linuxwacom/input-wacom/wiki/Kernel-Module-Signing for a little more information.

cm3z4, 2019-09-12:
Okay, well thanks for your feedback. I'll continue to troubleshoot on my own. I'm determined to find a solution.

cm3z4, 2019-09-13:
> @cm3z4 I'm seeing this same issue, on an X1 Yoga 3rd gen. Been trying to pin it down for ages.
> 
> One guess of mine there is some firmware race condition between the hardware lid switch and the touchscreen. For me, closing and opening the lid quicky and then powering it on from sleep with the power button gets it working.
> 
> I'm willing to provide logs or run tests, if it would help.

I suppose that could be the issue, I'm not sure. What baffles me is that a reboot doesn't fix the problem. I've been trying all kinds of things. Sometimes, it will work after a reboot, that is until it suspends again. I'll try your method that works on your X1 and report the outcome. Thanks, Caleb.

dubit0, 2019-11-21:
I'm experiencing a very similar issue on my Lifebook u939x. 

**uname -a**
`Linux cedar 5.3.11-arch1-1 #1 SMP PREEMPT Tue, 12 Nov 2019 22:19:48 +0000 x86_64 GNU/Linux`

**lsusb**
`Bus 001 Device 008: ID 056a:51dd Wacom Co., Ltd `

The Wacom touchscreen & stylus are working until the system is suspended. After suspend, neither the touchscreen nor the stylus are working. hid-recorder is showing no events on the hidraw devices after suspend. Also cat'ing the event device files ind /dev/input show no output after suspend.

However, in my case a cold start as well as reboot fix the issue and render the touchscreen working again. So if it is an issue of how the driver initializes the device, it is done correctly during booting but not after suspend.

Is there some debug trace I could record that could help to pin down the problem?

dubit0, 2019-11-22:
I just realized that the `wacom` kernel module is not even taking control over the USB device. Instead, the `usbhid` module claims the wacom touch screen. However, when coming back from suspend that wacom module seems to interfere with the usbhid module and renders the touchscreen not working.

Blacklisting the `wacom` module fixes the problem.

I'm not sure if this the intended behavior or if the wacom module should have claimed the device in the first place.

GRbit, 2020-01-21:
@dubit0 
> Blacklisting the `wacom` module fixes the problem.

Let me understand you right. Did you do something like this:
```bash
echo 'blacklist wacom' >> /etc/modprobe.d/blacklist.conf
update-initramfs -u
reboot
```
And then touchscreen worked after entering sleep by closing lid and waking after sleep with lid opening?

cronzevid, 2020-08-04:
Jesus Christ
I've been struggling with my U939X for about half a year (touchscreen not working after lid close/suspend), and I can confirm this works:
>Blacklisting the wacom module fixes the problem.

Noticeable thing is that earlier, for me cold boot/reboot was also fixing the issue.
`Linux host 5.8.0-1-MANJARO #1 SMP PREEMPT Mon Jul 13 09:41:40 UTC 2020 x86_64 GNU/Linux`

caleb-allen, 2021-01-03:
Is anybody able to try out the linux kernel v5.4.86? I found a commit in the changelog which sounds like it could be related.

https://github.com/torvalds/linux/commit/b08221c40febcbda9309dd70c61cf1b0ebb0e351
> Recently we met a touchscreen problem on some Thinkpad machines, the
touchscreen driver (i2c-hid) is not loaded and the touchscreen can't
work.

caleb-allen, 2021-01-03:
There is also this commit from Nov 2020:

https://github.com/torvalds/linux/commit/e40cc1b476d60f22628741e53cf3446a29e6e6b9
> platform/x86: thinkpad_acpi: Send tablet mode switch at wakeup time

mairin, 2021-01-20:
@caleb-allen where did you get 5.4.86 from? That commit is for 5.10-rc7... I'm currently running kernel-5.8.9-200 and it was released in Sept. (I'm looking to see if I can get a 5.10.x running now, since I have the same issue :-p )

caleb-allen, 2021-01-20:
@mairin I think I found that commit backported to 5.4.86, in addition for 5.10.3 (?) maybe? I don't recall exactly how I came up with that version though :\

mairin, 2021-01-20:
@caleb-allen so I just tried 5.10.8-100 (fedora 32) and the tablet / touch both worked ok after waking from suspend. It also uhh caused the system to randomly lock hard so maybe another 5.10.x series would do.

mairin, 2021-01-20:
Trying 5.10.9-100 now, it was built this morming 🤞

caleb-allen, 2021-01-20:
I seem to still have the issue on `5.4.89-1-MANJARO`. I'll give it a shot with `5.10.7-3`

caleb-allen, 2021-01-20:
Argh. Issue is still present for me on `5.10.7-3` until I wake up the device from sleep using the power button

mairin, 2021-01-20:
No luck with 5.10.9-100. I do a fresh boot, log in, fire up inkscape, touch pen to screen and as soon as pen touches to screen system locks up hard - screen/clock frozen, will not accept kb or mouse input. I ran with kernel debug option installed and running.... last few lines in journalctl appear to indicate it happens so fast it's not getting logged:

```
Jan 20 11:05:36 pandariomh systemd[1716]: dbus-:1.2-org.gnome.clocks@0.service: Succeeded.
Jan 20 11:05:39 pandariomh systemd[1716]: dbus-:1.2-org.gnome.Boxes.SearchProvider@0.service: Succeeded.
Jan 20 11:05:39 pandariomh systemd[1716]: dbus-:1.2-org.gnome.Contacts.SearchProvider@0.service: Succeeded.
Jan 20 11:05:41 pandariomh systemd[1716]: dbus-:1.2-org.gnome.Nautilus@0.service: Succeeded.
Jan 20 11:05:41 pandariomh systemd[1716]: dbus-:1.2-org.gnome.Documents@0.service: Succeeded.
Jan 20 11:05:41 pandariomh systemd[1]: systemd-hostnamed.service: Succeeded.
Jan 20 11:05:41 pandariomh audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 20 11:05:41 pandariomh audit[369]: NETFILTER_CFG table=nat family=7 entries=0 op=xt_unregister pid=369 subj=system_u:system_r:kernel_t:s0 comm="kworker/u16:3"
Jan 20 11:05:41 pandariomh audit[369]: NETFILTER_CFG table=broute family=7 entries=0 op=xt_unregister pid=369 subj=system_u:system_r:kernel_t:s0 comm="kworker/u16:3"
Jan 20 11:05:41 pandariomh audit[369]: NETFILTER_CFG table=filter family=7 entries=0 op=xt_unregister pid=369 subj=system_u:system_r:kernel_t:s0 comm="kworker/u16:3"
Jan 20 11:05:41 pandariomh audit: BPF prog-id=26 op=UNLOAD
Jan 20 11:05:41 pandariomh audit: BPF prog-id=25 op=UNLOAD
Jan 20 11:05:42 pandariomh tracker-store[2451]: OK
Jan 20 11:05:42 pandariomh systemd[1716]: tracker-store.service: Succeeded.
```

I suspect some sort of diff between 5.10.8 and 5.10.9 might help figure out why it might be locking hard as soon as pen touches screen in 5.10.9. Not sure about 5.10.8, the hangs seem random to me.

jigpu, 2021-01-20:
If your machine has a built-in ethernet port and you have access to a second machine, you might try enabling the kernel's "netconsole" debug option. You'll need to install netcat onto the second machine, and then run [my netconsole.sh script](https://gist.githubusercontent.com/jigpu/dc53bc5da68977b28d15eca083d33c30/raw/23169e90f9f53f36bbb48a57a3608b97cb27a737/netconsole.sh) to enable it. Provide the IP address of the second machine and the value 7 (corresponding to KERN_DEBUG log level) as command-line arguments.

Once you see dmesg logs appearing on the remote system, bring the pen into prox and cross your fingers that the kernel is able to send out the data...

mairin, 2021-01-20:
> If your machine has a built-in ethernet port and you have access to a second machine,

No ethernet port, but I may have a usb-c hub with an etherport port. Hopefully that'd work. I will give it a try this evening (my daughter is using said usb-c hub right now for her remote schooling :-p )

jigpu, 2021-01-20:
Think I may have found the source and fix for the crash. See https://github.com/linuxwacom/input-wacom/issues/230#issuecomment-763976561 and give it a shot :)

mairin, 2021-01-21:
@jigpu fix-230 got rid of the hard hangs, yay! However, if I suspend via shutting the lid and reopen the lid, the wacom still doesn't work. If I suspend via software (the GNOME menu) and then unsuspend, wacom works like a champ.

Some things I noticed, since I didn't realize before now there was even a difference between suspend via lid shut vs suspend via GNOME menu -
1. On this system if you suspend via lid shut, it auto unsuspends when you open the lid. You don't have to press a button to initiate wake up. So I guess lidswitch one way, suspend, lidswitch other way, wake up
2. On this system if you suspend from the OS menu, you wake it up by hitting the power button on the side. So in this use case there's no hw button or switch to enter suspend, only a hw button to wake up

Going to give it another try quick to see if I can get any errors in the journal

mairin, 2021-01-21:
OK here's my logs of the hw suspend, with just lines I think are relevant (it's 99% NetworkManager spew and GNOME shell deprecation warnings lol):

**From bootup - this shows us USB device number 5 (1-10) is the wacom stuff:**
```
Jan 20 16:53:05 pandariomh kernel: usb 1-10: new full-speed USB device number 5 using xhci_hcd
Jan 20 16:53:05 pandariomh kernel: usb 1-10: New USB device found, idVendor=056a, idProduct=5146, bcdDevice= 0.03
Jan 20 16:53:05 pandariomh kernel: usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jan 20 16:53:05 pandariomh kernel: usb 1-10: Product: Pen and multitouch sensor
Jan 20 16:53:05 pandariomh kernel: usb 1-10: Manufacturer: Wacom Co.,Ltd.
Jan 20 16:53:05 pandariomh kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor Touchscreen as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:056A:5146.0004/input/input7
Jan 20 16:53:05 pandariomh kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:056A:5146.0004/input/input8
Jan 20 16:53:05 pandariomh kernel: hid-generic 0003:056A:5146.0004: input,hiddev96,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input0
Jan 20 16:53:05 pandariomh kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:056A:5146.0005/input/input9
Jan 20 16:53:05 pandariomh kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:056A:5146.0005/input/input10
Jan 20 16:53:05 pandariomh kernel: input: Wacom Co.,Ltd. Pen and multitouch sensor Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:056A:5146.0005/input/input11
Jan 20 16:53:05 pandariomh kernel: hid-generic 0003:056A:5146.0005: input,hiddev97,hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input1
Jan 20 16:53:05 pandariomh kernel: input: Wacom Pen and multitouch sensor Finger as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:056A:5146.0004/input/input13
Jan 20 16:53:05 pandariomh kernel: wacom 0003:056A:5146.0004: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input0
Jan 20 16:53:05 pandariomh kernel: scsi 0:0:0:0: Direct-Access     Generic- SD/MMC CRW       1.00 PQ: 0 ANSI: 6
Jan 20 16:53:05 pandariomh kernel: sd 0:0:0:0: Attached scsi generic sg0 type 0
Jan 20 16:53:05 pandariomh kernel: input: Wacom Pen and multitouch sensor Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:056A:5146.0005/input/input15
Jan 20 16:53:05 pandariomh kernel: wacom 0003:056A:5146.0005: hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input1
Jan 20 16:53:05 pandariomh kernel: i915 0000:00:02.0: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
```

**From the moment the lid shut closed... see the wacom USB disconnect then the loss of the device?**
```
Jan 20 22:07:40 pandariomh kernel: usb 1-10: USB disconnect, device number 5
Jan 20 22:07:40 pandariomh systemd-logind[708]: Lid closed.
Jan 20 22:07:40 pandariomh systemd-logind[708]: Suspending...
Jan 20 22:07:40 pandariomh NetworkManager[784]: <info>  [1611198460.7309] manager: sleep: sleep requested (sleeping: no  enabled: yes)
Jan 20 22:07:40 pandariomh NetworkManager[784]: <info>  [1611198460.7312] device (p2p-dev-wlp2s0): state change: disconnected -> unmanaged (reason 'sleeping', sys-iface-state: 'managed')
Jan 20 22:07:40 pandariomh NetworkManager[784]: <info>  [1611198460.7318] device (18:54:CF:2D:39:A2): state change: disconnected -> unmanaged (reason 'sleeping', sys-iface-state: 'managed')
Jan 20 22:07:40 pandariomh NetworkManager[784]: <info>  [1611198460.7324] device (30:96:FB:D5:BF:9E): state change: disconnected -> unmanaged (reason 'sleeping', sys-iface-state: 'managed')
Jan 20 22:07:40 pandariomh NetworkManager[784]: <info>  [1611198460.7330] manager: NetworkManager state is now ASLEEP
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (EE) Wacom Pen and multitouch sensor Finger touch: Error reading wacom device : No such device
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (II) config/udev: removing device Wacom Pen and multitouch sensor Finger touch
Jan 20 22:07:40 pandariomh mono[2469]: gdk_device_update_tool: assertion 'GDK_IS_DEVICE (device)' failed
Jan 20 22:07:40 pandariomh WebKitWebProces[2723]: gdk_device_update_tool: assertion 'GDK_IS_DEVICE (device)' failed
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (II) UnloadModule: "wacom"
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (II) systemd-logind: releasing fd for 13:70
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (EE) Wacom Pen and multitouch sensor Pen eraser: Error reading wacom device : No such device
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (II) config/udev: removing device Wacom Pen and multitouch sensor Pen stylus
Jan 20 22:07:40 pandariomh WebKitWebProces[2723]: gdk_device_update_tool: assertion 'GDK_IS_DEVICE (device)' failed
Jan 20 22:07:40 pandariomh mono[2469]: gdk_device_update_tool: assertion 'GDK_IS_DEVICE (device)' failed
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (II) UnloadModule: "wacom"
[... cut ...]
Jan 20 22:07:40 pandariomh WebKitWebProces[2723]: gdk_device_update_tool: assertion 'GDK_IS_DEVICE (device)' failed
Jan 20 22:07:40 pandariomh mono[2469]: gdk_device_update_tool: assertion 'GDK_IS_DEVICE (device)' failed
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (II) UnloadModule: "wacom"
Jan 20 22:07:40 pandariomh /usr/libexec/gdm-x-session[1694]: (II) systemd-logind: releasing fd for 13:71
Jan 20 22:07:40 pandariomh upowerd[1305]: g_source_remove: assertion 'tag > 0' failed
```
Is this helpful? Anything else I could do to get useful logs?

mairin, 2021-01-21:
Oh in case the wakeup sections of the logs are useful.... so when I resumed from software suspend heres what it looked like (remembering usb 1-10 is wacom hw):

```
Jan 20 21:54:39 pandariomh systemd-sleep[11180]: System resumed.
Jan 20 21:54:39 pandariomh kernel: PM: suspend exit
Jan 20 21:54:39 pandariomh systemd[1]: systemd-suspend.service: Succeeded.
Jan 20 21:54:39 pandariomh audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-suspend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 20 21:54:39 pandariomh audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-suspend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 20 21:54:39 pandariomh systemd[1]: Finished Suspend.
Jan 20 21:54:39 pandariomh upowerd[1305]: treating change event as add on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-10

```

After wakeup from shutting the lid and reopening the lid, there is no mention of '1-10' 'wacom' 'device number 5' etc in the logs. the last mention in the log is from right after the lidswitch was triggered and the lid was shut:
`Jan 20 22:07:40 pandariomh kernel: usb 1-10: USB disconnect, device number 5`

jigpu, 2021-01-21:
Those are some _very_ insightful observations! The difference between lidswitch vs menu suspend behavior is not something I would have expected. The USB disconnect in the former case would also go a long way to explaining the difference in behavior and why I've never been able to replicate this locally (I'm pretty sure I've only tested on systems with I2C-connected sensors). 

This difference in behavior makes me immediately wonder if ACPI is involved at all, but I'm definitely no ACPI expert. I'll have to think about how to further investigate this particular disconnect / lidswitch correlation...

jigpu, 2021-01-27:
Just a minor update / note to future self: we had bug #232 filed the other day, and while it has a number of parallels with this issue (suspend/resume, behavior changed with recent kernel, blacklisting module fixes the problem) I'm suspecting that it is a distinct issue. In particular, this issue seems to have been around since at least Linux 5.0, while the other issue was probably introduced in Linux 5.8. Also, their sensor is connected over I2C while everyone in this issue all have USB-connected sensors. Finally, their symptom is the system as a whole not resuming, rather than just the sensor not resuming.

I'm still trying to determine a way to debug this particular issue further...

apteas, 2021-01-28:
I am a novice and just a user (with very little technical knowledge). I have the same problem of the touchscreen not showing up after waking from suspend on Yoga X1 3rd Gen. Suspend using `pm-suspend` or by closing the lid, and then waking up from suspend, both lead to the following outcome:
```
$ grep -i wacom /var/log/Xorg.0.log | tail -n 8
[  1140.202] (EE) Wacom Pen and multitouch sensor Finger touch: Error reading wacom device : No such device
[  1140.212] (EE) Wacom Pen and multitouch sensor Pen eraser: Error reading wacom device : No such device
[  1140.239] (II) config/udev: removing device Wacom Pen and multitouch sensor Finger touch
[  1140.266] (II) UnloadModule: "wacom"
[  1140.266] (II) config/udev: removing device Wacom Pen and multitouch sensor Pen stylus
[  1140.274] (II) UnloadModule: "wacom"
[  1140.274] (II) config/udev: removing device Wacom Pen and multitouch sensor Pen eraser
[  1140.310] (II) UnloadModule: "wacom"
```
After this, `xsetwacom --list` returns nothing, and the touchscreen does not show up in the list returned by `xinput` either.

On the other hand, if I use `systemctl suspend` then, after waking up, the touchscreen works just fine.
```
$ grep -i wacom /tmp/Xorg.0.log | tail -n 62
[  1146.681] (II) config/udev: Adding input device Wacom Pen and multitouch sensor Pen (/dev/input/mouse3)
[  1146.682] (II) config/udev: Adding input device Wacom Pen and multitouch sensor Finger (/dev/input/mouse2)
[  1146.818] (II) config/udev: Adding input device Wacom Pen and multitouch sensor Pen (/dev/input/event7)
[  1146.818] (**) Wacom Pen and multitouch sensor Pen: Applying InputClass "libinput tablet catchall"
[  1146.818] (**) Wacom Pen and multitouch sensor Pen: Applying InputClass "Wacom USB tablet class"
[  1146.818] (**) Wacom Pen and multitouch sensor Pen: Applying InputClass "Wacom tablet class"
[  1146.818] (II) Using input driver 'wacom' for 'Wacom Pen and multitouch sensor Pen'
[  1146.818] (**) Wacom Pen and multitouch sensor Pen: always reports core events
[  1146.877] (II) Wacom Pen and multitouch sensor Pen: type not specified, assuming 'stylus'.
[  1146.877] (II) Wacom Pen and multitouch sensor Pen: other types will be automatically added.
[  1146.877] (--) Wacom Pen and multitouch sensor Pen stylus: using pressure threshold of 851 for button 1
[  1146.877] (**) Wacom Pen and multitouch sensor Pen stylus: panscroll is 0
[  1146.877] (**) Wacom Pen and multitouch sensor Pen stylus: panscroll modified to 1300
[  1146.877] (--) Wacom Pen and multitouch sensor Pen stylus: maxX=30927 maxY=17397 maxZ=4095 resX=100000 resY=100000  tilt=enabled
[  1146.877] (II) Wacom Pen and multitouch sensor Pen stylus: hotplugging dependent devices.
[  1146.877] (EE) Wacom Pen and multitouch sensor Pen stylus: Invalid type 'cursor' for this device.
[  1146.877] (EE) Wacom Pen and multitouch sensor Pen stylus: Invalid type 'touch' for this device.
[  1146.877] (EE) Wacom Pen and multitouch sensor Pen stylus: Invalid type 'pad' for this device.
[  1146.877] (II) Wacom Pen and multitouch sensor Pen stylus: hotplugging completed.
[  1146.957] (II) XINPUT: Adding extended input device "Wacom Pen and multitouch sensor Pen stylus" (type: STYLUS, id 9)
[  1146.959] (**) Wacom Pen and multitouch sensor Pen stylus: (accel) keeping acceleration scheme 1
[  1146.959] (**) Wacom Pen and multitouch sensor Pen stylus: (accel) acceleration profile 0
[  1146.959] (**) Wacom Pen and multitouch sensor Pen stylus: (accel) acceleration factor: 2.000
[  1146.959] (**) Wacom Pen and multitouch sensor Pen stylus: (accel) acceleration threshold: 4
[  1147.019] (**) Wacom Pen and multitouch sensor Pen eraser: Applying InputClass "libinput tablet catchall"
[  1147.019] (**) Wacom Pen and multitouch sensor Pen eraser: Applying InputClass "Wacom USB tablet class"
[  1147.019] (**) Wacom Pen and multitouch sensor Pen eraser: Applying InputClass "Wacom tablet class"
[  1147.019] (II) Using input driver 'wacom' for 'Wacom Pen and multitouch sensor Pen eraser'
[  1147.019] (**) Wacom Pen and multitouch sensor Pen eraser: always reports core events
[  1147.020] (**) Wacom Pen and multitouch sensor Pen eraser: panscroll is 1300
[  1147.020] (**) Wacom Pen and multitouch sensor Pen eraser: panscroll modified to 1300
[  1147.020] (--) Wacom Pen and multitouch sensor Pen eraser: maxX=30927 maxY=17397 maxZ=4095 resX=100000 resY=100000  tilt=enabled
[  1147.045] (II) XINPUT: Adding extended input device "Wacom Pen and multitouch sensor Pen eraser" (type: ERASER, id 10)
[  1147.047] (**) Wacom Pen and multitouch sensor Pen eraser: (accel) keeping acceleration scheme 1
[  1147.047] (**) Wacom Pen and multitouch sensor Pen eraser: (accel) acceleration profile 0
[  1147.047] (**) Wacom Pen and multitouch sensor Pen eraser: (accel) acceleration factor: 2.000
[  1147.047] (**) Wacom Pen and multitouch sensor Pen eraser: (accel) acceleration threshold: 4
[  1147.050] (II) config/udev: Adding input device Wacom Pen and multitouch sensor Finger (/dev/input/event1)
[  1147.050] (**) Wacom Pen and multitouch sensor Finger: Applying InputClass "libinput touchscreen catchall"
[  1147.050] (**) Wacom Pen and multitouch sensor Finger: Applying InputClass "Wacom USB touchscreen class"
[  1147.051] (**) Wacom Pen and multitouch sensor Finger: Applying InputClass "Wacom touchscreen class"
[  1147.051] (II) Using input driver 'wacom' for 'Wacom Pen and multitouch sensor Finger'
[... cut ...]
I guess that there are still errors (which I don't understand), but finger, pen, eraser all three work fine.

p.s. I have an additional unusual problem: if I close the lid after the system suspends with `systemctl suspend` then the system wakes up! (Why would a 'wakeup' be triggered by closing the lid? I have no idea!) Anyway, here is a "fix" that works for now. I type in:
`systemctl suspend && sleep 20  && systemctl suspend`
After the system suspends with the first command, I close the lid. The system wakes up, but within 20 seconds, it suspends again! Terrible solution, but it works reproducibly!

mairin, 2021-01-29:
Hi @jigpu it appears my hardware enablement colleagues at Red Hat (Hans de Goede, Benjamin Tissoires, Jared Dominguez, and Carlos Soriano Sanchez) helped and they were able to figure this issue out.

**The Issue**

The BIOS firmware on this system (Lenovo Thinkpad Yoga Gen 3) has a feature under the "Config" tab called "Sleep State." On my system, this was set to "Linux". Hans noted that most modern hw (and windows defaults) to using s2idle suspend mode, so s2idle suspend is better tested. He suspects this is an embedded-controller bug triggered by s3 mode, due to s3 mode not really getting much testing.

Remember I noticed the bug doesnt occur with OS menu suspend... that triggers s2idle suspend. The lidswitch on these systems, when in Linux mode, triggers s3 suspend which is broken on them.

**The Solution**

Go into BIOS, Config, and change the "Sleep State" setting to "Windows" - this will apparently change the lidswitch to use s2idle suspend instead of s3 suspend, so it won't trigger the s3 suspend bug anymore.

If you have an affected system and have no Linux/Windows "Sleep State" setting in BIOS, here's some steps to try to force s2idle

run:

`cat /sys/power/mem_sleep`

output may look something like this:

`[s2idle] deep`

"deep" is s3 suspend. If both are listed, you can change it by running: 

`sudo sh -c 'echo s2idle > /sys/power/mem_sleep'`

To check that it changed, try

`cat /sys/power/mem_sleep`

And then try to suspend with the lid and wake up with the lid and see if the digitizer works.

mairin, 2021-01-29:
Oh this is a relevant kernel bug my colleague Benjamin Berg found:
https://bugzilla.kernel.org/show_bug.cgi?id=203667

apteas, 2021-01-31:
Hi @mairin, Thanks a lot for the pointers. The kernel bug you referred mentions device 056a:5146 whereas I have 056a:514a (on Yoga X1 3rd Gen, and `uname -srvmpio` gives `Linux 5.8.0-41-generic #46-Ubuntu SMP Mon Jan 18 16:48:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`) but the problem is identical .

I tried the following and both scenarios work reliably. (I don't understand anything about kernels or drivers so don't know whether this will be more useful to report here or on the thread https://bugzilla.kernel.org/show_bug.cgi?id=203667)

- If I change the "Sleep State" setting to "Windows" in BIOS:
  - when I close the lid, the system goes to sleep state S2
  - upon opening the lid, the touchscreen is detected and works fine

- If I change the "Sleep State" setting to "Linux" in BIOS:
  - when I close the lid, the system goes to sleep state S3
  - upon opening the lid, the touchscreen is _not_ detected
  - if I close and open the lid "quickly" then the touchscreen is detected. In this step, depending on how short the duration when the lid is closed, the sequence of events can be "touchscreen detected, then sleep-awake" ([dmesg-log1.txt](https://github.com/linuxwacom/input-wacom/files/5899425/dmesg-log1.txt)) or it can be "sleep-awake, then touchscreen detected" ([dmesg-log2.txt](https://github.com/linuxwacom/input-wacom/files/5899426/dmesg-log2.txt)) or for _very_ quick action, "touchscreen detected, without sleep-wake cycle" ([dmesg-log3.txt](https://github.com/linuxwacom/input-wacom/files/5899427/dmesg-log3.txt)).

For now, as a user, I don't mind the S3 sleep (less power used) with "open and then close-open quickly" sequence as a solution -- it works reliably and takes just a few extra seconds, even giving me time to think whether I really need to open the laptop in the first place!

p.s. After waking up from S3 sleep, `sudo /usr/sbin/rtcwake -m freeze -s 10` or `sudo /usr/sbin/rtcwake -m mem -s 10` (and even with smaller interval than 10 seconds, e.g., `-s 1` in the above command) does _not_ lead to the touchscreen being detected.

jigpu, 2021-02-05:
Thanks all so much for the continued research and workarounds. I've added this information to our ["Known Issues" wiki page](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#s3-suspend-bug) to make it easier to find in the future.

I've also stumbled across this link in the past few days which has a few other workarounds, including the `rtcwake` commands that @apteas had no luck with. Thought it might be another helpful source: https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_X1_Yoga_(Gen_3)#Fix_touchscreen_after_resume

aitorpazos, 2021-03-17:
For me, in a Lenovo Yoga X1 3rd gen, this workaround gives a seamless experience (as if it were working as expected):
- Install `acpi_call` and enable kernel module
- Add the following SystemD service:
```                                          
# /etc/systemd/system/activate-touch-hack.service
[Unit]
Description=Touch wake Thinkpad X1 Yoga 3rd gen hack
After=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target

[Service]
ExecStart=/bin/sh -c "echo '\\_SB.PCI0.LPCB.EC._Q2A'  > /proc/acpi/call"

[Install]
WantedBy=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
```

kubrickfr, 2021-05-09:
A very similar thing happens on the Dell XPS 9300 (touchscreen works after suspend only in s2idle mode).
Does anybody know how I can find the right ACPI call for this laptop?

middle-road, 2021-05-09:
I dug out my Lenovo Yoga 12 S1 this weekend to have another go at attempting to get the touchscreen to function properly. I last attempted trying to get it to function back around New Year's.

Re-visited this page to check for any developments.
Was interested in the mention of 5.1x perhaps working.
(Then again the Lenovo Yoga X1 is much better suited for Linux than the Yoga 12...)

I tried the latest release of Mint Cinnamon 20.1 first, fresh install. Same behavior as before.
uname -a 
`5.4.0-72-generic #80-Ubuntu SMP Mon Apr 12 17:35:00 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

Ask my Son for his take on it and he told me to try Fedora. He's running it and it's at 5.11
What a waste of time.
Fedora 34 / Cinnamon: Live Boot, TS was active until a 'suspend'. Did a fresh install and like before it would function until 'suspend'. Tried the 'mem_sleep' tweak to no avail.
uname -a 
`5.11.12-300.fc34.x86_64 #1 SMP Wed Apr 7 16:31:13 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

Fedora 34 / Wayland-KDE: TS active on Live boot, but when I did I fresh install of it, the TS did not function at all.

Fedora 34 Workstation: Same as KDE. 

LXQt: Didn't even boot up on Live.

Now comes the interesting part. Came across a blog post where the author detailed he experience with his new Yoga 12 and ubuntu 15.11 back in 2015.

I downloaded 15.11 and gave it a try and it works. Comes out of 'suspend' and is still functioning.
uname -a
`Linux ubuntu 4.2.0-16-generic #19-Ubuntu SMP Thu Oct 8 15:35:06 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux`
But now the problem is that I can't stand the DE of 15.11....

I then hunted down an image of Mint Cinnamon 17.3 and booted that, and it is functioning.
uname -a 
`Linux mint 3.19.0-32-generic #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015 x86_64 x86_64 x86_64` 

I have not tried a 'lid closed' action yet. I just need to get a laptop back out in my shop and really wanted to have a touch screen.
I'm going to dig out my other Yoga 12 and give that a try with a current distro - after I ask a question here of:
_Which might be the best distro for me to use_?
Now if I could pry our Yoga 14 from the Wife's clutches, I could try some troubleshooting on that also.

chethan0510, 2021-11-27:
I have a Lenovo Yoga 7i with Ubuntu 20.04 (Linux CK-Yoga-7-14ITL5 5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux). (To add, I am a novice when it comes to Linux.)

I also am having this issue: the touchscreen/stylus stops working after suspend or on folding (the occurence is quite random). Tried some of the solutions suggested here, but it didn't help.  What works for me is stopping and restarting the wacom module: 

`sudo rmmod wacom`
`sudo modprod wacom`

I'd be glad to provide any logs if it helps.

Pinglinux, 2022-08-03:
The issue is not in Wacom device/driver. It's probably a system or kernel issue. Jason has [summarized this discussion](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#s3-suspend-bug). A kernel [bug](https://bugzilla.kernel.org/show_bug.cgi?id=203667) related to the issue has been reported.

closed 2022-08-03 by Pinglinux