linuxwacom/input-wacom #111: Cintiq Pro 13 stops working after several minutes

zupazupa, 2019-08-19:
Hello,
I have a problem with my Cintiq Pro 13 (DTH-1320A). I've plugged it in to fresh Ubuntu 19.04 and it started working fine without any additional installations. But then I've realized that it is crashing after several minutes. By crashing I mean it is still working as a display but stops working as a tablet.
Turning it off and on helps, but only for a while. After several "offs and ons" whole graphic environment freezes (only mouse is working).
I've tried with installing input-wacom from sources (0.43.0) but it hasn't change anything.

I've checked the evemu-record command when tablet is crashed - there are no events.

_libwacom-list-local-devices_ still lists the tablet:

```
[Device]
Name=Wacom Cintiq Pro 13
DeviceMatch=usb:056a:034f;
PairedID=usb:056a:0353;
Class=Cintiq
Width=12
Height=7
IntegratedIn=Display;
Styli=0x842;0x84a;0x160802;0x16080a;0x100802;0x10080a;0x140802;0x14080a;0x120802;0x100804;0x10080c;0x100902;0x10090a;0x80842;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=true
TouchSwitch=false
StatusLEDs=
NumStrips=0
Buttons=0
```

_xsetwacom --list devices_ output:
```
Wacom Express Key Remote Pad pad	id: 15	type: PAD       
Wacom Cintiq Pro 13 Pen stylus  	id: 20	type: STYLUS    
Wacom Cintiq Pro 13 Pen eraser  	id: 21	type: ERASER  
```

I'm attaching two packages with output from your sysinfo.sh script: one when tablet is working fine and the second when it is not responding.
[sysinfo_zupa_crash.tar.gz](https://github.com/linuxwacom/input-wacom/files/3515973/sysinfo_zupa_crash.tar.gz)
[sysinfo_zupa_working.tar.gz](https://github.com/linuxwacom/input-wacom/files/3515974/sysinfo_zupa_working.tar.gz)

Please let me know if you need more information.

skomra, 2019-08-19:
Hi,
Is 19.04 the only OS you've tried this tablet with? Or was it previously working with a different OS?

skomra, 2019-08-19:
Also to make sure our driver is the only thing talking to the hardware please:

`sudo apt remove fwupd`

zupazupa, 2019-08-20:
Hi! Thanks for quick answer :)
Previously I've plugged it in to Ubuntu 18.04 and faced the same issue as described here #55 (I didn't even try to resolve it because I was changing the machine).
It is also working with Windows 10, but I guess that is not what you were asking about ;)
I can try it with Linux Mint and let you know.
I'll also try removing this packet and let you know about the results.

zupazupa, 2019-08-20:
Hello again! Things get a little bit more complicated... I had worked with this tablet on my old computer for about 2 weeks, and since I had a problem on Linux - I've been working on Windows without any problems. When I bought new machine I've just installed drivers on Windows, checked if it works, assumed that it is fine and switch to Linux. Today I've tried to work a little bit longer on Windows and found out that there is the same issue, as on Linux (after sevaral minutes it stopped working). 
I've started digging, and re-switching and finally discovered that Wacom works perfectly fine on both Linux and Windows, when connected via Wacom Link adapter and crashes on both systems when connected via USB-C directly to the graphic card. 
So I guess that the problem lies on Wacom or Nvidia side...? Do you have any idea where to report it?
Anyway thanks for replies :) I guess that the topic may be closed...

closed 2019-08-21 by zupazupa

Pinglinux, 2019-08-21:
Hi @zupazupa When you say"Wacom works perfectly fine on both Linux and Windows, when connected via Wacom Link adapter and crashes on both systems when connected via USB-C directly to the graphic card", do you mean tablet stopped working when you connected Cintiq to the graphic card directly via USB-C cable? That's understable since Cintiq needs to power to run properly. The two USB-C ports on Wacom Link Plus have dedicated roles: one to connect to Cintiq, the other one to connect to power. Without power, the setup won't work well. If you check Wacom Link Plus manual, you'll know what I mean.

zupazupa, 2019-08-22:
@Pinglinux no, that is not what I ment.
When I connect tablet with the USB-C cable to the graphic card and with second cable to power supply (according to manual) it works for several minutes and after that crashes - as I've described in the first post. It still displays, but does not receive any event from the stylus.
First I thought it was Linux specific bug, but than I've realized it happens also on Windows.

Pinglinux, 2019-08-22:
Ok, let’s make sure I get it right since I feel it could still be a connection issue. Those cables can be a bit confusing. How do you connect the USB-C to your graphic card, I mean where is the other end, on the Link or directly from the Cintiq?

zupazupa, 2019-08-22:
When the tablet is crashing: 

- USB-C cable is connected to USB-C socket on my graphic card and the other end is connected to the tablet (upper socket on left-hand side)
- Power supplier cable is connected to the USB-C  socket on the right side of the tablet

When tablet is working fine:

- HDMI cable between graphic card and Wacom link
- USB cable between computer and Wacom link
- USB-C cable between power supplier and Wacom link
- USB-C cable between Wacom link and tablet

Pinglinux, 2019-08-22:
Ok, I checked with our QA team. It is a known issue and it is on Nvidia side. We have contacted Nvidia. Their engineers are updating their driver. Please report the issue at Nvidia's website, if you want to speed up the process.

Thank you for using Wacom device on Linux!