linuxwacom/input-wacom #115: Cintiq 22hd not working

signup08, 2019-08-30:
Device doesn't work, no events from either of the two relevant /dev/input/ evdev entries, drivers in use listed under /sys/kernel/debug/usb/devices are usbhid and hub, I  assume it should be wacom, though the wiki is referencing /proc/bus/usb/devices, so it may be very outdated, device is in the list of supported devices on the wiki. Tried this under both Debian Buster (kernel 4.19) and a Fedora live cd (don't recall the kernel version), with the same results, device works as it should under Windows 10.

signup08, 2019-08-30:
For the record, the display itself connects just fine, forgot to mention that the first time around, issue is only with input.

jigpu, 2019-09-04:
Can you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) with the 22HD connected and attach the generated tarball to this issue?

bogdancovaciu, 2019-10-02:
Hello @jigpu and thanks for taking care of this. No particular issue with my Cintiq22hd regarding the drivers here, but i have to install the `xf86-input-wacom` to get the Stylus ready to be configured via Gnome Settings, or `kcm-wacomtablet` under KDE Plasma. Without `xf86-input-wacom` the Stylus works in relative mode over both monitors, and that is not desirable when drawing, and is not displayed in any of the Gnome or KDE Plasma Settings.
One thing i noticed in KDE Plasma is that LibreOffice or VirtualBOX interface do not respond to "Stylus Clicks". Had a short discussion on Manjaro forum about it when someone else mentioned it. See the discussion here:
https://forum.manjaro.org/t/wacom-pen-clicks-not-working-in-some-applications/105389

For me is not an issue, as i have a totally different workflow when drawing vs when i do "writing" or system tasks.
Because you mentioned the `sysinfo` but @signup08 never came back with it, i thought sharing mine will be helpful.
[sysinfo.AWZK6sbtqK.tar.gz](https://github.com/linuxwacom/input-wacom/files/3680391/sysinfo.AWZK6sbtqK.tar.gz)

Kind regards!

christiankaiser, 2020-04-26:
I came across the exactly same issue with my Cintiq 22HD. I did describe the problem [here](https://unix.stackexchange.com/questions/580143/wacom-pen-does-not-trigger-mouse-moving-on-cintiq-22hd-with-debian-buster), and the solution with switching the screen off and on again works fine.

I am on Debian Buster, but did try with a Ubuntu and KUbuntu live CD, with the same behaviour.

I can reproduce the problem easily as it appears again each time when I restart my computer.

Kind regards!

robertxgray, 2020-05-12:
Ubuntu 18.04 here. I am using the hack script described in #34 as a workaround, with $idProduct = 00f9 for my 22hd.

Pinglinux, 2022-08-12:
The root cause of those Cintiq related issues are not in Wacom driver, especially since the issue disappear after a power/USB cycle. It should be in the kernel/system configuration for kernels later than 4.15.17+. Feel free to open a new request if systems running kernel 5.18+ still have the issue.

closed 2022-08-12 by Pinglinux