linuxwacom/input-wacom #117: Building the driver makes my monitor unresponsive after logging in

linuxnoob123, 2019-09-09:
Hi, i tried to use this repository to install the driver for my wacom tablet. My machine is running Ubuntu 18.04.3 with a 5.1.0 kernel. The graphics card is a nvidia 1650 maxq. 

I followed [this](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) page on your wiki, however when i tried to `$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"` the terminal said that "then" was a syntax error. 

I then moved on to [this wiki site](https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver) and followed the guide.After executing this `$ sudo apt-get install autoconf pkg-config make xutils-dev libtool xserver-xorg-dev libx11-dev libxi-dev libxrandr-dev libxinerama-dev libudev-dev` command, i rebooted my system. 

Unfortunately after logging in, i am greeted by my regular desktop, except i cant move the curser nor do mouse buttons do anything nor can i launch programs from the gnome menu bar using system + (number). I have been able to circumvent this issue by enabling the wayland desktop or whatever its called (please ignore my ignorance). However i would be forever thankful if you just told me how to reverse this `$ sudo apt-get install autoconf pkg-config make xutils-dev libtool xserver-xorg-dev libx11-dev libxi-dev libxrandr-dev libxinerama-dev libudev-dev` command. I believe that this is the core of the issue because its affecting the x11 thing. 

I have tried to boot from 4.13 kernel, 5.0 Kernel, tried `sudo apt-get purge --auto-remove xserver-xorg-input-wacom` in the secure boot console. I have looked in my apt history log and the `...install autoconf ....`command was the only thing that had changed between the last reboot my system still properly worked and when it didnt work anymore.

If any files, information etc is needed I am happy to provide that. 
Best regards

jigpu, 2019-09-09:
Could you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball to this issue for review? Additionally, please provide a copy of your apt log.

skomra, 2019-09-09:
Hello,
Which tablet were you trying to use? Almost all tablets should be supported out of the box by the 5.1 kernel. 

It looks like you began an installation of the xf86-input-wacom driver, which you most likely also do not need. You can remove packages on ubuntu by issuing the command "sudo apt remove" in front of the package name.

linuxnoob123, 2019-09-10:
[sysinfo.D3MEha9jZH.tar.gz](https://github.com/linuxwacom/input-wacom/files/3595519/sysinfo.D3MEha9jZH.tar.gz)
[history.log](https://github.com/linuxwacom/input-wacom/files/3595530/history.log)
[term.log](https://github.com/linuxwacom/input-wacom/files/3595531/term.log)

jigpu, 2019-09-10:
Thanks. The logs indicate that running `apt-get install xserver-xorg-input-wacom` is what actually broke everything. You should be able to restore all of the removed packages by running the following:
~~~
$ sudo apt-get install xserver-xorg-input-libinput-hwe-18.04 xserver-xorg-video-vmware-hwe-18.04 xserver-xorg-video-intel-hwe-18.04 xserver-xorg-core-hwe-18.04 xserver-xorg-input-wacom-hwe-18.04 xserver-xorg-video-nouveau-hwe-18.04 xserver-xorg-input-all-hwe-18.04 xserver-xorg-video-radeon-hwe-18.04 ubuntu-desktop xserver-xorg-video-ati-hwe-18.04 xserver-xorg-video-qxl-hwe-18.04 xserver-xorg-video-vesa-hwe-18.04 xserver-xorg-video-amdgpu-hwe-18.04 xorg xserver-xorg-hwe-18.04 xserver-xorg-video-all-hwe-18.04 xserver-xorg-video-fbdev-hwe-18.04
~~~

Afterwards, if you could let us know what tablet you're trying to get working with your system, we can advise you about what steps to take.

We'll update the wiki to prevent this from happening to others.

***

**Technical details:**

It looks like Ubuntu is maintaining two versions of the X server: a vanilla `xserver-xorg-core` package and an updated(?) `xserver-xorg-core-hwe-18.04` package. The two servers have incompatible input driver stacks, so there are duplicates of all the input drivers as well -- notably `xserver-xorg-input-wacom` and `xserver-xorg-input-wacom-hwe-18.04`. Your system had the "hwe" server and stack installed.

When asked to install the non-hwe version of the Wacom driver (which was incompatible with the hwe sever you had installed), APT's solution was the following:
* Remove the incompatible hwe server and all of its input drivers
* Install the non-hwe server and the non-hwe Wacom driver

If you aren't paying close attention to the solution that APT proposes and just tell it to proceed, you're left with a system that is in a broken state since there are none of the other input drivers (e.g. for the keyboard or mouse) installed. (Note: apparently even some of the video drivers are removed, since it seems some of them also have incompatible "hwe" and "non-hwe" versions... If somebody's system depended on one of the removed video drivers, they'd be even worse off)

If we update the wiki to request people try installing `xserver-xorg-core-hwe-18.04` instead, this should resolve the problem.

linuxnoob123, 2019-09-11:
Thank you for the solution. Unfortunately I lacked time to wait and just went for a fresh install of Ubuntu. I definitely learned my lesson. 

If this thread is the right place, i would absolutely love to learn how to make my Wacom tablet work. The device is a Wacom Intuos S Bluetooth.

jigpu, 2019-09-11:
According to our [device support list](https://github.com/linuxwacom/input-wacom/wiki/Device-IDs), the Intuos S Bluetooth (CTL-4100WL) requires a Linux 4.17 kernel to work "out of the box". Your (old) sysinfo indicated that you were running Ubuntu 18.04.3, which includes a new enough kernel. If you run `uname -r` you can confirm that your new install also has a kernel version >= 4.17.

If your kernel version is >= 4.17, you should be able to connect the tablet over USB or pair to it over Bluetooth and move the pointer around without any issue. You may not be able to configure the tablet through the control panel, however. It looks like the control panel is relying on an old version of libwacom that we may need to help you update.

Can you describe what happens when you pair the device and try to use it? If you have any problems getting the pointer to move around, please try to pair the device and then re-run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) so that we can see if there are Bluetooth or other errors logged.

linuxnoob123, 2019-09-13:
I tried plugging it in via USB and connecting it via Bluetooth. It seems to behave the same regardless how i connect it.This is how the cursor behaves:
-the cursor is moved even before the pen touches the pad
-the left mouse button is pressed as soon as the pen touches the pen
-using the lower button acts as a left mousebutton click 
-using the upper button acts as a right mousebutton click
Furthermore, the Settings->Devices->Wacom lists says there is no tablet/pen detected.

jigpu, 2019-09-13:
> -the cursor is moved even before the pen touches the pad
> -the left mouse button is pressed as soon as the pen touches the pen
> -using the lower button acts as a left mousebutton click
> -using the upper button acts as a right mousebutton click

This all sounds mostly normal (lower button should send a middle click, but that might just be a typo). Do you want the pen or its buttons to behave differently?

> Furthermore, the Settings->Devices->Wacom lists says there is no tablet/pen detected.

I thought this might be the case. The version of "libwacom" included with Ubuntu 18.04.3 does not have a definition for your particular device. Without a definition, the control panel will not be able to detect the tablet. I would recommend downloading and installing the "libwacom-common" package from Ubuntu 19.04 which will contain the definitions:

1. Go to https://packages.ubuntu.com/disco/libwacom-common
2. Click the "all" link under the "architecture" heading of the table under "Download libwacom-common"
3. Click on one of the mirrors
4. Save the DEB package
5. Run `sudo dpkg -i <path to deb>` to install the package
6. Reboot

linuxnoob123, 2019-09-13:
Thank you for the guide. I followed it expect for when my brain went on autopilot and i just clicked "open with software install" while downloading the package. The install was successfull and I rebooted. After the grub menu showed up i was however greeted with a commandline. I can use my regular credentials to log in. "sudo startx" fails. xinit reports that the connection to X server was lost.
First i tried `sudo apt remove <package-name>` and `sudo dpkg remove <package-name>` which both were unsuccessfull. Then using grep i have tried to find the line in /var/log/dpkg.log that says the name of the package to i need to uninstall to revert this. Unfortunately i was unsuccessfull
At this point i dont know what to do. I could just go ahead to backup my data and once again make a fresh install of ubuntu.

linuxnoob123, 2019-09-14:
Hello. I apprechiate the efforts you have made but at this point i have decided not to use my tablet for now. I can not spend any more time figuring out how to fix my laptop right now, although the tablet would be a nice addition. Anyways, thank you.

closed 2019-09-14 by linuxnoob123

jigpu, 2019-09-19:
Just a note that I've updated the wiki again after getting reports that my previous update caused problems for users who did *not* have the HWE packages installed. The current instructions should be compatible with both HWE and non-HWE Ubuntu versions, as well as other Debian flavors.