linuxwacom/input-wacom #121: Motion Computing R12 lost of touch in kernel 5.3

revolverve, 2019-09-20:
Just ask for more details than the sysinfo provide there: [https://github.com/linuxwacom/wacom-hid-descriptors/issues/63](url)
main error log;
-input input11: Absolute device without dev->absinfo, refusing to register
-wacom: probe of 0003:056A:4004.0001 failed with error -2
-udevd[1073]: '/bin/sh -c 'echo add_0003:056A:4004.0001_hid___hid:b0003g0101v0000056Ap00004004 >> /tmp/udevtrace.log'' [3083] terminated by signal 1 (Hangup)

revolverve, 2019-10-15:
Looking at libwacom tablet database i just realize motion r12 is using both 
 -Wacom ISDv4 4004         	
 and 
 -Wacom ISDv4 EC    
   in database 4004 and EC are both having :
[Features]
Stylus=true
Touch=true
Buttons=0
 
 Did saw some tablets definitions got PairedID=usb:056a:**
 Should it be the case here for motionr12 data definition?

jigpu, 2019-10-17:
Note that a similar issue has been reported at https://bugzilla.redhat.com/show_bug.cgi?id=1754031

revolverve, 2019-10-20:
Checked motion r12 ubuntu-19.10-beta-desktop-amd64(k5.3) , finger touch  get same dev->absinfo error.

revolverve, 2019-10-22:
Test a working: Linux version 5.2.11-lxpup64  (gcc version 9.2.0 (GCC)) #1 SMP
Module wacom: vendor="X.Org Foundation"
 module version = 0.36.1
compiled input-wacom ,replace old wacom.ko wacom_w8001.ko
modprobe -r wacom,moprobe wacom.
   grep "" /sys/module/wacom*/version
v2.00-input-wacom-input-wacom-0.44.0
result,no touch,log:
  input input36: Absolute device without dev->absinfo, refusing to register
wacom: probe of 0003:056A:4004.0002 failed with error -22
 Note that i am not a dev...for what it's worth if it does here.

jigpu, 2019-10-23:
Thanks for the updates. If I'm reading your post correctly, it sounds like you got your pen working with a 5.2 kernel and then installed the input-wacom-0.44.0 driver and got the `dev->absinfo` error. Is that correct?

If so, I'd like you to try installing earlier versions of input-wacom (e.g. 0.43.0, 0.42.0, etc.) until you find the most recent working version. From there we can dig in further to find the exact change that is responsible for the error.

revolverve, 2019-10-23:
Yes jigpu ,in 5.2 all pen ,eraser and touch working untill i install 0.44 making breaking touch only.
Give me time and i will test earlier version.

revolverve, 2019-10-23:
Result for  Linux version 5.2.11-lxpup64
-0.43 no go,touch gone..pen work.
-0.42
 grep "" /sys/module/wacom*/version
v2.00-0.42.0
touch ,pen ok
with ¨make¨ log:
wacom_wac.c:3536:3: warning: this statement may fall through [-Wimplicit-fallthrough=]
   __clear_bit(ABS_MISC, input_dev->absbit);
Here attach sysinfo no xrestart,no reboot,so you get an ¨history¨ of dmesg.log
[sysinfo.e1AuVVnR1W.tar.gz](https://github.com/linuxwacom/input-wacom/files/3764562/sysinfo.e1AuVVnR1W.tar.gz)
Did it the same simple way as before:
./configure
make
replace wacom.ko & wacom_w8001.ko with compiled ones
modprobe -r wacom
modprobe wacom

tested 0.41,0.40 also:
both works[touch,pen,eraser] and gave the same ¨make¨ warning
If it can help somehow!

jigpu, 2019-10-23:
Okay, lets see if we can get an automated bisect to work on your machine... Try to copy/paste all of the following into a terminal and then press Enter:

~~~
# Install the 'git' command
sudo apt-get install git

# Get a copy of the source code repository
git clone https://github.com/linuxwacom/input-wacom.git
cd input-wacom

# Create a script to build and test the revision
cat <<EOF > bisect.sh
#!/usr/bin/env bash
set -e
trap 'exit 255' ERR
./autogen.sh
make clean
make || exit 125
sudo rmmod wacom
sudo dmesg --clear
sudo insmod 4.5/wacom.ko
sudo dmesg | grep absinfo && exit 1 || exit 0
EOF
sudo chmod +x ./bisect.sh

# Run the script
git bisect reset
git bisect start
git bisect bad input-wacom-0.44.0
git bisect good input-wacom-0.43.0
sudo -v
git bisect run ./bisect.sh
~~~

This set of commands should download the source code repository to your machine and then repeatedly build and run the versions between 0.43.0 and 0.44.0 to find out which one caused the bug. When it finishes, the last thing it prints will hopefully be "bisect run success".

When it finishes, run `git bisect log` and copy/paste the output into your reply.

revolverve, 2019-10-24:
jigpu ,
it failed  
[jigpu_bisect_sh.log.tar.gz](https://github.com/linuxwacom/input-wacom/files/3765656/jigpu_bisect_sh.log.tar.gz)

but should it be run from a working kernel or a ¨broken¨ kernel?

That is a useful script by the way..

revolverve, 2019-10-24:
Here kernel 5.3.7 with input-wacom 0.42(newest with touch working)
[input-wacom-0.42_k5.3.7.tar.gz](https://github.com/linuxwacom/input-wacom/files/3765718/input-wacom-0.42_k5.3.7.tar.gz)

jigpu, 2019-10-24:
Sorry for the confusion -- the bisect commands should work on either kernel, but to be safe I would recommend running them with the working 5.2.x kernel.

The failure appears to be due to a difference in how your system's "dmesg" command works. Replacing the line `sudo dmesg --clear` in the script above with `sudo dmesg -c > /dev/null` should allow it to work. You can try again by deleting the input-wacom directory and then re-running the edited set of commands.

revolverve, 2019-10-24:
[git_bisect_log.tar.gz](https://github.com/linuxwacom/input-wacom/files/3767615/git_bisect_log.tar.gz)
it did it,
but,as here it seems like it only work from 0.42.0 should something like:
git bisect bad input-wacom-0.44.0(or 0.43.0)
git bisect good input-wacom-0.42.0
be more relevant?
Hope we dont get over all this for only one problematic hardware or worse..one user!!!
Thanks for your time!

jigpu, 2019-10-24:
Ah, indeed -- I wrote the wrong version into the script. Your log for 0.44 through 0.42 is exactly what I needed, thanks for catching my error and running it as well :)

The responsible commit seems to be 0a844432469a7daf98b06a2d14644740ab88e953 as suspected. I'm continuing to try to understand why exactly this commit would be failing.

~~~
~~~

It sounds like there are several users and at least two different devices affected by this issue.

jigpu, 2019-10-24:
I've pushed up a potential fix to my personal repository. Can you please run the following commands to obtain a copy and then build/install it? (you'll want to remove the previous "input-wacom" directory generated by the bisect script first)
~~~
git clone https://github.com/jigpu/input-wacom.git -b fix-121
~~~

revolverve, 2019-10-25:
jigpu,
all done,
i guess  ,
grep "" /sys/module/wacom*/version
v2.00-0.42.0
0.42.0?? 
if so,It all work fine,pen ,touch,eraser.

revolverve, 2019-10-25:
jigpu, 
May i had details,feeling lucky with the ¨bisect 0.44 through 0.42¨  :)
,running make on your 0.42.0.108.g1de2650 did NOT warn the following as in 0.42.0:
***wacom_wac.c:3536:3: warning: this statement may fall through [-Wimplicit-fallthrough=]
   __clear_bit(ABS_MISC, input_dev->absbit);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
wacom_wac.c:3538:2: note: here
  case WACOM_MO:***
I'll be happy to test any possible fix.

jigpu, 2019-10-25:
It looks like my potential bugfix didn't get loaded/installed correctly. The version should be reported as `v2.00-0.44.0.3.g1de2650`. Could you try recompiling and installing the "fix-121" version?

The "fall through" warning that you mention above is harmless -- we actually _do expect_ the code to fall through at this location. Later versions of the driver fixed the warning by just telling the compiler that fall through is expected.

revolverve, 2019-10-26:
Jason,
you may have to help me in github wonderland,because;
git clone https://github.com/jigpu/input-wacom.git -b fix-121
Cloning into 'input-wacom'
remote: Enumerating objects: 33, done.
remote: Counting objects: 100% (33/33), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 6057 (delta 13), reused 16 (delta 6), pack-reused 6024
Receiving objects: 100% (6057/6057), 3.11 MiB | 262.00 KiB/s, done.
Resolving deltas: 100% (4981/4981), done.
=v2.00-0.42.0.108.g1de2650
tried ,thought ¨doable¨ downloading zip of fix-121 branche
=v2.00-input-wacom-fix-121
either of those get  the ¨refusing to register¨
or there is a  missing detail somewhere!!!!
- back  0.42.0 release...it work..

 oh! the make warning advice! good !Thanks

jigpu, 2019-10-28:
The kernel version printed out looks a little strange, but the commit ID is correct and working directly from the ZIP should work the same. I'm going to assume that you were able to get my fix compiled and loaded, but that it didn't actually fix the problem.

I've pushed up another potential fix. If you still have the input-wacom directory around, run `git pull` to get the update and then compile/install it. The version number printed out by `grep "" /sys/module/wacom/version` for this new update should be `v2.00-0.44.0.4.g30fb85f` (though the "g30fb85f" at the end is really the most important part if yours is different).

fabienbaron, 2019-10-28:
v2.00-0.44.0.4.g30fb85f still produces the "failed to register" message for me with 5.3.

jigpu, 2019-10-28:
Thanks -- I'll see what else can be done...

revolverve, 2019-10-28:
**jigpu** ,
Same  ¨refusing to register¨, assuming g30fb85f is good enough !?! 

 **fabien**,  
Does 0.42.0 work for you?

revolverve, 2019-10-30:
fabien
> 0.42 doesn't throw the -22 error

Here if no -22 error it work..
Just to show my ignorance on the subject,I taught wacom.ko version had to be compiled against kernel used,not at all here ,now using 0.42 compiled in 5.3.7 with 5.3.8..
  ..Just some ephemeral knowledge in this age of quantum physics.. :)

xax, 2019-11-13:
Just to confirm the issue:

The *Wacom ISDv4 101* (xinput), `ID 056a:0101` (lsusb) pen/eraser/finger device on a *Fujitsu Lifebook T902* worked well using *Linux Mint 19.2* with kernels up to and including 5.0; starting with kernel 5.3 finger touch does not work and `dmesg` shows
```
[…]  input input128: Absolute device without dev->absinfo, refusing to register
[…]  wacom: probe of 0003:056A:0101.0003 failed with error -22
```
Bisecting from tags `input-wacom-0.42.0` to `input-wacom-0.44.0` of this repository (against kernel 5.3) indeed leads to commit 0a84443 being designated *first bad commit*, thus commit cd47de4 being *last good commit*.

Latest master (b17704d as of now) yields error mentioned above.

revolverve, 2019-11-13:
Trying to find relation between 4004 and 5001 i 
 found this from 2014 ; https://patchwork.kernel.org/patch/4145891/
just sharing here..
Thumb up to devs here to over the years keep those ¨proprietary¨ old devices supported while adding support for new one!

fabienbaron, 2019-11-18:
On Ubuntu the fix doesn't work for me pre-5.3.

```baron@voyager:~$ uname -a
Linux voyager 5.0.0-36-generic #39-Ubuntu SMP Tue Nov 12 09:46:06 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
baron@voyager:~$ grep "" /sys/module/wacom/version
v2.00-0.42.0.109.g30fb85f
baron@voyager:~$ dmesg | grep wacom
[    2.918943] wacom: loading out-of-tree module taints kernel.
[    2.919015] wacom: module verification failed: signature and/or required key missing - tainting kernel
[    2.974756] wacom: probe of 0003:056A:5002.0001 failed with error -22
[    3.030344] wacom 0003:056A:5002.0002: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       MultiTouch Sensor  ] on usb-0000:00:14.0-4/input1
```

revolverve, 2019-11-19:
Fabien,
Only 0.42 ¨release¨ work here ;
https://github.com/linuxwacom/input-wacom/releases/download/input-wacom-0.42.0/input-wacom-0.42.0.tar.bz2.
If there could be confusion  as it seems your test with the fix appear like mine..
v2.00-0.42.0.109.g30fb85f...fix attempt,not working
v2.00-0.42.0  grep the release version should be.. working here in k5.3
Did saw wacom changelog in 5.3.11 but not for fixing our problem here that  does not seems to affect  too many hardware...

bionade24, 2019-12-15:
I have the bug on Arch Kernel 5.3.11, after downgrading to 5.2.1 everything works again:
I use `libinput` and `libwacom` instead of `xinput`.  Also on a T904, like mentioned in #125 .
```
[ 2101.583306] input: Wacom Co.,Ltd.       MultiTouch Sensor   Touchscreen as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:056A:5002.0001/input/input45
[ 2101.639265] hid-generic 0003:056A:5002.0001: input,hiddev0,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       MultiTouch Sensor  ] on usb-0000:00:14.0-4/input0
[ 2101.639504] input: Wacom Co.,Ltd.       MultiTouch Sensor   as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:056A:5002.0002/input/input47
[ 2101.639667] hid-generic 0003:056A:5002.0002: input,hiddev1,hidraw1: USB HID v1.11 Device [Wacom Co.,Ltd.       MultiTouch Sensor  ] on usb-0000:00:14.0-4/input1
[ 2103.353801] input input50: Absolute device without dev->absinfo, refusing to register
[ 2103.353823] wacom: probe of 0003:056A:5002.0001 failed with error -22
[ 2103.416194] input: Wacom ISDv4 5002 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:056A:5002.0002/input/input52
[ 2103.472552] wacom 0003:056A:5002.0002: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       MultiTouch Sensor  ] on usb-0000:00:14.0-4/input1
```

bionade24, 2020-02-15:
Is this bug fixed in kernel 5.5?

revolverve, 2020-02-19:
kern.notice kernel: Linux version 5.5.4-lxpup64 
kern.err kernel: input input11: Absolute device without dev->absinfo, refusing to register
kern.warn kernel: wacom: probe of 0003:056A:4004.0001 failed with error -22
....not here....

jonduncan05, 2020-04-19:
Running clean Debian 10 or Live Ubuntu 19.10, different kernel (v4 vs. v5.3 respectively), both behave the same on Lifebook T with the :5002 screen.  I haven't seen anyone mention... on mine, the stylus seems to work fine until I touch the screen with a finger and then neither work.  Which I find interesting because on first failing commit 0a84443 it looks like the changes were in "wacom_wac_finger_event"

jigpu, 2020-04-20:
@jonduncan05 Thanks for posting that edit -- I was about to ask if the latest master had been tested yet. The fix for #155 solved the exact same error message for a similar sensor. Could others please test to see if the master branch works for them?

fabienbaron, 2020-04-20:
I can confirm touch works again on Arch Linux with 5.6.5-arch3-1 kernel. Will test pen now.

jasch0825, 2020-04-21:
I can confirm touch an pen works fine on 5.3.0-46-generic #38~18.04.1-Ubuntu SMP with the last master.

Thanks to everyone.

bionade24, 2020-04-21:
I'll have a question: I only have libwacom installed, not input-wacom. Why does it not work for me then? What is diffrent on my setup?

jigpu, 2020-04-21:
@bionade24 The bug is in our kernel driver. The "input-wacom" package is an up-to-date copy of the kernel driver which can be installed by users with older kernels. In this particular case even the most recent released kernels (and even the most recent input-wacom release!) still have the issue, so the only way to get the fix is to install the "master" development branch of input-wacom as described above. Give it a shot :)

revolverve, 2020-04-21:
Yes, it did it on the R12 
grep "" /sys/module/wacom*/version
v2.00-input-wacom-master
 So ,if its in master it make it in the kernel? eventually?

jigpu, 2020-04-21:
@revolverve Yes, the fix has been accepted by the kernel's HID subsystem maintainer and just needs to be pulled by Linus into the kernel source tree. The fix has also been sent to the Linux stable mailing list. Unfortunately it will be several weeks before the fix makes its way onto people's computers. It will get there eventually, however.

I'll update [this post](https://github.com/linuxwacom/input-wacom/issues/155#issuecomment-608565932) with specific kernel versions that have the fix once I know what they are.

revolverve, 2020-05-16:
Latest kernel 5.6.13 integrated the fix for this issue ,it now behave like pre-5.3 kernel. 
For  motion R12 r001 its fixed.
Thanks all devs for your linux/wacom support.

closed 2020-05-19 by jigpu