linuxwacom/input-wacom #122: Build error

TechGamer21, 2019-09-22:
For the life of me I can't figure this one out, I had read on another post that the OP had restarted their computer, unfortunately that didn't work for me.

```
[anthony@anthony-pc input-wacom-0.44.0]$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... not found
configure: WARNING: Unable to find build config in any of:
	  [/lib/modules/5.2.11-1-MANJARO/source],
	  [/lib/modules/5.2.11-1-MANJARO/build],
	  [/usr/src/linux],
	  [/usr/src/linux-5.2.11-1-MANJARO],
	  [/usr/src/linux-2.6]
	

configure: WARNING: Kernel directory does not appear to have needed config files
checking kernel version... 
configure: WARNING: Spcified '5.2.11-1-MANJARO' kernel, but found '' instead
configure: error: We could not find the development environment to build modules for the '5.2.11-1-MANJARO' kernel within the '' directory. Please install the kernel source or the kernel development package and try again.
Build Failed
[anthony@anthony-pc input-wacom-0.44.0]$
```

jigpu, 2019-09-23:
This usually means the kernel headers have not been installed or the running kernel has a different version than the headers. Since it looks like you're running the Linux 5.2 kernel, can you double-check that the "linux52-headers" package is installed? If its not, please install it and try again.

TechGamer21, 2019-09-24:
> This usually means the kernel headers have not been installed or the running kernel has a different version than the headers. Since it looks like you're running the Linux 5.2 kernel, can you double-check that the "linux52-headers" package is installed? If its not, please install it and try again.

Thank you so much, can't believe that slipped my mind

closed 2019-09-24 by TechGamer21