linuxwacom/input-wacom #123: Build Failed on Kali Linux 5.2.0-kali2-amd64

johhnry, 2019-09-22:
Hi,

I am on Kali Linux (`SMP Debian 5.2.9-2kali1 (2019-08-22) x86_64 GNU/Linux`) and I am trying to build the `input-wacom-0.44.0` for my Bamboo Wacom.

This is what I am getting : 

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... not found
configure: WARNING: Unable to find build config in any of:
	  [/lib/modules/5.2.0-kali2-amd64/source],
	  [/lib/modules/5.2.0-kali2-amd64/build],
	  [/usr/src/linux],
	  [/usr/src/linux-5.2.0-kali2-amd64],
	  [/usr/src/linux-2.6]
	

configure: WARNING: Kernel directory does not appear to have needed config files
checking kernel version... 
configure: WARNING: Spcified '5.2.0-kali2-amd64' kernel, but found '' instead
configure: error: We could not find the development environment to build modules for the '5.2.0-kali2-amd64' kernel within the '' directory. Please install the kernel source or the kernel development package and try again.
Build Failed
```

Thanks in advance

jigpu, 2019-09-23:
Can you provide the model number for your tablet? I don't believe input-wacom-0.44.0 supports any more tablets than the 5.2 kernel already does. Its possible that you're running into some other problem that installing the input-wacom driver will not fix.

johhnry, 2019-09-23:
Hi,

Thanks for answering my issue.
The model number of my tablet is : `CTH-460`.
For example if I want to disable the tactile pressure / change left or right handed, is it possible on Linux with this kind of tablet?
Plus, I am working on Blender so is the software/os able to interpret the pen pressure?

Thanks again

jigpu, 2019-09-23:
The CTH-460 should be supported by your current Linux kernel. Installing this driver (or others) should not be necessary. You should be able switch between left- and right-handed use of the tablet through the GNOME Control Center (if you are using the GNOME desktop). I'm not quite sure what you mean by "disable the tactile pressure", but you can change other options like the pressure threshold through the command-line `xsetwacom` tool. You should read `man xsetwacom` to get an idea of what is possible.

Blender should support pen pressure for a few operations, e.g. sculpting.

closed 2019-09-23 by jigpu