linuxwacom/input-wacom #125: Wacom ISDv4 5002 issue (Fujitsu T904)

fabienbaron, 2019-10-02:
As of an update last week, the wacom touchscreen on Fujitsu laptop stopped working on Arch Linux. I've also tested this laptop with live USB sessions on Ubuntu 19.04 (works) vs 19.10 beta (doesn't).
dmesg output shows (same under arch and ubuntu 19.10):
[   11.268102] input input28: Absolute device without dev->absinfo, refusing to register
[   11.268128] wacom: probe of 0003:056A:5002.0001 failed with error -22
[   11.310324] input: Wacom ISDv4 5002 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:056A:5002.0002/input/input30
[   11.310488] wacom 0003:056A:5002.0002: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       MultiTouch Sensor  ] on usb-0000:00:14.0-4/i
nput1
The stylus seems detected but actually doesn't work either.

jasch0825, 2019-10-19:
I had the same issue after an update from Ubuntu 19.04 to 19.10. 
I did the following steps to solve the issue (just a workaround):
- update the Kernel from 5.3.0-18-generic to 5.3.7-050307-generic (doesn't solve the issue)
- downgrade the Kernel to 5.0.0-32-generic (solved the issue)

It is just a workaround for my Fujitsu T904

frumble, 2019-10-21:
I can confirm this issue on Manjaro with the recent October updates.

jigpu, 2019-10-23:
Likely the same issue as https://github.com/linuxwacom/input-wacom/issues/121. If anyone can run a pre-5.3 kernel (which should work) and then install older versions of the input-wacom driver to determine the last working version, we can then work on narrowing down the fault to the specific change.

jigpu, 2019-10-28:
Closing as a duplicate of #121 based on feedback from Fedora bug tracker.

closed 2019-10-28 by jigpu