linuxwacom/input-wacom #128: Unable to install input-wacom-0.44.0

SMarshall1971, 2019-10-13:
Hi, I'm a non-technical person trying to install a Wacom One tablet on Linux Mint 18.3.  I've followed the instructions on https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source as far as the step marked "Build/Install".  It tells me to copy/paste a line of text: 

$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"

It outputs:

bash: syntax error near unexpected token `then'
stuart@stuart-LinuxMint /usr/src/input-wacom-0.44.0 $

Please could someone advise?  As I said, I'm a non-technical person and I really benefit from instructions broken down into quite simple steps... thanks!

skomra, 2019-10-14:
Hi,

I don't know why, but sometimes we see users reporting that 'then' fails. Please try the directions below.

---------------------------------------

From within the extracted input-wacom directory, run the following commands to build the driver and install it where your system can find it.

Note that if the ./configure script does not exist (as will be the case with code from git or pre-release versions), you will need to run ./autogen.sh instead.

```
./configure
make
sudo make install 
```

jigpu, 2019-10-14:
The instructions above should get you sorted out. As for the the `unexpected token `then'` issue, it looks like that can be caused if you accidentally paste the leading `$` into your terminal. The `$` is a (typical but perhaps-confusing) way to indicate that the command needs to be run as a normal user. There might be a better way for us to document it.

rishabhmw, 2019-10-26:
@jigpu is right. Fixed the issue for me I was accidentally pasting the $ symbol.

jigpu, 2019-11-14:
I have removed the leading '$' from the various commands in our wiki's build instructions.

closed 2019-11-14 by jigpu