linuxwacom/input-wacom #129: Build fails if spaces are in path

jigpu, 2019-10-14:
~~~
make  all-recursive
make[1]: Entering directory '/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0'
Making all in 4.5
make[2]: Entering directory '/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/4.15.0-65-generic/build M=/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0/4.5
make[3]: Entering directory '/usr/src/linux-headers-4.15.0-65-generic'
make[3]: *** No rule to make target 'and'.  Stop.
make[3]: Leaving directory '/usr/src/linux-headers-4.15.0-65-generic'
Makefile:39: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0/4.5'
Makefile:377: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0'
Makefile:308: recipe for target 'all' failed
make: *** [all] Error 2
Build Failed
~~~

_Originally posted by @changeling420 in https://github.com/linuxwacom/xf86-input-wacom/issues/87#issuecomment-541775057_

jigpu, 2019-10-14:
@changeling420, I've given fixing this a shot but it looks like the build system for the Linux kernel (which we piggyback on) does not itself support having spaces in the path. The only way I can "fix" this at the moment is to produce an obvious error if a space is found.

Since we won't be able to build from paths that have spaces, please either rename the directories that have spaces or move your input-wacom directory somewhere else.

changeling420, 2019-10-15:
> @changeling420, I've given fixing this a shot but it looks like the build system for the Linux kernel (which we piggyback on) does not itself support having spaces in the path. The only way I can [...]

thanks jigpu, it worked fine

closed 2019-10-28 by jigpu