linuxwacom/input-wacom #132: [Regression] Using stylus on Gnome Shell 3.34 crashes applications

luyatshimbalanga, 2019-10-23:
As tested on HP Envy x360 15 Ryzen 2500u.
A regression occurred with the stylus input causing traceback on opened applications (for example Firefox and Gnome Settings) in Gnome Shell 3.34.1 Wayland running on Fedora 31. The touch screen input is unaffected.

jigpu, 2019-10-23:
You say that this is a regression, so could you let me know which component(s) you upgraded? Or did you just upgrade the whole distro to Fedora 31? Is it possible for you to roll back the upgrade so that we can try to isolate the faulty component?

This sounds similar to something that @bentiss was reporting. I wonder if he has any insight?

luyatshimbalanga, 2019-10-24:
It is a distro upgrade from Fedora 30 where the stylus worked as intended to Fedora 31.

luyatshimbalanga, 2019-10-26:
I run an updated Fedora 30 (Gnome 3.32) live media which includes the tablet file for the HP Envy x360 Ryzen 2500u. Using the stylus to move the applications, writing and editing is smooth. The issue occurred after upgrading to Fedora 31 (Gnome 3.34) leading to a segmentation fault. 
It seems the problem is specifically related to the input for stylus hence input-wacom.

luyatshimbalanga, 2019-10-27:
I think I figured out the issue. Gnome Shell 3.34 used on Fedora 31 running on Wayland session by default only use XWayland (Xorg) on demand depending of the application. The problem does not occur on Xorg session.

luyatshimbalanga, 2019-10-28:
I am closing this report as the issue is related to GTK. See the [GTK bug report](https://gitlab.gnome.org/GNOME/gtk/issues/2157)

closed 2019-10-28 by luyatshimbalanga