linuxwacom/input-wacom #133: Recompiling the driver after Kernel Updates

drpjkurian, 2019-10-28:
Dear developers
I woud like to suggest you to add the point of recompiling the drivers after Kernel updates in instruction manual.
It took a while for me to figure out the cause of my wacom failing after regular software updates. My machines are running on Lubuntu 18.04 and Lubuntu 16.04

With regards
Dr Kurian P J

skomra, 2019-10-28:
I added the following to the "Next steps" section. Let me know if that's not what you had in mind.

"If in the future your kernel is updated by your distribution, you will need to re-install the driver for that kernel version. To do so, rerun <code>./autogen.sh</code> or <code>./configure</code> and then follow the original installation instructions."

drpjkurian, 2019-10-29:
>  I added the following to the "Next steps" section. Let me know if that's not what you had in mind.
> 
> "If in the future your kernel is updated by your distribution, you will need to re-install the driver for that kernel version. To do so, rerun `./autogen.sh` or `./configure` and then follow the original installation instructions."

Thank you Skomra for your rapid response

closed 2019-10-29 by drpjkurian