linuxwacom/input-wacom #135: Stylus Not Recognized

wwderw, 2019-11-02:
Recently updated to Pop OS 19.10 with the 5.3x kernel.

Maybe related to #104, maybe not.  I still have to use the unbind/bind script after reboot for the stylus to even recognize, so I'm thinking it has something to do with a kernel issue and the driver.  Originally I was thinking it had to do with Plasma, but I'm using the relatively new Gnome DE here.

However, unlike 19.04 and older, after I do the unbind/bind script, `xsetwacom --list` will show the stylus then.  However, the GUI frontend to the Wacom Tablet will not recognize the stylus.

This does make it hard to use the stylus in VMs (on the host, its find using Bash scripts to assume everything, which is what I'm used to as the Gnome frontend doesn't have keybindings for the 27qhd for the Express Remote, I've always been using Bash Scripts in that regard) for programs that would benefit from stylus usage.

For instance, when I right click, the resulting menu is a few hundred px away, yet on the host, it's right next to each other.  While the mouse in the guest tracks, where it clicks doesn't seem to.

I don't know if this is an underlying driver issue or perhaps linked to the GUI GTK frontend (don't know if y'all handle that as well or if that's through say Gnome itself).

Thoughts? Ideas?

jigpu, 2019-11-06:
I'm not quite sure, but it sounds like you're describing two problems. The first is that GNOME's Wacom Tablet settings does not recognize the stylus after running the unbind/bind script. The second is that clicks do not appear in the correct location inside of your VMs. Is that accurate?

The second issue sounds *very* similar to a problem that we've been seeing with a few Qt-based applications. You wouldn't be using Virtualbox, would you? See https://github.com/linuxwacom/xf86-input-wacom/issues/93 for more information...

As for the first issue, is the stylus recognized after performing a logout and login?

wwderw, 2019-11-06:
> The first is that GNOME's Wacom Tablet settings does not recognize the stylus after running the unbind/bind script.

This actually happens with KDE Wacom GUI frontend as well.  It too, does not recognize the stylus after the unbind/bind script.

> You wouldn't be using Virtualbox, would you? See linuxwacom/xf86-input-wacom#93 for more information...

Yes, I am running Virtualbox and that does indeed appear to solve that issue.  I just ran it for a little bit though, just for a test.  But it appears to be working.  Time for another Bash script (I won't ever remember that command).

> As for the first issue, is the stylus recognized after performing a logout and login?

No, I actually have to run the script again after logging out and then back in.

Pinglinux, 2022-08-05:
So, the script is your weapon ;)

closed 2022-08-05 by Pinglinux