linuxwacom/input-wacom #136: Failed to install Wacom driver in Amazon Linux 2 system

skybear2014, 2019-11-07:
hi, it seemed that Wacom driver was not installed in Amazon Linux 2 system. And the installation failed for me. 
Reproduce steps.

1. Deploy EC2 instance(size: m4.large) based on AMI amzn2-ami-hvm-2.0.20191024.3-x86_64-gp2 (ami-0a85857bfc5345c38)
2. Enable the EPEL repository in the instance
3. Install wacom driver, and get the following error message

> configure:
> ----------------------------------------
>   BUILD ENVIRONMENT:
>        linux kernel - yes 4.5
>       kernel source - yes /lib/modules/4.14.146-120.181.amzn2.x86_64/source
> 
> NOTE: The kernel drivers included in this package are only tested with the
> X Wacom driver built from xf86-input-wacom. If you are running an X server
> version older than 1.7, please use the drivers provided by the linuxwacom
> package.
> 
> Please run 'make && make install'.
> make  all-recursive
> make[1]: Entering directory `/home/ec2-user/input-wacom-0.44.0'
> Making all in 4.5
> make[2]: Entering directory `/home/ec2-user/input-wacom-0.44.0/4.5'
>     Building input-wacom drivers for 4.5 kernel.
> make -C /lib/modules/4.14.146-120.181.amzn2.x86_64/source M=/home/ec2-user/input-wacom-0.44.0/4.5
> make[3]: Entering directory `/usr/src/kernels/4.14.146-120.181.amzn2.x86_64'
>   AR      /home/ec2-user/input-wacom-0.44.0/4.5/built-in.o
>   CC [M]  /home/ec2-user/input-wacom-0.44.0/4.5/wacom_wac.o
>   CC [M]  /home/ec2-user/input-wacom-0.44.0/4.5/wacom_sys.o
> /home/ec2-user/input-wacom-0.44.0/4.5/wacom_sys.c: In function ‘wacom_led_register_one’:
> /home/ec2-user/input-wacom-0.44.0/4.5/wacom_sys.c:1385:15: error: ‘struct led_trigger’ has no member named ‘name’
>    led->trigger.name = name;
>                ^
> /home/ec2-user/input-wacom-0.44.0/4.5/wacom_sys.c:1386:11: error: implicit declaration of function ‘devm_led_trigger_register’; did you mean ‘devm_led_classdev_register’? [-Werror=implicit-function-declaration]
>    error = devm_led_trigger_register(dev, &led->trigger);
>            ^~~~~~~~~~~~~~~~~~~~~~~~~
>            devm_led_classdev_register
> /home/ec2-user/input-wacom-0.44.0/4.5/wacom_sys.c: In function ‘wacom_led_next’:
> /home/ec2-user/input-wacom-0.44.0/4.5/wacom_sys.c:1541:25: error: ‘struct led_classdev’ has no member named ‘trigger’
>   } while (next_led->cdev.trigger != &next_led->trigger);
>                          ^
> cc1: some warnings being treated as errors
> make[4]: *** [/home/ec2-user/input-wacom-0.44.0/4.5/wacom_sys.o] Error 1
> make[3]: *** [_module_/home/ec2-user/input-wacom-0.44.0/4.5] Error 2
> make[3]: Leaving directory `/usr/src/kernels/4.14.146-120.181.amzn2.x86_64'
> make[2]: *** [all] Error 2
> make[2]: Leaving directory `/home/ec2-user/input-wacom-0.44.0/4.5'
> make[1]: *** [all-recursive] Error 1
> make[1]: Leaving directory `/home/ec2-user/input-wacom-0.44.0'
> make: *** [all] Error 2
> Build Failed

jigpu, 2019-11-07:
It looks like the kernel for that Amazon instance might not have `CONFIG_LEDS_TRIGGERS` enabled. Please test out my development branch by running `git clone https://github.com/jigpu/input-wacom.git -b fix-136` to get a copy of the fixed code and compile it as normal.

skybear2014, 2019-11-08:
Thank a lot! I got a different error now.

> configure:
> ----------------------------------------
>   BUILD ENVIRONMENT:
>        linux kernel - yes 4.5
>       kernel source - yes /lib/modules/4.14.146-120.181.amzn2.x86_64/source
> 
> NOTE: The kernel drivers included in this package are only tested with the
> X Wacom driver built from xf86-input-wacom. If you are running an X server
> version older than 1.7, please use the drivers provided by the linuxwacom
> package.
> 
> Please run 'make && make install'.
> make  all-recursive
> make[1]: Entering directory `/home/ec2-user/input-wacom'
> Making all in 4.5
> make[2]: Entering directory `/home/ec2-user/input-wacom/4.5'
>     Building input-wacom drivers for 4.5 kernel.
> make -C /lib/modules/4.14.146-120.181.amzn2.x86_64/source M=/home/ec2-user/input-wacom/4.5
> make[3]: Entering directory `/usr/src/kernels/4.14.146-120.181.amzn2.x86_64'
>   AR      /home/ec2-user/input-wacom/4.5/built-in.o
>   CC [M]  /home/ec2-user/input-wacom/4.5/wacom_wac.o
>   CC [M]  /home/ec2-user/input-wacom/4.5/wacom_sys.o
> /home/ec2-user/input-wacom/4.5/wacom_sys.c: In function ‘wacom_led_next’:
> /home/ec2-user/input-wacom/4.5/wacom_sys.c:1546:25: error: ‘struct led_classdev’ has no member named ‘trigger’
>   } while (next_led->cdev.trigger != &next_led->trigger);
>                          ^
> At top level:
> /home/ec2-user/input-wacom/4.5/wacom_sys.c:1364:13: warning: ‘wacom_led_readonly_brightness_set’ defined but not used [-Wunused-function]
>  static void wacom_led_readonly_brightness_set(struct led_classdev *cdev,
>              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /home/ec2-user/input-wacom/4.5/wacom_sys.c:1336:12: warning: ‘wacom_led_brightness_set’ defined but not used [-Wunused-function]
>  static int wacom_led_brightness_set(struct led_classdev *cdev,
>             ^~~~~~~~~~~~~~~~~~~~~~~~
> /home/ec2-user/input-wacom/4.5/wacom_sys.c:1325:28: warning: ‘__wacom_led_brightness_get’ defined but not used [-Wunused-function]
>  static enum led_brightness __wacom_led_brightness_get(struct led_classdev *cdev)
>                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
> make[4]: *** [/home/ec2-user/input-wacom/4.5/wacom_sys.o] Error 1
> make[3]: *** [_module_/home/ec2-user/input-wacom/4.5] Error 2
> make[3]: Leaving directory `/usr/src/kernels/4.14.146-120.181.amzn2.x86_64'
> make[2]: *** [all] Error 2
> make[2]: Leaving directory `/home/ec2-user/input-wacom/4.5'
> make[1]: *** [all-recursive] Error 1
> make[1]: Leaving directory `/home/ec2-user/input-wacom'
> make: *** [all] Error 2

jigpu, 2019-11-08:
Okay, I've pushed up an update that should fix the new error. You should be able to run `git pull` inside of the input-wacom directory and then attempt to build again. (Alternatively, you can just delete the directory and re-clone as per my previous instructions).

I don't think these patches are going to be accepted upstream, BTW. The kernel has `CONFIG_LEDS_TRIGGERS` marked as a dependency of the Wacom driver, and its only because input-wacom doesn't have access to the dependency list that it even attempts to build...

skybear2014, 2019-11-08:
Thanks for the prompt reply. The installation completed with some warning message, as shown below.  Thanks again for your help. 

> make[3]: Entering directory `/usr/src/kernels/4.14.146-120.181.amzn2.x86_64'

  CC [M]  /home/ec2-user/input-wacom/4.5/wacom_wac.o
  CC [M]  /home/ec2-user/input-wacom/4.5/wacom_sys.o
/home/ec2-user/input-wacom/4.5/wacom_sys.c:1364:13: warning: ‘wacom_led_readonly_brightness_set’ defined but not used [-Wunused-function]
 static void wacom_led_readonly_brightness_set(struct led_classdev *cdev,
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ec2-user/input-wacom/4.5/wacom_sys.c:1336:12: warning: ‘wacom_led_brightness_set’ defined but not used [-Wunused-function]
 static int wacom_led_brightness_set(struct led_classdev *cdev,
            ^~~~~~~~~~~~~~~~~~~~~~~~
/home/ec2-user/input-wacom/4.5/wacom_sys.c:1325:28: warning: ‘__wacom_led_brightness_get’ defined but not used [-Wunused-function]
 static enum led_brightness __wacom_led_brightness_get(struct led_classdev *cdev)
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~
  LD [M]  /home/ec2-user/input-wacom/4.5/wacom.o
  CC [M]  /home/ec2-user/input-wacom/4.5/wacom_w8001.o
  Building modules, stage 2.
  MODPOST 2 modules
  CC      /home/ec2-user/input-wacom/4.5/wacom.mod.o
  LD [M]  /home/ec2-user/input-wacom/4.5/wacom.ko
  CC      /home/ec2-user/input-wacom/4.5/wacom_w8001.mod.o
  LD [M]  /home/ec2-user/input-wacom/4.5/wacom_w8001.ko
make[3]: Leaving directory `/usr/src/kernels/4.14.146-120.181.amzn2.x86_64'
make[2]: Leaving directory `/home/ec2-user/input-wacom/4.5'
make[2]: Entering directory `/home/ec2-user/input-wacom'
make[2]: Leaving directory `/home/ec2-user/input-wacom'
make[1]: Leaving directory `/home/ec2-user/input-wacom'
Making install in 4.5
make[1]: Entering directory `/home/ec2-user/input-wacom/4.5'
make -C /lib/modules/4.14.146-120.181.amzn2.x86_64/source M=/home/ec2-user/input-wacom/4.5 modules_install mod_sign_cmd=''
make[2]: Entering directory `/usr/src/kernels/4.14.146-120.181.amzn2.x86_64'
  INSTALL /home/ec2-user/input-wacom/4.5/wacom.ko
/bin/sh: /lib/modules/4.14.146-120.181.amzn2.x86_64/extra/wacom.ko: Permission denied
  INSTALL /home/ec2-user/input-wacom/4.5/wacom_w8001.ko
/bin/sh: /lib/modules/4.14.146-120.181.amzn2.x86_64/extra/wacom_w8001.ko: Permission denied
  DEPMOD  4.14.146-120.181.amzn2.x86_64
make[2]: Leaving directory `/usr/src/kernels/4.14.146-120.181.amzn2.x86_64'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
PATH="/sbin:/bin:/usr/sbin:/usr/bin:/bin:/sbin" depmod -a 4.14.146-120.181.amzn2.x86_64
make[1]: Leaving directory `/home/ec2-user/input-wacom/4.5'
make[1]: Entering directory `/home/ec2-user/input-wacom'
make[2]: Entering directory `/home/ec2-user/input-wacom'
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
make[2]: Leaving directory `/home/ec2-user/input-wacom'
make[1]: Leaving directory `/home/ec2-user/input-wacom'

closed 2019-11-08 by skybear2014