linuxwacom/input-wacom #138: Kernel not recognising the device??

drpjkurian, 2019-11-19:
I was using 'One' by Wacom in Lubuntu 16.04. Recently the device stopped working. I presumed that regular Kernel updates killed my driver. Hence I reinstalled it using the instructions from the website. But that attempt to revive my wacom failed. I reinstalled the X driver and rebooted the machine, that too failed. 
I tried the troubleshooting commands namely
`sudo rmmod wacom
sudo rmmod wacom_w8001
sudo modprobe wacom
sudo modprobe wacom_w8001
` 
All the commands except the last one gave an error message.
A peculiar process noted is this which might throw light on the nature of the bug.
The light on the device is glowing and the command 
`lsusb` gives the following output

`Bus 002 Device 003: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 002 Device 005: ID 04ca:0061 Lite-On Technology Corp. 
Bus 002 Device 004: ID 056a:037a Wacom Co., Ltd 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0ac8:c412 Z-Star Microelectronics Corp. 
Bus 001 Device 006: ID 0a5c:4503 Broadcom Corp. Mouse (Boot Interface Subclass)
Bus 001 Device 005: ID 0a5c:4502 Broadcom Corp. Keyboard (Boot Interface Subclass)
Bus 001 Device 003: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
` 

But the command

`grep "" /sys/module/wacom*/version
` 
gives the following output
`grep: /sys/module/wacom*/version: No such file or directory
`
Interestingly the  command 
`modinfo wacom | grep version`
gives the following output as if the device is not connected or recognized
`version:        v2.00-0.44.0
srcversion:     158A607B42851BD4CBB5A83
`
Kindly help to resolve this issue. Please let me know if any additional information are required.

With regards
Dr Kurian P J

skomra, 2019-11-19:
Have you experienced any system crashes? Or has the device simply stopped working?

Please run the sysinfo script and post the results here:

https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh

drpjkurian, 2019-11-20:
Dear Skomra
There were no system crashes.
I figured out a way to use my Wacom. I installed version 43 and Voila, It worked.  I have both the versions namely 43 and 44 in my machine.
I do not know how whether version 43 installed along with version 44 may conceal the information you are trying to seek. Hence please let me know how to purge the unwanted version if required.

With regards
Dr Kurian
[sysinfo.N2n7o1vnRQ.tar.gz](https://github.com/linuxwacom/input-wacom/files/3870524/sysinfo.N2n7o1vnRQ.tar.gz)

skomra, 2019-11-20:
Hi Dr Kurian,

Thanks for reporting this. I have reproduced the issue with the 0.44 driver on the 4.15 kernel with the following relevant dmesg log:

[    1.437815] input input2: Absolute device without dev->absinfo, refusing to register
[    1.438370] wacom: probe of 0003:056A:037B.0001 failed with error -22
[    1.438397] wacom 0003:056A:037B.0002: Unknown device_type for 'Wacom Co.,Ltd. CTL-672'. Ignoring.

No need to modify your system if it is currently working for you.

Best,
Aaron

bionade24, 2019-11-23:
I have a similar bug on Arch Kernel 5.3.11, after downgrading to 5.2.1 everything works again:
```
[ 2101.583306] input: Wacom Co.,Ltd.       MultiTouch Sensor   Touchscreen as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:056A:5002.0001/input/input45
[ 2101.639265] hid-generic 0003:056A:5002.0001: input,hiddev0,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       MultiTouch Sensor  ] on usb-0000:00:14.0-4/input0
[ 2101.639504] input: Wacom Co.,Ltd.       MultiTouch Sensor   as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:056A:5002.0002/input/input47
[ 2101.639667] hid-generic 0003:056A:5002.0002: input,hiddev1,hidraw1: USB HID v1.11 Device [Wacom Co.,Ltd.       MultiTouch Sensor  ] on usb-0000:00:14.0-4/input1
[ 2103.353801] input input50: Absolute device without dev->absinfo, refusing to register
[ 2103.353823] wacom: probe of 0003:056A:5002.0001 failed with error -22
[ 2103.416194] input: Wacom ISDv4 5002 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:056A:5002.0002/input/input52
[ 2103.472552] wacom 0003:056A:5002.0002: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       MultiTouch Sensor  ] on usb-0000:00:14.0-4/input1
```
Is this the same bug?

jigpu, 2019-12-12:
@bionade24 -- others are reporting `Absolute device without dev->absinfo, refusing to register` errors for various tablet PC sensors (including your 056a:5002) in #121. It isn't clear at this time whether this is the same issue or not.

bionade24, 2019-12-12:
@jigpu Seems like you're right, should I simply delete or move my log? I use `libinput` instead of `xinput` like in the mentioned thread, but this shouldn't make any differences, should it?

jigpu, 2019-12-12:
If you'd like to copy the information over to the other issue that would be great. Its entirely possible that these two issues _are_ the same, but since your particular sensor has already been mentioned over there, it would help me to have all the information about it in one location :)

karlkappe, 2020-04-20:
I have a similar issue. I am trying to use a bamboo slate on Linux MINT 18.3. Installation of the kernel modul semmed to work as 
` modinfo wacom | grep version`
Results in
```
version:        v2.00-0.45.0
srcversion:     0AA19FE5A3E9C50CD127943
```
But i am not able to pair the device via bluetooth. Is there something I am missing?

drpjkurian, 2020-04-20:
The output shows that you have installed the driver version 45.
Let me share my experience,
In my netbook, I have Lubuntu 18.04.4, in this machine, the wacom one
tablet works on driver version 44 instead of version 45. Wherea in my
wife's machine which runs on Lubuntu 16.04, the same tablet works on driver
version 43 instead of 44.
Take home answer is you may try older driver version.

karlkappe, 2020-04-20:
Thanks for the answer. I tried ver 43 but also failed. I will try out 44 just in case :-).

jigpu, 2020-04-20:
@drpjkurian could you please try using the current development version of our driver? We recently fixed that possibly-related bug I mentioned (#155) and I am interested to know if it has also fixed the issue for you as well.

Follow the instructions at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source, only instead of downloading the latest release, please install the `git` command and then run the following to check out the current development version:

~~~
git clone https://github.com/linuxwacom/input-wacom.git
~~~

Next, enter the input-wacom directory and continue following the instructions from the "Build / Install" heading.

jigpu, 2020-04-20:
@karlkappe I suspect you are experiencing a different issue. Please open a new issue with a description of how you are trying to pair the tablet and any errors that you are aware of. We'll see what we can do :)

Zaffy, 2021-01-08:
The same issue with "refusing to register" happened to me with CTL-672.
I can confirm that building and using the development version fixed the issue.

jigpu, 2021-01-11:
Thanks for the confirmation @Zaffy. I'll close this issue for now since the fix seems to already be in place :)

closed 2021-01-11 by jigpu