linuxwacom/input-wacom #140: Cintiq 22HD on openSuSE

ivyallie, 2019-12-01:
I just installed openSuSE Tumbleweed with KDE, replacing an older Linux Mint KDE installation that was a few years old. (I think it may have been Mint 16.) On Mint my Cintiq 22HD worked beautifully and I had a shell script that ran a bunch of xsetwacom commands to assign the button functions. However, with openSuSE on the same machine, the monitor is recognized but does not work as a tablet at all. Neither the pen nor the buttons have any effect. The KDE Wacom config GUIs detect it but don't seem to have any effect. I suspect that they may be causing the problem but I'm not sure how to remove them.

The output of sysinfo.sh is attached.

[sysinfo.jzRA1RN92t.tar.gz](https://github.com/linuxwacom/input-wacom/files/3908331/sysinfo.jzRA1RN92t.tar.gz)

skomra, 2019-12-02:
Hi ivyallie,

Could you please also run the event capture script while using the pen on the device:

https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/1050d9aab9514ca0658c3cb415ccb11af5e4d6d6/capture.sh

Thanks!

bogdancovaciu, 2019-12-03:
Once your system is up and running, turn off and on the Cintiq from the back power button. That should make it working in KCM Wacom, and you should be able to set the stylus and the buttons. Another way is to go to **Settings > Startup and Shutdown > Background Services** and stop then restart the Wacom Tablet service from the bottom tab.

I provided more information about this issue here:
https://github.com/linuxwacom/input-wacom/issues/115#issuecomment-537412706

ivyallie, 2019-12-03:
Thanks bogdancovaciu! I turned it off and on again with the button and now it's recognized. I was even able to restore my old configuration with my script. I do feel silly for not having tried this already though!

closed 2019-12-03 by ivyallie

jigpu, 2020-04-27:
Explicitly marking as a duplicate of https://github.com/linuxwacom/input-wacom/issues/115