linuxwacom/input-wacom #145: Screen rotation and "pan"

sebasmonia, 2020-01-15:
I'm using a script that automatically rotates the devices via xsetwacom when I go into table mode (Lenovo L390 Yoga)

```
# more code before this
xinput set-prop "Wacom Pen and multitouch sensor Finger touch" --type=float "Coordinate Transformation Matrix" $matrix
xinput set-prop "Wacom Pen and multitouch sensor Pen stylus" --type=float "Coordinate Transformation Matrix" $matrix
xinput set-prop "Wacom Pen and multitouch sensor Pen eraser" --type=float "Coordinate Transformation Matrix" $matrix
```

Everything worked as expected until I changed my eraser button from middle click to `pan`:

Old: `xsetwacom set "Wacom Pen and multitouch sensor Pen eraser" Button 1 "2"`
New: `xsetwacom set "Wacom Pen and multitouch sensor Pen eraser" Button 1 pan`

Now when I rotate the laptop, the `pan` directions to scroll **do not** get rotated. So dragging the stylus left/right scrolls up/down and viceversa.

jigpu, 2020-02-07:
To be clear, are you using only the Coordinate Transformation Matrix to achieve the sensor rotation? Or do you use something like `xsetwacom set <id> rotate <direction>` in your script?

The driver should pan correctly if you use the rotate option because that changes the underlying input device coordinate system. The coordinates are rotated before the "pan" code ever takes effect, so everything just works. Using the Coordinate Transformation Matrix for rotate is _not_ compatible with "pan", however. Here, the transformation takes place in the X server itself after our driver has finished processing the event.

sebasmonia, 2020-02-10:
I only use the matrix approach, had no idea about the `rotate <direction>` approach.
Will revisit the script I copied from and try to convert it, then re-test.
Thank you!

sebasmonia, 2020-02-22:
Modified my script to:

```
# more code before this
xsetwacom set "Wacom Pen and multitouch sensor Finger touch" rotate $wacomrotate
xsetwacom set "Wacom Pen and multitouch sensor Pen stylus" rotate $wacomrotate
xsetwacom set "Wacom Pen and multitouch sensor Pen eraser" rotate $wacomrotate
```

where I determine the value of `wacomrotate` as explained in https://github.com/linuxwacom/xf86-input-wacom/wiki/Rotation and now it all works as expected (plus the script is less cryptic than with the matrix!).

Closing the issue, not a linuxwacom problem. Thank you!

closed 2020-02-22 by sebasmonia