linuxwacom/input-wacom #146: Touch cause Cintiq Pro 32 Touch to randomly lose pen input unless disable with physical button

tysontan, 2020-01-25:
I recently noticed an oddity on my Cintiq Pro 32 Touch. The pen display would randomly lost pen input completely when touch is enabled. The only workaround is to disable touch sensitivity using the physical button on the face of the display.

Although I can also disable touch in Linux driver, doing so would not prevent the problem from happening. 

Extensive tests have been conducted, changing every possible thing on the BIOS settings, port arrangement, Linux driver settings ... to no effect.

It doesn't happen on my Cintiq Pro 24 (no touch), it doesn't happen under Windows. But it happens under Gnome (wayland), and KDE Plasma (xorg). 

My distro is Manjaro 18.1.5, Linux 5.4. AMDGPU Polaris 10/11 cards, tested on C232 / C236 / Z370 / Z390 motherboards, Wacom Link Plus. The Cintiq Pro 32 Touch is a special model for China, it has only USB type-C input, Wacom Link Plus is required to use this product on a desktop PC, although it wasn't shipped with one. -_-

This bug must have been around for quite some time. The memory of horror goes way back in time, although I didn't draw so often back then to notice the pattern.

tattsan, 2020-02-07:
Same here. I use Cintiq Pro 32 on ThinkPad X1 Carbon gen3, with Arch Linux.
This problem has been seen since kernel 5.3.x, so I used 5.2.14 for a while.

jigpu, 2020-02-07:
@tattsan can you let us know if you are also using one of the China models? If not, are you also using USB-C or using the USB-A / DisplayPort connections instead?

tattsan, 2020-02-07:
My Cintiq Pro 32 has DisplayPort, HDMI, USB-A, USB-C ports. I am using USB-A / DisplayPort connection.

I don't need touch function, so I always disable it by touch key panel. But every time I start the tablet up, it is enabled;-)

dantempla, 2020-02-21:
I have a similiar problem on my **Wacom Intuos Pro S** when using **bluetooth**. I am resting my palm on the tablet while having the pen close to the surface and sometimes it recognizes touch instead of the pen. It sounds similiar to the problem described here. It does not occur when I am using it wired with usb-c.

Up-to-date input-wacom-dkms from git repository here and am on Arch Linux with Kernel 5.5.4

tysontan, 2020-02-21:
@dantempla No, It's a totally different thing. You situation is exactly how the device should normally behave. Palm rejection depends on touch sensitivity and stylus height, human cannot sense or control how high exactly the stylus is off the surface so the pen goes in or out the range as your hands moving. It'll never be reliable.

The bug we are discussing here, does not require the hand or the pen to be placed on the surface to trigger. It happens randomly as long as the touch function is ON. When it happens, the pen input is totally dead, and stay dead until it randomly decided to come back.

dantempla, 2020-02-21:
@tysontan Oh, okay. I understand and thank you for the guesses on my problem, they sound reasonable. The problem does not occur on Windows 10, so I guess it must be a software thing.

tysontan, 2020-02-22:
I used the scripts from @jigpu (https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh) to capture a log when the bug was triggered. On Arch you need to install the foloowing package to run the script:
1) evemu
2) hid-tools

During the capture, pen input stay dead for most of the time until it got back to life in the last few seconds. Here is the captured file:
[record_1582336262.tar.gz](https://github.com/linuxwacom/input-wacom/files/4238699/record_1582336262.tar.gz)

@tattsan maybe you can also attempt the capture so we have more data? Thanks!

tysontan, 2020-02-22:
Additionally, I found out that even after turning OFF Touch by using the physical button, the pen input would still die at rare occasions. It's much less frequent than leaving the Touch is ON though.

tattsan, 2020-02-22:
Thanks @tysontan !  I will try that script.
I hadn't seen this bug when I was using kernel 5.2.14 or older, and came across it when I upgraded the kernel to 5.3.

tattsan, 2020-02-22:
Here is my log:
[record_1582354444.tar.gz](https://github.com/linuxwacom/input-wacom/files/4238936/record_1582354444.tar.gz)

I waited until the pen is recognized again, but the pen did't came back.

tattsan, 2020-02-22:
The above record_1582354444.*.log are logs of the Pen device.
I also recorded the Touch device, here is the archive:
[record_1582357476.tar.gz](https://github.com/linuxwacom/input-wacom/files/4238981/record_1582357476.tar.gz)

jigpu, 2020-02-24:
Thanks @tysontan. Looks like we're still getting valid data from the tablet at the beginning of the log, but no events are leaving the kernel. This smells a lot like touch state corruption, which could make the kernel accidentally enable arbitration when it shouldn't. I know I wrote a script to find corruption in the touch logs, but I'm not sure where I left it... Hopefully I don't have to end up re-writing it...

jigpu, 2020-03-05:
Made a bit of progress on this. I've [built a tool](https://gist.github.com/jigpu/2cc07e3086de368222a6b9e62c4b4930) to monitor the status of multitouch devices to see when the touch state gets corrupted and used this to find a minimal HID recording which triggers the bug (below). Together, this has allowed me to bisect the problem to commit 13b5f0318d37bb869b1d167c2f041b5ffffa639b

The issue seems to be with how we're counting the number of expected touches, and this commit definitely changes that behavior, so I'll see if I can understand what is wrong.

~~~
D: 0
R: 833 06 00 ff 09 01 a1 01 75 08 26 ff 00 15 00 85 06 95 3f 09 01 91 02 85 05 95 3f 09 01 81 02 c0 05 0d 09 04 a1 01 85 81 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 75 08 09 51 95 01 81 02 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 56 55 0c 66 01 10 47 ff ff 00 00 27 ff ff 00 00 75 10 95 01 81 02 09 54 95 01 75 08 15 00 25 0a 81 02 09 55 85 84 95 01 75 08 15 00 25 0a b1 02 85 87 06 00 ff 09 c5 15 00 26 ff 00 75 08 96 00 01 b1 02 c0 05 0d 09 04 a1 01 85 88 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 95 07 81 03 05 01 65 11 55 0e 35 00 46 43 1b 26 00 3c 75 10 95 01 09 30 81 02 46 56 0f 26 c0 21 09 31 81 02 c0 05 0d 09 56 55 0c 66 01 10 47 ff ff 00 00 27 ff ff 00 00 75 10 95 01 81 02 c0 05 0d 09 0e a1 01 85 83 09 23 a1 02 09 52 09 53 15 00 25 0a 75 08 95 02 b1 02 c0 c0
N: Wacom Co., Ltd. Cintiq Pro 32 Touch
P: usb-0000:00:14.0-4.4.3.1/input0
I: 3 056a 0356
D: 0
E: 64.058422 64 81 01 00 8d 0d 17 13 01 01 aa 0f ff 0f 01 02 23 13 ca 0e 01 03 29 19 67 16 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 60 08 04
E: 64.063657 64 81 00 00 8d 0d 17 13 01 01 bb 0f f7 0f 01 02 2a 13 cb 0e 01 03 23 19 64 16 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 97 08 04
E: 64.069584 64 81 01 01 da 0f ea 0f 01 02 32 13 cc 0e 00 03 23 19 64 16 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff cd 08 03
E: 64.074451 64 81 01 01 fb 0f dd 0f 01 02 3b 13 cd 0e ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 03 09 02
E: 64.079446 64 81 01 01 27 10 cf 0f 01 02 45 13 ce 0e ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 3a 09 02
E: 64.085413 64 81 01 01 4f 10 c6 0f 01 02 51 13 d1 0e ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 70 09 02
E: 64.090428 64 81 01 01 7b 10 c0 0f 01 02 5d 13 d4 0e ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff a6 09 02
E: 64.096749 64 81 01 01 93 10 c0 0f 01 02 70 13 d9 0e ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff dd 09 02
E: 64.101632 64 81 00 01 93 10 c0 0f 01 02 7f 13 df 0e ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 13 0a 02
E: 64.107450 64 81 00 02 7f 13 df 0e ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 49 0a 01
~~~

jigpu, 2020-03-23:
I've spent a long time looking at this issue and I think the best way forward is to revert the buggy commit and then potentially revisit the feature it was introducing at a later time. I've pushed up a new branch to my repository which contains the revert as well as another touch-related fix.

[Follow the instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) to build the driver. Instead of downloading the latest driver release, however, run the following two commands to check out the branch with the fix:
~~~
git clone https://github.com/jigpu/input-wacom
git checkout touch-fixes
~~~

Please let me know if this branch fixes the behavior for you. I'll be submitting these fixes upstream once I get the green light from you and after finishing my own final tests.

tysontan, 2020-03-29:
Thank you Jason! I'm rushing some tasks at the moment. I will test it a bit later and report back to you.

tattsan, 2020-04-05:
Thank you so mutch. I built and install the module yesterday, with no trouble until now.
I will continue testing the module.

tysontan, 2020-04-05:
Thank you tattsan!

I've been testing the patch today as well. So far I haven't noticed any problem. Please give me a few more days so that I can be absolutely sure of it.

tysontan, 2020-04-08:
After using the patched version heavily for a few days, I never noticed any problem. I think it's safe. Thank you again jigpu!

jigpu, 2020-04-08:
Thanks for the thorough testing @tysontan and @tattsan. I've just sent the patch upstream and will let you know when its been accepted.

tattsan, 2020-04-09:
I've also been with no troubles too. Thank you!