linuxwacom/input-wacom #149: Intuous S tablet mounts in "low-power" mode

dbazile, 2020-02-12:
When plugged into my Fedora machine, my Wacom Intuos S (CTL-4100) mounts in some strange "low-power" mode where the hardware light is half as bright as it normally should be and DPI is unusably low (1 cm pen stroke hops approx. 200-300 pixels on screen).

The device reports two different USB device IDs depending on the "mode" it's in: `056a:0374` for full-power and `2d1f:0374` for low-power.  Plugging in the tablet starts at that first full-power ID, then immediately falls through to that low-power device ID by default under normal conditions.

## System Info

```
Tablet:
    Wacom Intuos S (model number CTL-4100)

Fedora 30:
    libwacom  0.33-1.fc30
    kernel    5.3.16-200.fc30

Fedora 31:
    libwacom  1.1-2.fc31
    kernel    5.4.17-200.fc31
    usbguard  0.7.6-6.fc31
```

## Logs and Outputs

The following is journalctl output when I plug in the tablet to my machine (Fedora 30):

```
$ journalctl --follow

Feb 11 19:05:45 kefka kernel: usb 1-1: new full-speed USB device number 66 using xhci_hcd
Feb 11 19:05:46 kefka kernel: usb 1-1: New USB device found, idVendor=056a, idProduct=0374, bcdDevice= 1.11
Feb 11 19:05:46 kefka kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 11 19:05:46 kefka kernel: usb 1-1: Product: Intuos S
Feb 11 19:05:46 kefka kernel: usb 1-1: Manufacturer: Wacom Co.,Ltd.
Feb 11 19:05:46 kefka kernel: usb 1-1: SerialNumber: REDACTED
Feb 11 19:05:46 kefka kernel: input: Wacom Intuos S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0374.0055/input/input156
Feb 11 19:05:46 kefka kernel: input: Wacom Intuos S Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0374.0055/input/input158
Feb 11 19:05:46 kefka kernel: wacom 0003:056A:0374.0055: hidraw6: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos S] on usb-0000:00:14.0-1/input0
Feb 11 19:05:46 kefka mtp-probe[7101]: checking bus 1, device 66: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1"
Feb 11 19:05:46 kefka mtp-probe[7101]: bus: 1, device: 66 was not an MTP device
Feb 11 19:05:46 kefka systemd-udevd[7105]: Process 'u2f_hidraw_id --udev' failed with exit code 1.
Feb 11 19:05:46 kefka gnome-shell[2200]: Could not open device /dev/input/event25: Could not get device info for path /dev/input/event25: No such file or directory
Feb 11 19:05:46 kefka gnome-shell[2200]: Could not open device /dev/input/event24: Could not get device info for path /dev/input/event24: No such file or directory
Feb 11 19:05:46 kefka kernel: input: Wacom Intuos S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0374.0056/input/input159
Feb 11 19:05:46 kefka kernel: input: Wacom Intuos S Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0374.0056/input/input161
Feb 11 19:05:46 kefka kernel: wacom 0003:056A:0374.0056: hidraw6: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos S] on usb-0000:00:14.0-1/input0
Feb 11 19:05:46 kefka mtp-probe[7108]: checking bus 1, device 66: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1"
Feb 11 19:05:46 kefka mtp-probe[7108]: bus: 1, device: 66 was not an MTP device
Feb 11 19:05:46 kefka upowerd[1821]: treating change event as add on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0374.0056/power_supply/wacom_battery_4
Feb 11 19:05:46 kefka kernel: usb 1-1: USB disconnect, device number 66
Feb 11 19:05:46 kefka kernel: debugfs: Directory '17' with parent 'devices' already present!
Feb 11 19:05:46 kefka kernel: usb 1-1: new full-speed USB device number 67 using xhci_hcd
Feb 11 19:05:47 kefka kernel: usb 1-1: New USB device found, idVendor=2d1f, idProduct=0374, bcdDevice= 1.11
Feb 11 19:05:47 kefka kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 11 19:05:47 kefka kernel: usb 1-1: Product: Intuos S
Feb 11 19:05:47 kefka kernel: usb 1-1: Manufacturer: Wacom Co.,Ltd.
Feb 11 19:05:47 kefka kernel: usb 1-1: SerialNumber: REDACTED
Feb 11 19:05:47 kefka kernel: input: Wacom Co.,Ltd. Intuos S as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:2D1F:0374.0057/input/input162
Feb 11 19:05:47 kefka kernel: hid-generic 0003:2D1F:0374.0057: input,hidraw6: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos S] on usb-0000:00:14.0-1/input0
Feb 11 19:05:47 kefka mtp-probe[7154]: checking bus 1, device 67: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1"
Feb 11 19:05:47 kefka mtp-probe[7154]: bus: 1, device: 67 was not an MTP device
Feb 11 19:05:47 kefka mtp-probe[7159]: checking bus 1, device 67: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1"
Feb 11 19:05:47 kefka mtp-probe[7159]: bus: 1, device: 67 was not an MTP device
Feb 11 19:05:47 kefka gnome-shell[2200]: Could not get tablet information for 'Wacom Co.,Ltd. Intuos S': (null)
```

I can force the tablet to mount in "full-power" mode by plugging the tablet into a different machine (Fedora 31) that has usbguard installed (configured to block all unknown devices by default) and running the command `usbguard allow-device SOME_NUM`.  The device attached will be device ID `056a:0374`, short-circuiting whatever mechanism triggers the fall-through to the low-power device ID: the device status light is ultra bright and the DPI is as high as it's supposed to be.  journalctl output for this scenario follows:

```
$ journalctl --follow

Feb 11 19:09:21 biggs usbguard-daemon[6353]: [1581466161.638] (W) Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0
[... 77 lines cut ...]
$ usbguard watch

[device] PresenceChanged: id=7
 event=Insert
 target=block
 device_rule=block id 056a:0374 serial "REDACTED" name "Intuos S" hash "REDACTED" parent-hash "REDACTED" via-port "1-4" with-interface 03:00:00 with-connect-type "hotplug"
[device] PolicyChanged: id=7
 target_old=block
 target_new=allow
 device_rule=allow id 056a:0374 serial "REDACTED" name "Intuos S" hash "REDACTED" parent-hash "REDACTED" via-port "1-4" with-interface 03:00:00 with-connect-type "hotplug"
 rule_id=3
[device] PresenceChanged: id=7
 event=Remove
 target=allow
 device_rule=allow id 056a:0374 serial "REDACTED" name "Intuos S" hash "REDACTED" parent-hash "REDACTED" via-port "1-4" with-interface 03:00:00 with-connect-type "hotplug"
[device] PresenceChanged: id=8
 event=Insert
 target=block
 device_rule=block id 2d1f:0374 serial "REDACTED" name "Intuos S" hash "REDACTED" parent-hash "REDACTED" via-port "1-4" with-interface 03:00:00 with-connect-type "hotplug"
[device] PolicyChanged: id=8
 target_old=block
 target_new=allow
 device_rule=allow id 2d1f:0374 serial "REDACTED" name "Intuos S" hash "REDACTED" parent-hash "REDACTED" via-port "1-4" with-interface 03:00:00 with-connect-type "hotplug"
 rule_id=4
```

dbazile, 2020-02-12:
Found this explanation that fits what I'm seeing on the [libwacom wiki](https://github.com/linuxwacom/libwacom/wiki/Troubleshooting#special-cases):

> ## Special Cases
> 
> The third generation Wacom Intuos received a firmware update to enable Android support. This support was added using the generic HID driver in Android (Android doesn't usually ship with the Wacom driver). This update sees the firmware now "detecting" if it is connected to an Android system. If the tablet thinks it is connected to an Android system, it will show the system a 2d1f Vendor ID (VID), and thus libwacom (and the Control Center) will not recognize the tablet. Unfortunately this detection mechanism will in some cases identify a Linux system as an Android one. To manually put the tablet back into its original mode (056a VID), hold down buttons 1 & 4 for N number of seconds.
> 
> A list of devices affected by the 2d1f VID problem:
> 
> | Name              | Model Name | VID:PID   |
> |-------------------|------------|-----------|
> | Wacom Intuos M    | CTL-6100   | 056a:0375 |
> | Wacom Intuos BT M | CTL-6100WL | 056a:0378 |
> | Wacom Intuos S    | CTL-4100   | 056a:0374 |
> | Wacom Intuos BT S | CTL-4100WL | 056a:0376 |

Apparently it's a thing Wacom's doing via firmware.  I'd submit a bug report asking them if they're going to fix this if they offered _any_ electronic means communicating with them aside from via telephone...
Either way, since this isn't something input-wacom or libwacom is doing, I'm closing this ticket.

closed 2020-02-12 by dbazile

jigpu, 2020-02-13:
Yeah, unfortunately the Android detection done by the firmware isn't 100% reliable. The difference between "Linux" and "Android" is pretty minimal, so I'm not exactly surprised. The fwupd daemon (which is operating perfectly fine) can unintentionally trigger the Android-mode switch, among other things.

BYT3M3, 2020-10-30:
I have a additional problem with this Android mode. Whenever I unplug the CTL-6100WL and plug it back in, it goes into Android mode and if I then switch to PC mode, I cannot get the pen to be mapped to a single monitor. This is a very important feature for my online teaching. Could anyone please help me with this?

dbazile, 2020-10-30:
Hi @BYT3M3,

You may have more luck opening a new ticket. Not sure if anyone's going to be looking at this old closed ticket. :)

stevebranam, 2020-10-30:
I had the Android problem and the dual-screen problem on my Thinkpad running 18.04 with an extra monitor. Holding buttons 1 and 4 down fixed the Android mode problem. To fix the dual-screen problem, I used a variation of this post: [](https://github.com/linuxwacom/xf86-input-wacom/wiki/Dual-and-Multi-Monitor-Set-Up).

I used xrandr to get my display info, and then xsetwacom with MapToOutput to set the proper tracking:

`stephen.branam@ubuntu:~$ uname -a
Linux ubuntu 5.4.0-52-generic #57~18.04.1-Ubuntu SMP Thu Oct 15 14:04:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

stephen.branam@ubuntu:~$ xrandr
Screen 0: minimum 8 x 8, current 3840 x 1080, maximum 32767 x 32767
DP-0 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-0 connected 1920x1080+1920+0 (normal left inverted right x axis y axis) 598mm x 336mm
   1920x1080     60.00*+  74.97    59.94    50.00  
   1680x1050     59.95  
   1440x900      59.89  
   1280x1024     75.02    70.00    60.02  
   1280x720      60.00    59.94    50.00  
   1152x864      75.00  
   1024x768      75.03    70.07    60.00  
   800x600       75.00    72.19    60.32  
   720x576       50.00  
   720x480       59.94  
   640x480       75.00    72.81    59.93    59.94  
DP-2 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 344mm x 194mm
   1920x1080     60.00*+  48.00  
`
The DP-2 is my laptop display, the HDMI-0 is my second monitor. However, to reference them properly, I needed to use the coordinate info:

`stephen.branam@ubuntu:~$ # Set Wacom to built-in monitor:
stephen.branam@ubuntu:~$ xsetwacom set "Wacom Intuos BT S Pen stylus" MapToOutput 1920x1080+0+0

stephen.branam@ubuntu:~$ # Set Wacom to monitor plugged into HDMI:
stephen.branam@ubuntu:~$ xsetwacom set "Wacom Intuos BT S Pen stylus" MapToOutput 1920x1080+1920+0
`
Now it works fine on either one, depending which one I switch it to.

jigpu, 2020-11-05:
@BYT3M3 if you're running the GNOME desktop it should also be possible to set up the monitor mapping within the Wacom tablet settings. KDE may also have the ability to set the monitor mapping through settings, but I am unsure. Any other distribution will require manual mapping as mentioned above.

@stevebranam the output names like "HDMI-0" and "DP-2" from xrandr should also work. The raw coordinates are definitely a failsafe backup though :)

Davidy22, 2021-08-18:
Is there no corrective procedure we can take after we detect that the tablet is in low power mode? xsetwacom gives pretty distinct responses between the modes so I'm sure we can do the detection side, is there any signal we can send to imitate the left/right button reset that I'm doing right now to workaround this? Because this is pretty unexpected behavior and I didn't discover the workaround the first time I googled this because the keywords needed were a little obtuse, plugging a tablet into a computer is pretty normal usage and it's probably better for linux if we do the workaround instead of hope for wacom to do it.

dbazile, 2021-08-18:
> Is there no corrective procedure we can take after we detect that the tablet is in low power mode? xsetwacom gives pretty distinct responses between the modes so I'm sure we can do the detection [...]

Holding down the first and fourth button for a few seconds tells the tablet to switch to full power mode. Hope this helps.

Davidy22, 2021-08-18:
> is there any signal we can send to imitate the left/right button reset that I'm doing right now to workaround this?

I already know how to invoke the workaround manually. This isn't good behavior for the driver though, and if there's a signal the driver can send to invoke the same switch we can attempt to actually fix this instead of hoping that artist types find the wiki page here, or that wacom fixes this for us. Detecting that we're in android mode and sending the reset signal is still a bit of a workaround, the ideal would be somehow emitting the right information for the tablet to correctly determine we're on linux like it already does when the computer boots with the tablet plugged in, but making it detect correctly at all's better than leaving it like this.

dbazile, 2021-08-18:
Heh totally skipped over that part...