linuxwacom/input-wacom #150: wacom_i2c support for w9002 and friends

Mis012, 2020-02-16:
Hi,
since the downstream drivers (I looked at samsung's android trees) look quite different compared to the mainline driver, I was wondering if the mainline driver is likely to work with w9002? There is no information on the i2c protocol on the wiki as far as I can see, so I'm not sure if it's compatible with all the i2c-speaking chips or not.

(I do not have one of these android devices, though some of them look mainlinable with a bit of effort; I have a Surface Pro display assembly and a raspberry pi)

jigpu, 2020-02-19:
I don't know much about the W9002, so its hard for me to say how similar the protocols are. In the past, Android sensors have relied on our ["wacom_i2c" driver](https://github.com/torvalds/linux/blob/master/drivers/input/touchscreen/wacom_i2c.c) driver. It looks like the Samsung driver is quite a bit larger, however, which doesn't make very confident.

Its possible your Surface Pro display assembly uses a chipset which is more similar to that found in many tablet PCs. These also communicate over the I2C bus but provide a HID descriptor that allows them to be picked up by our standard ["wacom" driver](https://github.com/torvalds/linux/blob/master/drivers/hid/wacom_sys.c).

Mis012, 2020-02-27:
I don't think it's hid over i2c. it's an android-y chip, w9002. Microsoft uses a sensor hub to convert it to usb hid, but I'd rather use the i2c driver

Mis012, 2020-03-05:
hmm, it's called wacom_g5sp_i2c here:https://github.com/haoyangw/android_kernel_samsung_klte/blob/28c0f3c3da523469e3f1cebd56a41cff9652d529/drivers/input/touchscreen/wacom_hl/wacom_i2c.c  
It would include more than just W9002 I assume, but not really sure which chips exactly.

g5 could mean generation 5, so protocol 5? perhaps it bears resemblance to that protocol?

Mis012, 2020-03-05:
seems it's on address 0x09

Mis012, 2020-03-08:
all samsung's android downstream kernels I can find suggest 0x56, so that's a difference

jigpu, 2020-03-09:
If you can capture the I2C data stream while the pen is in use, that should help with decoding the protocol. I wouldn't expect the protocol to be particularly exotic. I'm pretty sure there's a way to have the kernel itself dump I2C communications to dmesg, but I can't remember how off the top of my head. You may need to do some digging in the I2C subsystem to find kernel command-line flags used for debugging.

Mis012, 2020-03-10:
it seems to work fine with the wacom-i2c driver in mainline, except that I get
```
Event: time 1583865289.984757, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1583865289.984757, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 1
Event: time 1583865289.984757, type 1 (EV_KEY), code 331 (BTN_STYLUS), value 1
Event: time 1583865289.984757, type 1 (EV_KEY), code 332 (BTN_STYLUS2), value 1
Event: time 1583865289.984757, type 3 (EV_ABS), code 0 (ABS_X), value 65535
Event: time 1583865289.984757, type 3 (EV_ABS), code 1 (ABS_Y), value 65535
Event: time 1583865289.984757, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 65535
Event: time 1583865289.984757, -------------- SYN_REPORT ------------
Event: time 1583865289.987398, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1583865289.987398, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 0
Event: time 1583865289.987398, type 1 (EV_KEY), code 331 (BTN_STYLUS), value 0
Event: time 1583865289.987398, type 1 (EV_KEY), code 332 (BTN_STYLUS2), value 0
Event: time 1583865289.987398, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1583865289.987398, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1583865289.987398, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1583865289.987398, -------------- SYN_REPORT ------------
```
continuously when idle

Mis012, 2020-03-10:
but rpi irqsare weird /shrug

Mis012, 2020-03-10:
oh, well, it seems all that was needed was configuring a pull-up :man_facepalming: