linuxwacom/input-wacom #151: cintiq pro 13 does not work on ubuntu 18.04.2 LTS

sinamone, 2020-02-18:
Hi everyone, 
I have look on the different issues topics and some of them seems similar to mine but i am an ubuntu baby user so i didn't understand a single word i have read. 
Sorry if I am not writing at the good place.
I have a problem with my cintiq 13 pro, which is working a few second and then suddenly become a usb thing, just detected as a screen and no more as a cintiq (and the pen stop being detected to).
I am using a wacom link plus adaptator.
The cintiq works perfectly well on windows (but i don't like it).
I don't understand what's the problem or what should i do and i am afraid of destroying everything if i just try by myself. 
Could someone help me please ?

sinamone, 2020-02-19:
Hi everyone
Finally i resolve it y myself, i just upgrade everything and it works now. If another ubuntu baby come here : 
`sudo apt update
sudo apt full-upgrade -y`

jigpu, 2020-02-19:
Glad to hear that you were able to resolve the issue with an upgrade.

closed 2020-02-19 by jigpu