linuxwacom/input-wacom #153: Build Failed - Kernel doesn't have the configuration files

Twiggeh, 2020-03-01:
Terminal Dump

jigpu, 2020-03-01:
Your issue of the Cintiq being stretched to both displays won't be fixed by a kernel driver update. The display mapping is entirely controlled by userspace. If you are using GNOME or Cinnamon, it should automatically map the Cintiq correctly. If you are using any other desktop you will need to use the "MapToOutput" option of xsetwacom to map the pen to the correct display. What desktop environment are you using?

For the build issue: can you provide the output from `ls /lib/modules` and `uname -r` please?

For the xsetwacom issue: can you provide the output from `echo $XDG_SESSION_TYPE` please?

Twiggeh, 2020-03-01:
Hi ! 
I'm using xfce, and I tried reinstalling the driver because `xsetwacom --list devices` returned nothing. I have no idea why, but it started working and I could remap my output by using `xsetwacom set "Wacom Cintiq 13HD Pen stylus" MapToOutput "HEADER-0"`, so that resolved my issue :) 

I did not continue trying to build the drivers, but here are the outputs that you asked for :

ls /lib/modules : 
```
3.16.82-2-MANJARO
5.6.0-2-MANJARO
extramodules-5.6-MANJARO
4.14.171-1-MANJARO
extramodules-4.14-MANJARO
```
uname -r
```
5.6.0-2-MANJARO
```
echo $XDG_SESSION_TYPE 
```
x11
```
Interestingly enough a new issue arose. Drawing a line doesn't result in a continuous line, but sometimes the stroke disconnects as if I've lifted the pen. Would you know how to troubleshoot that ? [image]

Thank you so much for your help so far and please tell me if I should try building the drivers if that is of any value to you :)

closed 2020-10-07 by Twiggeh