linuxwacom/input-wacom #154: Is there support for Cintiq ID 056a:00f9

hoyanf, 2020-03-04:
Hi, 
I'm just wondering if there is official support for Wacom Cintiq with USB ID 056A:00F9.

Thanks

jigpu, 2020-03-04:
056a:00f9 is a USB hub and I believe should work fine. I'm not aware of anything special about this particular device. The "hub nested too deep" error that you see is due to the Cintiq being connected through too many layers of USB hubs. Try connecting the Cintiq to a different USB port, ideally one directly on the system itself.

closed 2020-03-04 by jigpu

hoyanf, 2020-03-05:
[image]
The reason am asking because the device id is not within the one listed as supported.

Thanks

jigpu, 2020-03-05:
The "usb.ids" list is only used to provide "friendly" device names in programs like `lsusb`. It is in no way a list of supported devices, or has any impact on if a device will work. Many tablet PC sensors, for instance, have not been added to the list (new ones keep popping up all the time, and we aren't notified about most of them...)

I've just added the device to the list anyway, and it should be showing up in the future. Note that it can take weeks or months for updates to the list to roll out...