linuxwacom/input-wacom #155: regression: touchscreen functionality lost in Panasonic Toughbook CF-C2A

alexbobp, 2020-03-08:
I have a touchscreen laptop that has had multiple hardware revisions.  Between September 20, 2019 and September 23, 2019, (from revision 31070 to 31080), the distro I've been using, Clear Linux, stops working with the touchscreen, but only on the CF-C2A unit.  The later hardware revision (which has a different usbid for the wacom device) keeps working fine.  I've also seen the touchscreen go from not working to working in older and newer releases of other distros, and if desired, I could do more detailed testing on another distro to try to figure out when it breaks.

As of revision 31080, finger-touch stops working on the Panasonic CF-C2A laptop, but still works fine on the CF-C2C (hardware revision of the same model laptop). Stylus input still works either way.
In revisions 31070 and prior, finger-touch works fine on both units.

The multitouch sensor on the A unit has a usb id of 056a:0101
The multitouch sensor on the B unit has a usb id of 056a:0117

I have already made a ticket downstream on the Clear Linux project, at https://github.com/clearlinux/distribution/issues/1808 which contains lsmod output, a comparison of dmesg output, and a link to the release notes for the clear linux update.  Sadly those release notes have no mention of wacom.

The most notable part is the difference in dmesg output from the wacom driver.  On the earlier, working revision, I see:
`
6.585001] calling wacom_driver_init+0x0/0x1000 [wacom] @ 377 [ 6.587126] input: Wacom ISDv4 101 Pen as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0/0003:056A:0101.0003/input/input11 [ 6.587587] wacom 0003:056A:0101.0003: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and MultiTouch Sensor ] on usb-0000:00:1a.0-1.5/input0 [ 6.589117] input: Wacom ISDv4 101 Finger as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.1/0003:056A:0101.0004/input/input15 [ 6.589234] wacom 0003:056A:0101.0004: hidraw2: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and MultiTouch Sensor ] on usb-0000:00:1a.0-1.5/input1 [ 6.589269] initcall wacom_driver_init+0x0/0x1000 [wacom] returned 0 after 4158 usecs
`
On the later, broken revision, I see:
`
6.956628] calling wacom_driver_init+0x0/0x1000 [wacom] @ 382 [ 6.960998] input: Wacom ISDv4 101 Pen as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0/0003:056A:0101.0003/input/input11 [ 6.961447] wacom 0003:056A:0101.0003: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and MultiTouch Sensor ] on usb-0000:00:1a.0-1.5/input0 [ 6.963594] wacom: probe of 0003:056A:0101.0004 failed with error -22 [ 6.963611] initcall wacom_driver_init+0x0/0x1000 [wacom] returned 0 after 6362 usecs
`

alexbobp, 2020-03-08:
Note that I still have both the CF-C2A and the CF-C2C laptops available for any testing that's needed.

jigpu, 2020-03-09:
As I mentioned on the downstream Clear Linux bug report, the update from 31070 to 31080 changes the Linux kernel from 5.2.16-835 to 5.2.17-836, and introduces a change to the Wacom kernel driver. To test if this change is the source of the issue, I would recommend performing the following steps.

1. Install "git" which will be necessary to build the different versions.
2. Run `git clone https://github.com/linuxwacom/input-wacom.git`
3. Enter the input-wacom directory and run `git checkout cd47de45b8558d8d9941e8a485a0d1b0879dd755` to get a hopefully-good version of the driver.
4. Follow the wiki's [build instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) to install other pre-requisites and build / install the driver.
5. Run `sudo rmmod wacom && sudo insmod 4.5/wacom.ko` to reload the kernel driver without a reboot
6. Test if the touchscreen works
7. If the toushcreen works, run `git checkout 0a844432469a7daf98b06a2d14644740ab88e953` to get a potentially buggy version of the driver.
8. Re-run the build/install command from the wiki
9. Run Run `sudo rmmod wacom && sudo insmod 4.5/wacom.ko` again
10. Test if the touchscreen works
11. If it doesn't work, the change is probably the source of your problem.

If the 1st commit doesn't work, or the 2nd commit does, then we'll need to cast a bit wider net to see if some other change introduced the problem. You can, of course, try running steps 3 - 11 multiple times if you want to be sure that the 1st commit works fine and the 2nd commit breaks things.

If the 2nd commit does indeed appear to be the source of the trouble, we'll need to work with you to understand why exactly it is causing problems for your particular touchscreen.

jigpu, 2020-03-09:
See also https://bugzilla.redhat.com/show_bug.cgi?id=1754031 and https://github.com/linuxwacom/input-wacom/issues/121 for other "failed with error -22" bugs reported for Wacom touchscreens that could be related.

alexbobp, 2020-03-10:
One key difference from those two bugs is that stylus input still works perfectly normally.  I will gladly try out your instructions above.

alexbobp, 2020-03-10:
Good news.  I've confirmed that when I build the module with the githash you give in 3, it works.  When I rebuild it with the githash you give in 7, it breaks again.  So hopefully that info helps, and this also gives me a good workaround for the immediate term.

Also interesting...  My distro had kernel module signing enforcement by default, and I had to figure out how to turn that off.  But before I figured that out, when I rebooted and the kernel refused to load the wacom driver, it fell back to hid-generic and the touchscreen and stylus both seemed to work fine.  I didn't test extensively, but basic stuff like touching and dragging worked.

Let me know if there's any other useful testing I can do.

jigpu, 2020-03-10:
Thanks for the quick test results. Can you next run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball here? It will provide some additional information about your system and sensor that may be helpful. Getting results fromboth the working and broken versions would be ideal.

alexbobp, 2020-03-11:
Here's the results when running the working driver from git.
[sysinfo.working.tar.gz](https://github.com/linuxwacom/input-wacom/files/4315428/sysinfo.working.tar.gz)

Here's the results after installing the non-working driver version from git and rebooting.
[sysinfo.broken.tar.gz](https://github.com/linuxwacom/input-wacom/files/4315429/sysinfo.broken.tar.gz)

Were you also wanting results from the other hardware, that doesn't error with either driver?

jigpu, 2020-03-18:
Sorry for the delay in getting back to you. I've been looking at this and have a theory for what the source of the bug is. Before I go too much further down the path though, I'd like to get some additional debug info from you.

If you still have the "input-wacom" directory from earlier available, please run the following commands inside it to checkout my debug modifications:

~~~
$ git remote add jigpu https://github.com/jigpu/input-wacom.git
$ git remote update
$ git checkout bug-155-debug
~~~

With the `bug-155-debug` branch checked out, run `make` to rebuild the driver and then `sudo dmesg -C` to clear the dmesg log buffer. Finally, run `sudo rmmod wacom && sudo insmod 4.5/wacom.ko` to force the driver to reload. This is based on the non-working commit, so do not be surprised if touch stops working. With the driver reloaded, run `dmesg > dmesg.log` and attach the dmesg.log file here.

alexbobp, 2020-03-20:
[dmesg.log](https://github.com/linuxwacom/input-wacom/files/4359232/dmesg.log)

jigpu, 2020-03-20:
Thanks, that's helped significantly. I've pushed a new commit which should hopefully fix the issue. You can get the updated code by running the following commands. Afterwards, follow the previous instructions to rebuild the driver, clear dmesg, and reload the driver. Touch should hopefully work now. If not, save and attach the new dmesg output for review.

~~~
$ git remote update jigpu
$ git remote pull
~~~

alexbobp, 2020-03-21:
`git remote pull` didn't seem to exist so I tried `git pull`
Output looked like `Updating 266ba65..fd37ad6
Fast-forward
 4.5/wacom_sys.c | 5 ++++-
 1 file changed, 4 insertions(+), 1 deletion(-)`
I rebuilt and loaded, and touch was restored.  Thanks!
Here's the log just in case.
[dmesg.log](https://github.com/linuxwacom/input-wacom/files/4362700/dmesg.log)

jigpu, 2020-03-23:
The `git remote pull` was a typo on my part, but it looks like you've figured it out. I'll be repackaging this into a kernel patch and sending it out shortly.

jigpu, 2020-03-23:
I've pushed up a new branch with the code that I'm planning on sending up to the kernel. In addition to your fix it also has a fix for another touch-related bug. Please do the following to switch to the new branch and then rebuild/reinstall the driver to test that it works correctly.
~~~
git remote update
git checkout touch-fixes
~~~

alexbobp, 2020-04-01:
I'm sorry, I must have missed the email notification for this one.  I can confirm that this still works for me!

jigpu, 2020-04-03:
Thanks! The fix was accepted upstream, and should be available in the following releases (whenever they are actually made):
* input-wacom 0.46.0
* Linux 5.7
* Linux stable 5.6.TBD
* Linux stable 5.5.TBD
* Linux stable 5.4.TBD

closed 2020-04-03 by jigpu