linuxwacom/input-wacom #157: Possible feature: adjust sensitivity

amm042, 2020-03-21:
I've been using the Wacom intuos (CTL-4100 small) for teaching. It works great under Ubuntu 19.10 but I would like to reduce the XY (positional) sensitivity. I am using Xournal++ to annotate PDFs for video lectures. I have a 27 inch 4k screen (I'm not sure if this is related to DPI at all?). When I draw a line between 2 of the dots on the tablet (0.5 inches) it shows up as 3 inches on my screen. As a result, I have to write very small on the tablet. 

If I turn my mouse sensitivity down, it does affect the tablet XY resolution, however, it makes scrolling across my screen with the mouse tedious. Also, even with my mouse sensitivity all the way down, the 0.5" tablet line is still 1.5" on screen. I would like it to be about the same size on screen as I write on the tablet.

I did also try both relative and absolute positioning and they behave slightly different but both are too sensitive.

Is adjusting this possible? Thanks!

jigpu, 2020-03-23:
If you'd prefer to use absolute mode, the best option would be to re-map the tablet to a smaller portion of your screen and then move the Xournal window into that portion. You can run a command like the following, which says to map the pen into a rectangle of size 1920x1080 anchored at (0,0) (that is to say, the top-left quarter of your 4K screen):

~~~
xsetwacom set "<stylus id>" MapToOutput 1920x1080+0+0
~~~

This command can be placed into a startup script to ensure that ti applies whenever you login.

If you prefer using relative mode, I would suggest adding a config file to ensure that the pointer acceleration options only affect the tablet and not your mouse. The snippet below can be saved as `/etc/X11/xorg.conf.d/90-relative-pen.conf` and should automatically apply its settings whenever you login. You'll need to adjust the "ConstantDeceleration" value up or down to achieve a good feel. Note that you have to logout/login to apply any changes.

~~~
Section "InputClass"
    Identifier "Relative CTL-4100 pen"
    MatchUSBID "056a:0374"
    MatchIsTablet "true"
    MatchIsTouchscreen "false"
    MatchIsTouchpad "false"
    Option "Mode" "Relative"
    Option "AccelerationProfile" "-1"
    Option "ConstantDeceleration" "1"
EndSection
~~~

The "ConstantDeceleration" parameter controls how much slower the pointer should move. A value of 2 will halve the pointer speed, 3 will reduce it to a third, etc. You can of course also increase the speed by using fractional values (e.g. 0.5 would double the pointer speed).

Note that I've also disabled pointer acceleration in this snippet (AcceleratioProfile -1). This will probably make it harder to use the pen for navigation, but should ensure that handwriting remains natural looking. You can, of course, remove this option if you'd like.

amm042, 2020-04-07:
@jigpu I went with relative with a deceleration of 3x and it is exactly what I was looking for. Thank you 1000x. The only minor annoyance is I now have to switch to the mouse to click on things. But still way better in my opinion. Someday exposing these bits (accel, decel) in the GUI widget would be a plus.

jigpu, 2020-04-07:
Hmmm. Does your system go to sleep in between it working as intended and then loosing the setting? The setting will reset any time the device is disconnected, which could happen at reboot, logout, and possibly even sleep.

amm042, 2020-04-10:
I'm using xournal++ (https://github.com/xournalpp/xournalpp). When I initially open the app, it works with the deceleration. When I open a pdf to annotate it reverts to normal speed. The only way I found to get it back to decelerated is to reboot.

Could xournal be calling some pen input function that causes reinit without obeying the x11 conf file? Is there some way to log pen API calls?

jigpu, 2020-04-10:
Its unlikely that Xournal is doing anything like that. Programs generally do not modify those controls, though the desktop environment might. I installed Xournal++ onto my Arch Linux system and quickly tried setting the same acceleration parameters but don't see any change in behavior after opening a PDF for annotation.

You can watch the properties for modification by running the command `xinput watch-props <id>` with the ID of your stylus. It will first print out all the current values of the properties and then print out any subsequent changes to the device properties as they happen. For example, I ran the command and then manually changed the deceleration factor to 4 through the command-line and immediately saw the following message printed out:

~~~
Property 'Device Accel Constant Deceleration' changed.
	Device Accel Constant Deceleration (282):	4.000000
~~~

If you see something similar happen, then you're right that something is changing the properties under your nose. Unfortunately, this particular command won't show _what_ program is triggering the change...

Pinglinux, 2022-08-03:
This is not a kernel driver issue. Something in userland was responsible for the issue

closed 2022-08-03 by Pinglinux

acl21, 2025-03-18:
I had the exact same problem as @amm042 and I was able to get around it by scaling the stylus area x3. Here is what I did: 

1. First, I got my stylus name from 
```
$ xsetwacom --list devices
> <stylus-name>
``` 
2. Second, I got the default stylus area from:
```
$ xsetwacom get "<stylus-name>" Area
> 0 0 X Y
```
3. Then I scaled the stylus area with:
```
$ xsetwacom set "<stylus-name>" Area 0 0 X*3 Y*3
```