linuxwacom/input-wacom #159: onebywacom doesnt work linux mint

jojartb, 2020-03-31:
Hi Everybody,
Today I get my brand new OnebyWacom but unfortunatelly it doeasn't work.
The sysinfo.sh was performed and the output is attached.
Would you be so kind to help me?
Thank you for your help in advance,
Balazs
[sysinfo.mMVDqdXU7v.tar.gz](https://github.com/linuxwacom/input-wacom/files/4410559/sysinfo.mMVDqdXU7v.tar.gz)

Wochozka, 2020-11-05:
Hi,
you have Linux Mint 19.1 Thesa... Upgrade to 19.5 or above. I had the same problem. I still have time to time problems on 19.5, but Linux Mint 20 works with onemywacom perfectly.

Good Luck,

David.

jigpu, 2020-11-05:
You are currently running the Linux 4.15 kernel but the One by Wacom only works out of the box with a Linux 4.16 kernel (see [here](https://github.com/linuxwacom/input-wacom/wiki/Device-IDs)). Your options are to upgrade your distro or to [build and install the input-wacom from source](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source). I would strongly recommend going the upgrade route if possible.

closed 2020-11-05 by jigpu