linuxwacom/input-wacom #160: Mobile Studio Pro

Harpwind, 2020-04-01:
This is not an issue so much as a question.   If this is an inappropriate place to ask, please redirect me to the proper location.

I have a Mobile Studio Pro 16/2016.   I want to install Ubuntu on it and use it directly rather than as a linked device.

Would this be supported by this project?

Harpwind, 2020-04-01:
I went ahead and installed Ubuntu 18.04.   Everything works except the Pro Pen Slim and also doesn't seem to be registering the pressure differences, but I assume that belongs in the other section.

jigpu, 2020-04-01:
We've has a few people install Linux onto MobileStudio Pro devices in the past and it seemed to work well. I would definitely report your issues with the Pro Pen Slim and pressure differences as separate bugs.

I'd normally be able to try and debug what's going on with this particular scenario myself, but unfortunately I left both my MobileStudio Pro and Pro Pen Slim at the office, and probably won't have the ability to return for a while... I can still try to help by analyzing debug logs from you, however.

Harpwind, 2020-04-01:
I'll first apologize to start with.  I am pretty new to Linux and even newer to ubuntu.   And heretofore all my experiences were with command line with no GUI installed.

I had been trying to draw with the Pro Pen Slim.  Which  drew in GIMP, but with no pressure sensitivity.   Sometime after that I found the Wacom tab in the setup section and discovered it didn't detect my stylus.   I then tried the Pro Pen 2 and it was detected.    I have not had the chance to go back and test pressure sensitivity as I left work...which is were my pro still is, with both pens.

When I am back in the AM, I will try the Pro Pen 2 with pressure sensitivity.   I have also been reading and believe that I am still using libwacom .29 when I should be using .32 which may fix my other problem.

After I double check the the Pro Pen 2 is working properly and update to .32, I'll see if any problems still exist and report back.

Thank you for your help!

Harpwind, 2020-04-02:
I checked first thing this AM, pressure sensitivity was not working in GIMP, but when I looked on the wacom stylus properties in settings, I saw a test windows, turns out pressure sensitivity worked there.   I doesn't recognized my Pro Slim as a Slim, but it seems to work the same as the Pro Pen 2.

I then found some settings in GIMP that enabled pressure sensitivity.    If it should be recognized as a Slim, then I guess there is an issue somewhere, otherwise I guess it's good.

jigpu, 2020-04-02:
Glad to hear that you figured out pressure-sensitivity in GIMP. Its not nearly as straightforward as it should be...

As for the Pro Pen Slim, it seems that its definition was added to libwacom 0.33. Have you updated libwacom yet, and if so, to what version?

Harpwind, 2020-04-02:
I just upgraded to .32 this AM.   Didn't realize there was a .33....

closed 2020-04-13 by Harpwind