linuxwacom/input-wacom #161: Build Failed - Kernel directory does not appear to have needed config files

femmebeachtop, 2020-04-03:
trying to get the driver for a wacom intuos but cant seem to make it work at all. tablet not recognized and doesnt work at all when connected via usb

terminal dump is:
$ if test -x ./autogen.sh; then ./autogen.sh; else sh configure; fi && make && sudo make install || echo "Build Failed"
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking if build path '/home/lucaslodias/input-wacom-0.45.0' has spaces... no
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... not found
configure: WARNING: Unable to find build config in any of:
          [/lib/modules/4.19.87-07826-g55ab70aff690/source],
          [/lib/modules/4.19.87-07826-g55ab70aff690/build],
          [/usr/src/linux],
          [/usr/src/linux-4.19.87-07826-g55ab70aff690],
          [/usr/src/linux-2.6]

configure: WARNING: Kernel directory does not appear to have needed config files
checking kernel version... 
configure: WARNING: Spcified '4.19.87-07826-g55ab70aff690' kernel, but found '' instead
configure: error: We could not find the development environment to build modules for the '4.19.87-07826-g55ab70aff690' kernel within the '' directory. Please install the kernel source or the kernel development package and try again.
Build Failed

$ ls /lib/modules
4.19.0-8-amd64

jigpu, 2020-04-03:
What distribution are you using? That kernel version appears to be unique to ChromeOS, and if that's the case, can you provide additional information about the build environment (e.g. if you are using the native shell, running Debian or something else inside a Crostini container, or using the Crouton environment)?

femmebeachtop, 2020-04-03:
its a acer chromebook 514 and i think (new here) im using the native shell (linux beta)
Ive tried a few things and Im pretty much figuring everything out as i go :P
but even with a delete and reinstall of linux i got the same build error message

jigpu, 2020-04-03:
Thanks for the clarification.

The Linux beta is also known as "Crostini", and is actually a special virtual environment rather than the actual native shell that is running under the hood. From what I understand, the virtual environment does not have access to USB devices, so even if you could compile the driver within it, you still wouldn't be able to use the tablet within it.

You could potentially compile the driver in the native shell (if you switch your Chromebook into developer mode to get root access and find a source for the kernel headers), which would give you the ability to use the tablet within ChromeOS. Unfortunately, that alone doesn't appear to be enough to get you full support. Last I heard, several Pixelbook users were complaining that applications run in the Linux beta did not respond to pressure from their otherwise-working pen.

It _has_ been a few months since I stumbled across those reports from Pixelbook users, however, so maybe Google has recently added pressure support to the Linux beta? I would ask around to see if any Pixelbook users are able to get pressure sensitivity inside applications installed to the Linux beta. If they are, then you might be able to make things work with the native shell compile...

Sorry I don't have better news for you :(

closed 2020-04-03 by jigpu