linuxwacom/input-wacom #162: Pen draws straight lines between strokes

abhijit86k, 2020-04-05:
I have recently started seeing this weird behaviour with the factory-included pen/stylus that came with my Lenovo Thinkpad L390. Essentially what happens is that between strokes, from the point where the pen is lifted off the screen to when it makes contacts again, I get a straight line. The undesired strokes are indicated by the red arrows below. This exact behaviour is seen in both Ubuntu (Xournal, Gimp, etc) and in Windows (Windows Ink) which leads me to believe that the sensitivity of the screen+pen is not detecting the the lifting of the pen reliably.

**Attachments:**

**1. Screenshot of the issue**:
The red arrows point to the segments where the pen was lifted:

 [image]

**2. System Info:**

Results of [sysinfo.sh script from linuxwacom project](https://github.com/linuxwacom/wacom-hid-descriptors/tree/master/scripts): 

Tarball: [sysinfo.LRKYocWQHX.tar.gz](https://github.com/linuxwacom/input-wacom/files/4433233/sysinfo.LRKYocWQHX.tar.gz)

**3. Results of the [HID capture script]** (https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/82fb9362998dd8ca582b4efb558ec3abfb11668d/capture.sh):

Screenshot:

[image]

and log file:

url
[record_1586079135.tar.gz](https://github.com/linuxwacom/input-wacom/files/4433247/record_1586079135.tar.gz)

feijoas, 2020-04-05:
Might be related to my issue: https://github.com/linuxwacom/xf86-input-wacom/issues/107

jigpu, 2020-04-06:
Thanks for the recording and associated screenshot. Looks like the hardware is sending us bad data. The moment the pen contacts the display, it reports 100% pressure, and then only reports a "pen up" event in between the "1" and "2" that you drew above.

There's a chance this could be a kernel issue, however, since it appears the sensor is sending us standard pen reports instead of the vendor-defined ones that we normally switch to. It also looks like you're on Ubuntu 18.04.4, which shipped out a kernel update back in February, which might explain why this issue only "recently" appeared. I don't see any particularly suspicious changes, but I'll have a second look and then provide you with instructions for debugging with our input-wacom driver.

abhijit86k, 2020-04-06:
Thank you @jigpu . Do you want me to try booting an earlier Kernel?
As far as I can recall, this problem started to appear some time in Oct/Nov 2019 to the best of my knowledge.

jigpu, 2020-04-06:
Thanks for the clarification. Were you still running Ubuntu 18.04 at that time? If you have earlier kernels available in your boot menu, it would be helpful to know which versions the pen does and does not work with (and if it doesn't work with any of them, let me know the earliest kernel version that you were able to test). That will help me narrow down the range of changes to examine.

abhijit86k, 2020-04-06:
Yes I have been on Ubuntu 18.04 throughout. I am currently on 18.04.4. Not sure if the minor version number changed. I will check with previous kernels and update this message.

abhijit86k, 2020-04-07:
The issue exists in Windows. It appears that the hardware (pressure sensor) is, indeed broken. I think we can put this issue on hold until I can try with a different pen, which, under the current lockdowns, will take a while.

Pinglinux, 2022-08-05:
From what you described and discussed, it is not a Linux driver issue.

closed 2022-08-05 by Pinglinux