linuxwacom/input-wacom #163: input wacom install problem with kernel

Hkyx, 2020-04-10:
Hi, my tanlet is found with the cli but seems not the pen.

I try all your install files like input-wacom and get this error:
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Install DKMS modules
==> dkms install input-wacom/0.44.0 -k 5.4.28-rt19-MANJARO
==> dkms install input-wacom/0.44.0 -k 3.16.82-3-MANJARO
==> Unable to install module input-wacom/0.44.0 for kernel 5.4.28-1-MANJARO: Missing kernel headers.
==> Unable to install module input-wacom/0.44.0 for kernel 5.4.30-1-MANJARO: Missing kernel headers.

it seems input-wacom is looking for 5.4.28 kernel but it's a old kernet, and it don't found the kernel same if it's here...

jigpu, 2020-04-10:
I would double-check that you have fully updated your system and rebooted so that the latest kernel and headers are installed and the latest kernel is running. DKMS is sometimes difficult to get working properly, and you may need to make sure that it has access to everything that it needs.

Out of curiosity, why are you using the input-wacom driver? The PTK-640 should be supported out-of-the-box by Manjaro, and should work identically to input-wacom for you. You may be better off removing the input-wacom DKMS module. The lack of output from xsetwacom is most likely due to you running a Wayland session or something causing a misconfiguration of Xorg.

Hkyx, 2020-04-12:
thank you for your reply, in fact I got the 5.6 on a new install and got the same problem. I tried all I can like input-wacom drive just due to the documentation on the website but same issue if I remove it.

For the wayland. I use Gnome and the new one is made on wayland, so it's not a choice to use it or not, but if I test installing gnome with xorg, or in my awesomewm which is not with waylanf, get the same error.

jigpu, 2020-04-13:
I would suspect your issue is most likely due to some kind of misconfiguration of the X driver. Could you please run our [sysinfo.sh scritpt](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) from within GNOME-on-Xorg or AwesomeWM and attach the generated tarball here?

Hkyx, 2020-04-14:
just restart my setup this week-end for others stuff, so I tried 
- brand new install, 
- brand new install + kernel 5.6.x
- brand new install + kernel 5.6.x + xf86-input-wacom

same results.
[sysinfo.hNLfPanklq.tar.gz](https://github.com/linuxwacom/input-wacom/files/4476860/sysinfo.hNLfPanklq.tar.gz)

ty for your help,

Hk

jigpu, 2020-04-14:
Thanks for the sysinfo. As I expected, the Manjaro kernel includes a Wacom driver that explicitly lists your 056a:00b9 tablet in its support list. Oddly, however, the sysfs data seems to indicate that the hid-generic driver is powering the driver instead. This creates device nodes that xf86-input-wacom doesn't recognize, which is why it doesn't show up in xsetwacom.

Your kernel log didn't seem to come though in the sysinfo for some reason. Could you run `dmesg > dmesg.log` and then attach the log here? That may have some more information pointing to a failure by the Wacom driver to probe the hardware, or other issues.

Hkyx, 2020-04-14:
of course here is the file. 
[dmesg.log](https://github.com/linuxwacom/input-wacom/files/4477372/dmesg.log)

ty, hk

Hkyx, 2020-04-19:
Hi @jigpu  any chance you got a idea to solve that ?

Hkyx, 2020-04-23:
Solved: 

got the solution from reddit: https://www.reddit.com/r/awesomewm/comments/g6mkp1/help_with_xorgmanjaro/

Did you have installed the linux-headers package? In Manjaro it named with version of the kernel in it, e.g. linux316-headers. You need to install headers according to kernel version which you use.

so solution was: 

- Check your kernel version (uname -a)
- (I use yay so...), yay -Sy linux-headers
- Select the repository in the list with your kernel version (ex: linux56-headers-5.6.6-1)
- reboot

If it still working, check you got all xorg needed packages.

closed 2020-04-23 by Hkyx