linuxwacom/input-wacom #166: Lenovo Thinkpad L13 Yoga pencil issue

Jojeker, 2020-04-16:
Hello!

I recently got my Thinkpad L13 Yoga and I installed Ubuntu 19.10 and Manjaro 19.02 on it.

I am currently using this device for taking notes with Xournal++, but I am running into an issue regarding the pro pen. When drawing with the pen, it randomly draws vertical lines. However one thing I noticed, is the fact that this probably has something to do with the integrated palm rejection.

I attached a photo in order to present what I mean. I know this device is pretty new, so I didn't find may information on issues with it, so please help me with this!

Here some additional information on my system (Ubuntu):

`~$ uname -a

Linux ***-ThinkPad 5.3.0-26-generic #28-Ubuntu SMP Wed Dec 18 05:37:46 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

xinput --list

Virtual core pointer id=2 [master pointer (3)]

⎜ ↳ Virtual core XTEST pointer id=4 [slave pointer (2)]

⎜ ↳ Wacom Pen and multitouch sensor Pen stylus id=11 [slave pointer (2)]

⎜ ↳ Wacom Pen and multitouch sensor Finger id=12 [slave pointer (2)]

⎜ ↳ Elan Touchpad id=14 [slave pointer (2)]

⎜ ↳ Elan TrackPoint id=15 [slave pointer (2)]

⎜ ↳ Wacom Pen and multitouch sensor Pen eraser id=18 [slave pointer (2)]

⎣ Virtual core keyboard id=3 [master keyboard (2)]

↳ Virtual core XTEST keyboard id=5 [slave keyboard (3)]

↳ Power Button id=6 [slave keyboard (3)]

↳ Video Bus id=7 [slave keyboard (3)]

↳ Sleep Button id=8 [slave keyboard (3)]

↳ Integrated Camera: Integrated C id=9 [slave keyboard (3)]

↳ Integrated Camera: Integrated I id=10 [slave keyboard (3)]

↳ Integrated 5M Camera: Integrate id=13 [slave keyboard (3)]

↳ AT Translated Set 2 keyboard id=16 [slave keyboard (3)]

↳ ThinkPad Extra Buttons id=17 [slave keyboard (3)]

~$ xsetwacom --list devices

Wacom Pen and multitouch sensor Pen stylus id: 11 type: STYLUS

Wacom Pen and multitouch sensor Pen eraser id: 18 type: ERASER`

[image]

jigpu, 2020-04-16:
Could you please run our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh) to get a recording of the bug occurring? The best way to do this is to open Xournal, run the script (it will prompt to to install any dependencies if they are missing), and select the pen device. When it starts recording, switch to Xournal with Alt+Tab and then begin writing. When the issue occurs, take a screenshot and then switch back to the terminal with Alt+Tab. Hit CTRL+C to stop recording and then attach both the generated tarball and the screenshot for comparison.

It would also be helpful to run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball here as well.

Jojeker, 2020-04-17:
I did some testing on my machine, encountered the bug some times and recorded it. I have attached the screenshots, the .pdf document (which is nonsense) aswell as the recordings and the sysinfo. I hope to have correctly provided everything you asked for.

[sysinfo.ESTsQTLxqZ.tar.gz](https://github.com/linuxwacom/input-wacom/files/4493699/sysinfo.ESTsQTLxqZ.tar.gz)
[record_1587134274.tar.gz](https://github.com/linuxwacom/input-wacom/files/4493700/record_1587134274.tar.gz)
[Analysis.pdf](https://github.com/linuxwacom/input-wacom/files/4493701/Analysis.pdf)
[image]
[image]

.

jigpu, 2020-04-17:
Thanks for the data. I wasn't able to line up the recording with the picture (maybe you erased it or made a new document?) but I still think I was able to find a few instances of the bug in the first 30k events. The data below is for one of those points. Other buggy points include a jump to (15684, 4581) around T=422.713689, and a jump to (19953, 3638) at T=428.193802.

These points all look pretty similar in the logs, with the jump being associated with the loss of serial number and battery level information. This probably signals that the pen is near the edge of proximity (and indeed, some of the jumps are rapidly followed by an out-of-prox event) and is having trouble being tracked by the sensor. It sounds like it might be the same issue as https://github.com/linuxwacom/xf86-input-wacom/issues/107

I suspect this is more of a hardware problem than a driver problem, but there may be something we can do as the driver to mask the issue. I'll do some thinking...

### HID
~~~
E: 422.213682 18 06 21 db 39 e5 25 dc 0a 11 00 2f e8 51 b0 00 ff 00 00
E: 422.223705 18 06 21 ff 39 ee 25 91 0a 11 00 2f e8 51 b0 00 ff 00 00
E: 422.228676 18 06 21 28 3a f0 25 14 0a 11 00 2f e8 51 b0 00 ff 00 00
E: 422.238707 18 06 21 51 3a e9 25 f6 08 11 00 2f e8 51 b0 00 ff 00 00
E: 422.243654 18 06 21 7a 3a db 25 ea 07 11 00 2f e8 51 b0 00 ff 00 00
E: 422.253652 18 06 21 a4 3a c6 25 78 07 11 00 2f e8 51 b0 00 ff 00 00

E: 422.312707 18 06 21 b4 3a 8e 0e bc 03 00 00 00 00 00 00 00 00 00 00
E: 422.353627 18 06 21 69 3a 89 19 45 08 11 00 2f e8 51 b0 00 00 00 00

E: 422.354609 18 06 21 1e 3a 84 24 74 08 11 00 2f e8 51 b0 00 00 00 00
E: 422.357569 18 06 21 1c 3a 89 24 a1 08 11 00 2f e8 51 b0 00 00 00 00
E: 422.367607 18 06 21 1b 3a 93 24 04 09 11 00 2f e8 51 b0 00 00 00 00
E: 422.372604 18 06 21 1a 3a 9b 24 56 09 11 00 2f e8 51 b0 00 00 00 00
E: 422.382565 18 06 21 1c 3a af 24 99 09 11 00 2f e8 51 b0 00 00 00 00
E: 422.387565 18 06 21 23 3a c5 24 cd 09 11 00 2f e8 51 b0 00 00 00 00
~~~

### Evemu
~~~
E: 422.213726 0003 0000 14811	# EV_ABS / ABS_X                14811
E: 422.213726 0003 0001 9701	# EV_ABS / ABS_Y                9701
E: 422.213726 0003 0018 2780	# EV_ABS / ABS_PRESSURE         2780
E: 422.213726 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.213726 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 422.223763 0003 0000 14847	# EV_ABS / ABS_X                14847
E: 422.223763 0003 0001 9710	# EV_ABS / ABS_Y                9710
E: 422.223763 0003 0018 2705	# EV_ABS / ABS_PRESSURE         2705
E: 422.223763 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.223763 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 422.228744 0003 0000 14888	# EV_ABS / ABS_X                14888
E: 422.228744 0003 0018 2580	# EV_ABS / ABS_PRESSURE         2580
E: 422.228744 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.228744 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 422.238762 0003 0000 14929	# EV_ABS / ABS_X                14929
E: 422.238762 0003 0001 9707	# EV_ABS / ABS_Y                9707
E: 422.238762 0003 0018 2294	# EV_ABS / ABS_PRESSURE         2294
E: 422.238762 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.238762 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 422.243722 0003 0000 14970	# EV_ABS / ABS_X                14970
E: 422.243722 0003 0001 9691	# EV_ABS / ABS_Y                9691
E: 422.243722 0003 0018 2026	# EV_ABS / ABS_PRESSURE         2026
E: 422.243722 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.243722 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 422.253723 0003 0000 15012	# EV_ABS / ABS_X                15012
E: 422.253723 0003 0001 9670	# EV_ABS / ABS_Y                9670
E: 422.253723 0003 0018 1912	# EV_ABS / ABS_PRESSURE         1912
E: 422.253723 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.253723 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms

E: 422.312768 0003 0000 15028	# EV_ABS / ABS_X                15028
E: 422.312768 0003 0001 3726	# EV_ABS / ABS_Y                3726
E: 422.312768 0003 0018 0956	# EV_ABS / ABS_PRESSURE         956
E: 422.312768 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.312768 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +59ms
E: 422.353743 0003 0000 14953	# EV_ABS / ABS_X                14953
E: 422.353743 0003 0001 6537	# EV_ABS / ABS_Y                6537
E: 422.353743 0003 0018 2117	# EV_ABS / ABS_PRESSURE         2117
E: 422.353743 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.353743 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +41ms

E: 422.354726 0003 0000 14878	# EV_ABS / ABS_X                14878
[... 8 lines cut ...]
E: 422.357691 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +3ms
E: 422.367687 0003 0000 14876	# EV_ABS / ABS_X                14876
E: 422.367687 0003 0001 9363	# EV_ABS / ABS_Y                9363
E: 422.367687 0003 0018 2308	# EV_ABS / ABS_PRESSURE         2308
E: 422.367687 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.367687 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 422.372687 0003 0000 14875	# EV_ABS / ABS_X                14875
E: 422.372687 0003 0001 9371	# EV_ABS / ABS_Y                9371
[... cut ...]
E: 422.382681 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 422.387682 0003 0000 14883	# EV_ABS / ABS_X                14883
E: 422.387682 0003 0001 9413	# EV_ABS / ABS_Y                9413
E: 422.387682 0003 0018 2509	# EV_ABS / ABS_PRESSURE         2509
E: 422.387682 0004 0000 -1336809425	# EV_MSC / MSC_SERIAL           -1336809425
E: 422.387682 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
~~~

### First 30k points
Events while pen is up not drawn. Point circled in red corresponds to the events above.
[image]

Jojeker, 2020-06-07:
Hello!

Recently I have been looking around for the problem that was occurring and I found out, that the error was not only affecting Linux users but many Microsoft users as well. So many thought this was related to a hardware error until a week ago there was posted a driver update on the Lenovo driver Webpage issuing a new Wacom driver update, which fixes the following Issue:

> [New functions or enhancements]
- Solve ghost line issue which occurred when hand writing with palm.

The link to the site is: [https://pcsupport.lenovo.com/gb/en/products/laptops-and-netbooks/thinkpad-l-series-laptops/thinkpad-l13-yoga-type-20r5-20r6/downloads/DS544571](url) and the forum (which is a german one): [https://forums.lenovo.com/t5/A-L-SL-und-R-Serie-ThinkPad-Notebooks/L13-Yoga-Stift-springt-und-zieht-ungewollte-Linien/m-p/4621959?page=7](url)

So my question is, is there is a possibility to port this fix to Linux? It would make not only me but also other Linux users very happy to write without artrefacts :)

Thank you in advance!

Best regards,
Eduard

jigpu, 2020-06-09:
It looks like that site has a firmware update to address the issue, which makes sense given that the hardware was sending bogus data. Unfortunately there's nothing we can "port", and I don't believe its possible to run the firmware updater from within Linux. You'll need to temporarily use / install Windows to run the update.

Jojeker, 2020-06-09:
Ok, so this means the only thing I have to do is to update the driver from within Windows? Does it persist even if I remove Windows directly afterwards, because it is overriding the EEPROM or is there a different reason?

I will report how the update was going and tell if everything works afterwards!

I want to thank you very much for your help and time you have invested in helping :+1: :)

martinlanger90, 2020-07-01:
This one occurs on X1 Yoga Gen 5 as well. Tried to reach out the premier support. Until now not really a response. 
Tried with different Windows Versions 1909, 2004. Tried in Ubuntu 20.04, less but not gone. 
Any clue if this is the same issue? (https://drive.google.com/file/d/1uERRlez44dyFV4hS-0g_J9Vd7uyWKaId/view?usp=sharing) I would be glad if someone would watch this (issue appears at the end, it's less often but I would say you can trigger it every minute).

jigpu, 2020-07-02:
@martinlanger90 it certainly has a similar visual appearance, and it appearing under both Windows and Linux is a sign of the problem being with firmware/hardware. Unfortunately there's not much to do but wait for Lenovo to address it...

martinlanger90, 2020-07-03:
Seems really like it's the same as for L13 Yoga.
Recorded another one with pencil and hand visible (though in Windows again): https://drive.google.com/file/d/1ggflygFF7dUgZSu8s0FeGsbG2LtUHfaw/view?usp=sharing

Sadly I cannot reach lenovo anymore in the near future because I'm returning the X1. Hopefully they will fix it in the near future. The X1 Yoga Gen 5 is very good besides this annoying issue.

Jojeker, 2020-07-03:
I looked at your video and indeed it looks very similar to the issue I was running into. For me it fixed the issue with updating the firmware for the Wacom panel which seemed to produce the problem. So I would suggest trying to update the drivers if have not until now, probably there is a fix for it. 
[https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-yoga-5th-gen-type-20ub-20uc/downloads/driver-list](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-yoga-5th-gen-type-20ub-20uc/downloads/driver-list)

And additionally, I would suggest opening a forum on their website so they can at least fix the problem for others, but this depends if you want to wait to wait for it or not. I waited approximately 4 frustrating months until it worked, but now I am happy. 

I hope I could help you a bit.

Best regards,
Jojeker

martinlanger90, 2020-07-03:
Looked at it, no firmware update for that. Already contacted premier support, they are looking into the issue. But I think, that will take a long time for being fixed by lenovo / wacom. At least, they know already about this problem (also wrote a article in their forum, but it's not commented much by other people). I guess nobody has the X1 Yoga Gen 5 bought until now 😄

martinlanger90, 2020-07-05:
How to find any errors in those logs from the capture script? I don't know what to look for in those 3 log files 😃

jigpu, 2020-07-07:
@martinlanger90 Finding errors in the logs isn't terribly straightforward... This shell one-liner will comb through the recording's evemu log file to find changes in X or Y that seem to move "too fast" (a jump of more than 1000 points between two consecutive events), but you'll still need to use the timestamp it prints out to look at the surrounding events to see if it really is an error or not.

`cat <evemu log file> | awk '/E:/{T=$2; K=$3$4; V=$5; STATE[K]=V; if (K=="00000000") {print T, STATE["00030000"], STATE["00030001"], STATE["0001014a"]}}' | awk '{T=$1; X=$2; Y=$3; TIP=$4; dX=X-oldX; dY=Y-oldY; if (NR==1 || TIP==0) {dX=0; dY=0;} oldX=X; oldY=Y; if (dX > 1000) {AXIS="X"} else if (dY > 1000) {AXIS="Y"} else {AXIS=""} if (AXIS!="") {print "Possible jump in " AXIS " at T=" T}}'`