linuxwacom/input-wacom #167: Wacom Intuos S scrolling does not work

LukeLR, 2020-04-20:
I want to use scrolling when pressing a button and moving the pen up or down. `man xsetwacom` says, the `pan` action has this behaviour when bound to a button.

However, `xsetwacom set 14 Button 3 pan` does not work, even if applied to all 4 IDs detected by `xsetwacom --list devices`. Instead, the action applied to the button is empty, as `xsetwacom -s get 14 all` contains `xsetwacom set "14" "Button" "3" ""`. Holding down button 3 does not change the behavior of the pen movements.

What am I missing? Am I doing something wrong, or is this a bug? Thanks in advance!

jigpu, 2020-04-20:
Unfortunately it is not currently possible to use a pad button to trigger the pan function. The driver would need to correlate events across devices (button press on the pad device, pen drag on the stylus device), which can be tricky. This could be a feature request, but we probably won't be able to get around to it for a while :slightly_frowning_face:

jigpu, 2020-04-20:
FYI: I'm moving this to xf86-input-wacom and marking as a feature request so at least it isn't forgotten