linuxwacom/input-wacom #169: Intuos pro S (PTH-460) touch ring center button does nothing

RandomReaper, 2020-04-21:
The center button generates no event (tested with xev).
xserver-xorg-input-wacom 1:0.39.0-0ubuntu1

jigpu, 2020-04-21:
Looks like this is an issue specific to the Bluetooth connection. I believe I've pinned down the problem and pushed a fix to my repository in the "fix-112" branch.

Please install git (`sudo apt install git` or `sudo yum install git`) and clone the development code by running the following command:
~~~
git clone https://github.com/jigpu/input-wacom.git -b fix-112
~~~
Next, follow the [input-wacom build instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source), skipping the "Download" section and instead using the code cloned from git. Once the driver has been built and installed, please either reboot or follow the "Module Loading" instructions to ensure the driver is loaded. Connect your tablet as normal and test to see if you see button events from the center button now.

RandomReaper, 2020-04-22:
You where right, the problem was bluetooth specific. The problem is fixed by this patch.

Nice job, thank you!

jigpu, 2020-04-22:
Thanks -- I'll go ahead and forward the fix off to the Linux kernel for integration.

jigpu, 2021-01-13:
Fixed with commit dcce8ef8f70a8e38e6c47c1bae8b312376c04420 in both stable (v5.3+) and upstream kernel (v5.7+) trees.

closed 2021-01-13 by jigpu