linuxwacom/input-wacom #170: Fast movement up causes no button release event

ClovisPJ, 2020-04-24:
Hi!

So I'm using a Bamboo Ink stylus with my Lenovo Yoga 520.
This has a touchscreen as the display (2in1 laptop), which I want to use in ubuntu.

Unfortunately, I've found an issue.
If the pen is raised too quickly, no button release event is created.
This is a problem as it locks the cursor in pen down mode, which can be quite frustrating.

I am unsure whether it is a problem with the kernel or the x.org driver.
So I've created the issue here, hopefully that is okay.

I've attached all the diagnostic information.
I have updated both 'input-wacom' & 'xf86-input-wacom' from source (GitHub) to the latest version.
As well as a video of the bug (the video and logfile recording are simultaneous).

Hopefully you might be able to help?
Thanks in advance.

[record_1587752881.tar.gz](https://github.com/linuxwacom/input-wacom/files/4530877/record_1587752881.tar.gz)
[sysinfo.m8GuzEfXNS.tar.gz](https://github.com/linuxwacom/input-wacom/files/4530878/sysinfo.m8GuzEfXNS.tar.gz)
[video recording](https://drive.google.com/open?id=1e2PMYbDMB5ZMIOAbv3SFPs66cRjdyL5O)

jigpu, 2020-04-24:
Thanks for the detailed report. Unfortunately, this looks like a hardware issue... When you remove the pen like that, the sensor never sends an event that says the pen left. Our first indication that something is wrong is when the pen finally re-enters proximity, but by then it is far too late.

An issue like this would probably require the kernel to keep track of the time since the last event and force a prox-out event if one doesn't come after a while.

### Relevant HID Events
~~~
E: 4.690877 16 06 21 5f 47 22 2b 52 07 49 00 6d 31 c6 39 00 ff
E: 4.695750 16 06 21 76 47 ff 2a de 06 49 00 6d 31 c6 39 00 ff
E: 8.141198 16 06 20 76 47 ff 2a 00 00 00 00 00 00 00 00 00 00
E: 8.147458 16 06 20 56 42 06 27 00 00 00 00 00 00 00 00 00 00
E: 8.151205 16 06 20 56 42 06 27 00 00 00 00 00 00 00 00 00 00
~~~

### Relevant Kernel Events
~~~
E: 4.690973 0003 0000 18271	# EV_ABS / ABS_X                18271
E: 4.690973 0003 0001 11042	# EV_ABS / ABS_Y                11042
E: 4.690973 0003 0018 1874	# EV_ABS / ABS_PRESSURE         1874
E: 4.690973 0004 0000 969290093	# EV_MSC / MSC_SERIAL           969290093
E: 4.690973 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 4.695803 0003 0000 18294	# EV_ABS / ABS_X                18294
E: 4.695803 0003 0001 11007	# EV_ABS / ABS_Y                11007
E: 4.695803 0003 0018 1758	# EV_ABS / ABS_PRESSURE         1758
E: 4.695803 0004 0000 969290093	# EV_MSC / MSC_SERIAL           969290093
E: 4.695803 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +5ms
E: 8.141231 0003 0018 0000	# EV_ABS / ABS_PRESSURE         0
E: 8.141231 0001 014a 0000	# EV_KEY / BTN_TOUCH            0
E: 8.141231 0004 0000 969290093	# EV_MSC / MSC_SERIAL           969290093
E: 8.141231 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +3446ms
E: 8.147313 0003 0000 16982	# EV_ABS / ABS_X                16982
E: 8.147313 0003 0001 9990	# EV_ABS / ABS_Y                9990
E: 8.147313 0004 0000 969290093	# EV_MSC / MSC_SERIAL           969290093
E: 8.147313 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +6ms
E: 8.151305 0004 0000 969290093	# EV_MSC / MSC_SERIAL           969290093
E: 8.151305 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +4ms
~~~

ClovisPJ, 2020-04-24:
Ah fair enough, that makes sense. Yeah hardware figures.
Thanks for looking into this for me though. I'm super thankful.

closed 2020-04-24 by ClovisPJ