linuxwacom/input-wacom #172: xsetwacom fails to list finger touch device for Intuos Pro M (PTH-651)

chef694, 2020-04-25:
I can't find a way to disable touch for the Intuos Pro M on Fedora 31 (GNOME). Info below

[oven@fedora ~]$ xsetwacom list devices
Wacom Intuos Pro M Pen stylus   	id: 14	type: STYLUS    
Wacom Intuos Pro M Pen eraser   	id: 15	type: ERASER    
Wacom Intuos Pro M Pen cursor   	id: 16	type: CURSOR    
Wacom Intuos Pro M Pad pad      	id: 17	type: PAD       
[oven@fedora ~]$

jigpu, 2020-04-27:
Can you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball here?

chef694, 2020-04-27:
Hello and thanks for the reply, here's the file.

[sysinfo.Ck8ilWcCkk.tar.gz](https://github.com/linuxwacom/input-wacom/files/4542486/sysinfo.Ck8ilWcCkk.tar.gz)

jigpu, 2020-05-13:
Sorry for the delay. It looks like you've manually created an `/etc/X11/xorg.conf.d/71-libinput-overrides-wacom.conf` file that specifically forces the touchpad input to be handled by the libinput driver. You'll either need to delete this file and let xf86-input-wacom handle the touchpad (at which point you can disable touch with `xsetwacom set <touch_id> touch off`) or find some alternate way of disabling through through libinput.

closed 2020-05-19 by jigpu