linuxwacom/input-wacom #175: Pen Detection problem : my tablet's Pen should be CTH-670 but is CTH-470

nyam-hub, 2020-05-01:
Hi,

So, I've spent 6 hours today trying to figure out where the problem was. As I couldn't configure the Pen for my "CTH-470" (as I've refered to it by it's model on the back of the tablet) because it wasn't detected on Wacom Tablet settings. "No stylus" found. 

So, at some point, I've taken a close look at the CTH-470's pen's picture. And it was a different pen. It would seem that I have the Pen of the CTH-670, which would have 2 buttons + eraser, and the Tablet of the CTH-470. 

Maybe this one would be my tablet, and but it's CTH-470/S instead of CTH-470/S4 for the model number.

[https://www.amazon.com/Wacom-Photoshop-Essentials-CTH-470-S4/dp/B00965DLYC/ref=sr_1_4?dchild=1&keywords=cth+470&qid=1588352618&sr=8-4](url)

So when my tablet is plugged in, it's detected as CTH-470, which has no eraser and only one button.

<details>
  <summary>From libwacom/bamboo-16fg-s-pt.tablet : </summary> 

```
# Wacom
# Bamboo Capture
# CTH-470
#
# third generation BambooPT
#
# stylus with one button no eraser (LP170G); 16FG touch
# Pen active area: 5.8 x 3.62in
# Touch active area: 4.92 x 3.35in

[Device]
Name=Wacom Bamboo Capture
ModelName=CTH-470
DeviceMatch=usb:056a:00de
Class=Bamboo
Width=6
Height=4
Layout=bamboo-16fg-s-pt.svg
IntegratedIn=

[Features]
Stylus=true
Reversible=true
Touch=true
Buttons=4

[Buttons]
Left=A;B;C;D
EvdevCodes=0x110;0x115;0x116;0x111
```
</details>

we can see that it's referring to the model of CTH-470 with the pen : "stylus with one button no eraser (LP170G); 16FG touch"

Meanwhile, this one 
<details>
  <summary>from libwacom/bamboo-16fg-m-pt.tablet :</summary> 

 ```
# Wacom
# Bamboo Create
# CTH-670
#
# third generation BambooPT
#
# stylus with two buttons and eraser (LP170ES); 16FG touch
# Pen active area: 8.52 x 5.39 in
# Touch active area: 7.48 x 5.12 in

[Device]
Name=Wacom Bamboo Create
ModelName=CTH-670
DeviceMatch=usb:056a:00df
Class=Bamboo
Width=6
Height=4
Layout=bamboo-16fg-m-pt.svg
IntegratedIn=

[Features]
Stylus=true
Reversible=true
Touch=true
Buttons=4

[Buttons]
Left=A;B;C;D
EvdevCodes=0x110;0x115;0x116;0x111
```

</details>

Refers to a pen : # stylus with two buttons and eraser (LP170ES); 16FG touch

Which I possess. 
But the size of the tablet is different. 

Anyway here's the detection : 
<details>
  <summary>From xsetwacom --list</summary>

> Wacom Bamboo 16FG 4x5 Finger touch	id: 18	type: TOUCH
> Wacom Bamboo 16FG 4x5 Pad pad   	id: 19	type: PAD
> Wacom Bamboo 16FG 4x5 Pen stylus	id: 20	type: STYLUS
> Wacom Bamboo 16FG 4x5 Pen eraser	id: 21	type: ERASER 

</details>

<details>

<summary>From libwacom-list-local-devices : </summary>

> Device node: /dev/input/event9
> [Device]
> Name=Wacom Bamboo Capture
> ModelName=CTH-470
> DeviceMatch=usb:056a:00de;
> Class=Bamboo
> Width=6
> Height=4
> IntegratedIn=
> Layout=bamboo-16fg-s-pt.svg
> Styli=0xffffe;0xfffff;
> 
> [Features]
> Reversible=true
> Stylus=true
> Ring=false
> Ring2=false
> Touch=true
> TouchSwitch=false
> StatusLEDs=
> NumStrips=0
> Buttons=4
> 
> [Buttons]
> Left=A;B;C;D;
>  Right=
> Top=
> Bottom=
>  Touchstrip=
>  Touchstrip2=
>  OLEDs=
> Ring=
> Ring2=
> EvdevCodes=0x110;0x115;0x116;0x111;
> RingNumModes=0
> Ring2NumModes=0
> StripsNumModes=0
> 
> ---------------------------------------------------------------
> Device node: /dev/input/event26
> [Device]
> Name=Wacom Bamboo Capture
> ModelName=CTH-470
> DeviceMatch=usb:056a:00de;
> Class=Bamboo
> Width=6
> Height=4
> IntegratedIn=
> Layout=bamboo-16fg-s-pt.svg
> Styli=0xffffe;0xfffff;
> 
> [Features]
> Reversible=true
> Stylus=true
> Ring=false
> Ring2=false
> Touch=true
> TouchSwitch=false
> StatusLEDs=
> NumStrips=0
> Buttons=4
> 
> [Buttons]
> Left=A;B;C;D;
>  Right=
> Top=
> Bottom=
> Touchstrip=
> Touchstrip2=
> OLEDs=
> Ring=
> Ring2=
> EvdevCodes=0x110;0x115;0x116;0x111;
> RingNumModes=0
> Ring2NumModes=0
> StripsNumModes=0
> 
> ---------------------------------------------------------------
> Device node: /dev/input/event10
> [Device]
> Name=Wacom Bamboo Capture
> ModelName=CTH-470
> DeviceMatch=usb:056a:00de;
> Class=Bamboo
> Width=6
> Height=4
> IntegratedIn=
> Layout=bamboo-16fg-s-pt.svg
> Styli=0xffffe;0xfffff;
> 
> [Features]
> Reversible=true
> Stylus=true
> Ring=false
> Ring2=false
> Touch=true
> TouchSwitch=false
> StatusLEDs=
> NumStrips=0
> Buttons=4
> 
> [Buttons]
> Left=A;B;C;D;
> Right=
> Top=
> Bottom=
> Touchstrip=
> Touchstrip2=
> OLEDs=
> Ring=
> Ring2=
> EvdevCodes=0x110;0x115;0x116;0x111;
> RingNumModes=0
> Ring2NumModes=0
> StripsNumModes=0

</details> 

OS : Ubuntu 18.04 LTS. 

So anyway, how to change the pen of the DeviceMatch=usb:056a:00de ? 

It would just seem that to be honest, you forgot about that version of the CTH-470 or something? Because the pen of my tablet models shouldn't be "stylus with one button no eraser (LP170G); 16FG touch". My tablets model isn't Bamboo Capture but Bamboo Fun something... 

Also, on "Left handed" mode, the touchpad isn't reversed.

jigpu, 2020-05-04:
This is actually a GNOME issue. The CTH-470 and CTH-670 aren't capable of distinguishing between one-button and two-button pens, and don't report any identifying information that our software can use to distinguish them. Our driver creates a "fake" identifier for any pen seen by these devices, and GNOME does not properly handle this fake identifier. The end result is the "No stylus found" screen that you see.

You may have better luck with the most current GNOME 3.36 release which is used by Ubuntu 20.04. I'm not sure if there are any usable workarounds that don't involve upgrading, unfortunately.

closed 2020-05-04 by jigpu

nyam-hub, 2020-05-05:
Okay thanks! 
I'll try to upgrade to 20.04 after the finals.

By any chance, is there a command to activate the eraser with xsetwacom? 
I've tried 
`xsetwacom set "Wacom Bamboo 16FG 4x5 Pen eraser" Button 1 "Button 1"`
or
`xsetwacom set "Wacom Bamboo 16FG 4x5 Pen eraser" Button 1 "Button 3"`
and some other combinations. But it seems that then tip of the stylus doesn't work or act weird (well sometimes it seems like they act similarly). 
(I assume this is related to the issue you've mentioned. But I'm just asking in case I missed a fix during many hours of searching.) 

Thanks in advance.

jigpu, 2020-05-13:
Unfortunately there is currently no way to configure the stylus to send eraser events :(