linuxwacom/input-wacom #177: Lenovo Thinkpad L13 Yoga pen doesn't work at all

talestomaz, 2020-05-04:
Hi community,
I've just bough a Lenovo Thinkpad L13 Yoga and installed Manjaro KDE in it. Everything is working fine except the set touchscreen + pen. The touchscreen was working very well before installing `kcm-wacomtablet`, `libwacom` and `xf86-input-wacom`. Now, whenever I intend to click on something with my finger, the click isn't released, as if I were constantly pressing on the screen to move something. But the worse part is that the pen doesn't work at all. It didn't work before installing those packages and has had no change after it.
This is what `xinput --list` returns:

`⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Elan TrackPoint                           id=10   [slave  pointer  (2)]
⎜   ↳ Elan Touchpad                             id=9    [slave  pointer  (2)]
⎜   ↳ Wacom Pen and multitouch sensor Pen stylus        id=16   [slave  pointer  (2)]
⎜   ↳ Wacom Pen and multitouch sensor Pen eraser        id=17   [slave  pointer  (2)]
⎜   ↳ Wacom Pen and multitouch sensor Finger touch      id=18   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Sleep Button                              id=8    [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=11   [slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                    id=12   [slave  keyboard (3)]
    ↳ Integrated Camera: Integrated I           id=13   [slave  keyboard (3)]
    ↳ Integrated 5M Camera: Integrate           id=14   [slave  keyboard (3)]
    ↳ Integrated Camera: Integrated C           id=15   [slave  keyboard (3)]
`
Any suggestion on how should I begin with it?

talestomaz, 2020-05-04:
Result of running [the sysinfo script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh):
[sysinfo.KaSCn9REzO.tar.gz](https://github.com/linuxwacom/input-wacom/files/4576783/sysinfo.KaSCn9REzO.tar.gz)

jigpu, 2020-05-13:
Didn't see that you'd already filed an issue over here. I'll go ahead and close #182 as a duplicate of this, but please still try my suggestion of installing input-wacom and let me know how it goes.

talestomaz, 2020-05-14:
Thanks @jigpu! I'm kind of a newbie regarding building, so you can correct me and recommend further reading whenever you see I'm doing dumb stuff.

To make things easier, I've made a fresh install of Manjaro KDE and can do it again whenever it's necessary. The laptop is new and I have a /home partition to facilitate reinstallations. My fresh Manjaro install includes no "kcm-wacomtablet", "libwacom" and "xf86-input-wacom" and, as said in the original post, touchscreen works perfectly - though no stylus.

Instead of installing first those packages, I tried to go direct to the newest driver you indicated. Following your instructions, I tried to begin with the dependencies (`sudo pacman -S base-devel linux-headers`). It asked me to select a "member" of the "base-devel" group, giving me the following options: `1) autoconf  2) automake  3) binutils  4) bison  5) fakeroot  6) file  7) findutils  8) flex 9) gawk  10) gcc  11) gettext  12) grep 13) groff  14) gzip  15) libtool  16) m4  17) make  18) pacman  19) patch  20) pkgconf  21) sed  22) sudo  23) texinfo 24) which`. I had no idea what to choose. It said that standard choice would be `all` (`todos` in my version in Portuguese). I inserted "todos" and "all" as answers, but it said this was an invalid answer.

Then I decided to have a look at Pamac and found this "input-wacom-dkms 0.46.0-1". It pointed pretty much to the same page you indicated before (https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) and was last modified 2 days ago. I thought it would likely be the same and simply asked Pamac to install it. It was installed with no evident dependencies and, after reboot, there was no change - touchscreen worked perfectly, stylus didn't work at all.

Then I installed the other three packages ("kcm-wacomtablet", "libwacom" and "xf86-input-wacom"). Reboot. Now also touchscreen didn't work at all. This was the first time I had this behaviour. Further I uninstalled Pamac's "input-wacom-dkms 0.46.0-1". Reboot. Now, with those three packages and no "input-wacom-dkms 0.46.0-1", I have exactly the same behaviour from my original post: broken touchscreen - touch gets stuck - and no stylus working.

I don't know how much this helps, but here is the result of my script after all of these changes: 
[sysinfo.XlKH1J7jB5.tar.gz](https://github.com/linuxwacom/input-wacom/files/4628887/sysinfo.XlKH1J7jB5.tar.gz)

What to do now? Maybe I assumed wrong that the AUR package in Pamac would be the same of your instructions. In this case, any recommendation how to proceed regarding the group choice?

Thanks!

talestomaz, 2020-05-14:
Only a correction: the script I've eventually attached is the one WITH, not without, the "input-wacom-dkms 0.46.0-1" package installed. I assume that the script without it would be very similar to my first one. But, if it helps, I can uninstall "input-wacom-dkms 0.46.0-1" and run the script again.

Behaviour now, having installed "input-wacom-dkms 0.46.0-1" again: absolutely no response to touch.

jigpu, 2020-05-14:
> Instead of installing first those packages, I tried to go direct to the newest driver you indicated. 

Not strictly necessary, but a good precaution :+1: 

> Following your instructions, I tried to begin with the dependencies (`sudo pacman -S base-devel linux-headers`). It asked me to select a "member" of the "base-devel" group, giving me the following options: [snipped]. I had no idea what to choose. It said that standard choice would be `all` (`todos` in my version in Portuguese). I inserted "todos" and "all" as answers, but it said this was an invalid answer.

In this instance (and many others) you can just press the Enter key without typing anything to get the default behavior.

> Then I decided to have a look at Pamac and found this "input-wacom-dkms 0.46.0-1".

The input-wacom-dkms package is indeed based on our driver. It "should" work fine, but we have less control over exactly how things are installed through dkms, so we usually don't recommend it for debugging.

> I don't know how much this helps, but here is the result of my script after all of these changes:
> [sysinfo.XlKH1J7jB5.tar.gz](https://github.com/linuxwacom/input-wacom/files/4628887/sysinfo.XlKH1J7jB5.tar.gz)

Thanks! Given that your correction says that input-wacom-dkms was installed when this was run, I expected to find the dkms version of our driver installed on the system. Unfortunately I don't see it anywhere. This may be due to an error in the dkms build process or something else...

Go ahead and make sure input-wacom-wacom-dkms is uninstalled and then retry the build steps in our wiki. Hopefully they work fine once you get past the base-devel selection :)

talestomaz, 2020-05-14:
> In this instance (and many others) you can just press the Enter key without typing anything to get the default behavior.

Lol... beginner's mistake

> Go ahead and make sure input-wacom-wacom-dkms is uninstalled and then retry the build steps in our wiki. Hopefully they work fine once you get past the base-devel selection :)

Uninstalled input-wacom-dkms via Pamac. Then followed your instructions and apparently everything is installed and loaded. Command `$ grep "" /sys/module/wacom*/version` returns `v2.00-0.46.0`. The behaviour now is the same as in my original post: touch works, but gets stuck; stylus doesn't work. This should help now as well: the result of my screen after all these processes: [sysinfo.XAQ7BBGqiu.tar.gz](https://github.com/linuxwacom/input-wacom/files/4630255/sysinfo.XAQ7BBGqiu.tar.gz)

Thanks again!

jigpu, 2020-05-14:
Hmm, thanks. Unfortunately I still don't see anything out of the ordinary, and its clear that the input-wacom driver is properly loaded, so our earlier touch tracking issue must not be the source of your problem.

Please run our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh) to gather data about what kind of data the hardware and drivers are sending. You'll probably be prompted to install some dependencies. Here are the Manjaro equivalents:

* **xinput:** `sudo pacman -S xorg-xinput`
* **evemu-record:** `sudo pacman -S evemu`
* **hid-replay:** hid-replay-git (from AUR)

Once the dpenOnce the program is running, select the stylus device and then make three short strokes with the pen on the screen. Press CTRL+C to quit recording. Next, run the program again and choose the touch input. Make three short single-finger swipes on the touchscreen. Press CTRL+C to quit recording. Finally, attach both of the generated tarballs to this issue for review.

talestomaz, 2020-05-15:
Great. Installed all dependencies. Run the script both times. First, pen selected: 
[record_1589551475.tar.gz](https://github.com/linuxwacom/input-wacom/files/4634831/record_1589551475.tar.gz).

Now, touch selected: 
[record_1589551519.tar.gz](https://github.com/linuxwacom/input-wacom/files/4634832/record_1589551519.tar.gz).

Is it possible that the problem with the stylus is hardware? This is a brand new PC from Lenovo, unlikely, but I have no other means to test. Don't have Windows here.

jigpu, 2020-06-02:
Sorry for the delay in getting back to you. Unfortunately I don't see any evidence of the touch issue in the logs. Each of the three touches go down, drags, and then is released before the next drag begins. Does this issue always occur for you, or only sometimes?

As for the pen, I see no events recorded from the hardware. Can you double-check that the pen's battery is properly inserted, and -- failing that -- that it has sufficient voltage?

talestomaz, 2020-06-04:
Hi @jigpu thanks. Well, as I've read your message, I've taken the stylus again (for the first time since then) and it turns out that it has worked... I've made some tests and it really seems to work perfectly. Unbelievable... In terms of software, the only change I had since then was a huge Manjaro update. Maybe the stylus was uncharged and, by being kept into the charging spot, it got full. But I cannot see where is a place for battery in this stylus. It seems to be completely sealed.

The touch (with fingers, and not stylus) is still bad. As I said, I'm not able to click on an icon to open the application, because even if I release my finger after the touch, it keeps "touched", pressed. I can record a video and narrate it, if this helps. After I meet some work deadlines, I'll install my Manjaro again from scratch and see how things go. Thanks again!

jigpu, 2020-06-04:
Glad the pen is (mysteriously) working now at least. Any additional info about the touch would help whenever you have the time. If it happens all the time, but the recording didn't see any issue, there's a small chance that the issue is somewhere else in the environment (though I would expect KDE to be pretty solid...). It would be interesting to boot into a Fedora 32 live environment to see if pen and touch work differently there, though of course don't worry about it until you have some spare time :)

talestomaz, 2020-07-15:
Hi @jigpu  I've just made a fresh installation of Manjaro and can confirm that everything works fine out-of-the-box, including touch and pen. No need of extra package, nothing. Fazit: the pen was very likely not charged when I attempted to use it in the first days. Thanks for your help!

closed 2020-07-21 by jigpu