linuxwacom/input-wacom #178: Pen becomes unresponsive shortly after boot (Fujitsu Stylistic ST5112)

ZappaMC, 2020-05-08:
It's an older tablet with an integrated Wacom Penabled screen. Pen works fine during bootup, and a few seconds after entering the desktop, but becomes unresponsive shortly afterwards.
Currently using Lubuntu 19.10 64-bit version - I've tried with 18.04, with 32 and 64 bit.

modinfo wacom | grep version returns:
`version:        v2.00-0.45.0`
`srcversion:     0AA19FE5A3E9C50CD127943`

apt list --installed | grep wacom returns

`WARNING: apt does not have a stable CLI interface. Use with caution in scripts.`
`libwacom-bin/eoan,now 1.1-1 amd64 [installed,automatic]`
`libwacom-common/eoan,now 1.1-1 all [installed,automatic]`
`libwacom2-dbg/eoan,now 1.1-1 amd64 [installed]`
`libwacom2/eoan,now 1.1-1 amd64 [installed,automatic]`
`xserver-xorg-input-wacom/eoan,now 1:0.36.1-0ubuntu2 amd64 [installed,automatic]`

I'm relatively new to Linux, so help on this would be appreciated.

jigpu, 2020-05-13:
Can you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball here so we can get a bit more information about your setup?

If your system boots to a login screen that the pen works fine at, have you checked to see what happens if you stay at the login screen? Does the pen continue to work fine after a few minutes, or does it stop after a few seconds, similar to as if you had logged in?

ZappaMC, 2020-10-24:
Apologies for the huge delay, here's the tarball: [sysinfo.1MI5kwE33N.tar.gz](https://github.com/linuxwacom/input-wacom/files/5433717/sysinfo.1MI5kwE33N.tar.gz)
As for the login screen, if you stay at the login screen I believe that it stops working on the login screen, similar to if you'd logged in.

Pinglinux, 2022-08-05:
Your Xorg.0.log shows you have a 'Serial Wacom Tablet FUJ02e5'. Those devices were supported more than 10 or 15 years ago. Using an older Linux version may get the job done easier...

closed 2022-08-05 by Pinglinux