linuxwacom/input-wacom #180: Wacom Intuos M (CTL-6100WL) works only in Android mode

michaelpasek, 2020-05-12:
I have installed the latest git repo version of input-wacom. My tablet Wacom Intuos CTL-6100WL switches automatically to Android mode when plugged in USB port of Ubuntu 18.04 laptop (as expected from https://github.com/linuxwacom/input-wacom/issues/149#issuecomment-585432688 ). In Android mode the tablet works, but "improperly", and it is recognized by the command $libwacom-list-local-devices.
Holding down buttons 1 and 4 on the tablet for a couple of seconds switches the red light from dim to normal, however the tablet then stops working and does not show up in libwacom command.
In case that's useful, the USB device ID from command $lsusb in Android mode is 2d1f:0378 while normal mode is 056a:0378.

jigpu, 2020-05-13:
Can you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) with the tablet connected and in "normal" mode? You can attach the generated tarball directly to this bug.

michaelpasek, 2020-05-14:
Here it is. [sysinfo.wJViQBnJdQ.tar.gz](https://github.com/linuxwacom/input-wacom/files/4627187/sysinfo.wJViQBnJdQ.tar.gz)

michaelpasek, 2020-05-14:
Ah ok. Actually white LED just means the battery is charged, sorry.

jigpu, 2020-05-14:
Thanks for the logs.

Just to be clear, have you installed input-wacom to make use of the tablet's Bluetooth interface? The tablet should work fine over USB with the stock driver for your system, and if that's sufficient you can just `sudo make uninstall` from within the input-wacom directory to restore the original behavior.

Otherwise, you'll need to do a little more work to get input-wacom working. Specifically, I see the following message in your kernel log after the tablet connects:

~~~
[  220.799504] Lockdown: Loading of unsigned modules is restricted; see man kernel_lockdown.7
~~~

This indicates that the kernel is refusing to load the input-wacom driver because it is not signed with a trusted key and your system currently has secure boot enabled. Your options are to either disable secure boot from within the BIOS, or to generate and enroll a signing key and then rebuild/reinstall the driver. You can find out more, as well as instructions for each option, on our wiki page at https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot.

michaelpasek, 2020-05-15:
Thank you very much for the fast response! 

My goal was indeed to use the tablet in Bluetooth mode. Unfortunately, I had completely overlooked that I chose to enable secure boot back when I installed my Ubuntu. Following the steps at https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot made the tablet work like a charm wirelessly. 

Thanks a lot for the great work on the driver.

closed 2020-05-15 by michaelpasek