linuxwacom/input-wacom #182: Lenovo Thinkpad L13 Yoga pen doesn't work at all

talestomaz, 2020-05-13:
Hi community,
I've just bough a Lenovo Thinkpad L13 Yoga and installed Manjaro KDE in it. Everything is working fine except the set touchscreen + pen. The touchscreen was working very well before installing kcm-wacomtablet, libwacom and xf86-input-wacom. Now, whenever I intend to click on something with my finger, the click isn't released, as if I were constantly pressing on the screen to move something. This makes touchscreen useless. Besides, the pen doesn't work at all. It didn't work before installing those packages and has had no change after it.

I've run the script suggested here and have attached the file below. Any advice how to fix the issue? Thanks!

[sysinfo.KaSCn9REzO.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/4608900/sysinfo.KaSCn9REzO.tar.gz)

jigpu, 2020-05-13:
I've moved this to "input-wacom" since this sounds like a kernel issue. I've reviewed the sysinfo that you've provided and see no obvious issues with device creation or installed software. Your pen issue is probably caused by events not being sent by some part of the stack. If its having trouble tracking the state of your fingers (touches being "stuck" like you mentioned), that could possibly cause such behavior.

My first recommendation would be to install this input-wacom driver, following [these instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source). After rebooting see if there has been any change in behavior. The most recent driver release includes some fixes for touch tracking, which may be helpful.

jigpu, 2020-05-13:
Closing as duplicate of #177 (sorry -- mea culpa!)

closed 2020-05-13 by jigpu