linuxwacom/input-wacom #183: Unknown device_type for 'Wacom Co.,Ltd. Pen and multitouch sensor'. Ignoring.

aarjaneiro, 2020-05-14:
I seem to run into an issue during bootup. Calling  `dmesg | grep -i wacom,` I recieve

```
[    2.823675] usb 1-10: Manufacturer: Wacom Co.,Ltd.
[    2.824990] hid-generic 0003:056A:0094.0001: hiddev96,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input0
[    3.182184] wacom 0003:056A:0094.0001: Unknown device_type for 'Wacom Co.,Ltd. Pen and multitouch sensor'. Ignoring.
```
Moreover, lsusb tells me `Bus 001 Device 005: ID 056a:0094 Wacom Co., Ltd Pen and multitouch sensor` is an input whereas `libinput | grep wacom` gives no results. I currently have no pen or multitouch (or any touch) functionality and have tried installing the xf86-input-wacom drivers from source using the standard procedure outlined in the installation instructions (the above error was given after this installation).

Edit: I should of course mention that it is not a usb device, rather it is my laptop's touchscreen + pen.
Edit 2: Also, `#define BUILD_VERSION "0.39.0"` is my config-ver result.

jigpu, 2020-05-14:
Please follow the instructions at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source to try installing our input-wacom kernel driver. We have recently fixed an "unknown device_type" issue which may be the same as yours.

If that does not seem to work, you can also run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) to gather additional logs and hardware information. Simply attach the generated tarball directly to this issue.

aarjaneiro, 2020-05-16:
I tried the instructions but still no initialization. I've attached the file in a public repo, aarjaneiro/sysinfotarball, so I can delete it after the issue is resolved and any necessary diagnostics are retained (for obvious privacy reasons).

aarjaneiro, 2020-05-18:
More potentially useful information: 
`    |__ Port 10: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 056a:0094 Wacom Co., Ltd 
`
is given from lsusb -t -v. Reading online of people with similar problems, it seems possible my device is mislabeled. Could that be the case?

jigpu, 2020-06-02:
It looks like the 056a:0094 sensor isn't explicitly supported by the driver. When this happens we fall back to using the HID descriptor to set up the device. Unfortunately, I suspect that your particular sensor doesn't use a HID descriptor given the error, the ID number (which is similar to other non-HID sensors like 056a:0090, 056a:0093, and 056a:0097), and other info from the logs.

I've added an explicit entry for this sensor on the theory that its probably very similar to the others with nearby PIDs and pushed the code up to my repository. Please install git and clone the branch by running `git clone https://github.com/jigpu/input-wacom.git -b isdv4-0094`. Then build, install, and this modified version of the driver as per the previous instructions.

Also, if you have any suggestions about ways I can improve the anonymization pass done by the sysinfo script, please feel free to [open a wacom-hid-descriptors issue](https://github.com/linuxwacom/wacom-hid-descriptors/issues).

jigpu, 2021-01-13:
Any update on this issue? Did my previous patch solve your problem? If so, I'd like to send it up the kernel mailinglist so that others can benefit from it as well.

aarjaneiro, 2021-01-13:
@jigpu 
Apologies -- it seems to actually not have worked in my case. At this point, I think it is likely some hardware issue as this is not a common problem for other users of my system.

jigpu, 2021-01-25:
This is the first time I've heard of this particular sensor, so I wouldn't be too quick to dismiss it as a hardware issue. It isn't uncommon for variants of a particular tablet PC to ship with different sensors, and since the driver doesn't know about 056a:0094 it isn't unexpected that your device doesn't work.

That said, it _is_ strange that the patch didn't seem to help at all. If you still have the test branch installed, it would be useful to confirm if the driver is properly loaded by running `cat /sys/module/wacom/version`.

aarjaneiro, 2021-01-25:
> This is the first time I've heard of this particular sensor, so I wouldn't be too quick to dismiss it as a hardware issue. It isn't uncommon for variants of a particular tablet PC to ship with [...]

I reckon I might of meddled improperly with the firmware a while back thereby causing the issue (either that or I truly do have a weirdly-unique sensor). 

As for the version:
`>>cat /sys/module/wacom/version
v2.00
`

jigpu, 2021-01-26:
> I reckon I might of meddled improperly with the firmware a while back thereby causing the issue (either that or I truly do have a weirdly-unique sensor).

Hard to say for sure. We don't have an authoratitive list of all IDs, so its always possible that we're missing some. That said, 0094 probably would have been released ages ago (random guess: 7+) so it is a little weird that we haven't seen it before.

> As for the version:

That indicates you're currently using the stock kernel driver. It could be that the kernel got updated in the time since you tested input-wacom (since a new version has to be compiled for each kernel). If you could try compiling and testing again, the version should be `v2.00-0.46.0.1.ga5c2fcd` when properly installed. If you still get the "Unknown device_type" message with that version reported, then I'd suspect something else is going on.

Pinglinux, 2022-08-05:
I am not aware of an ID 0094. We can't shoot in the dark...

closed 2022-08-05 by Pinglinux