linuxwacom/input-wacom #184: Cintiq 13HD, Raspbian Buster: Cursor stuck, jumps when lifting pen

neutr0nis, 2020-05-15:
Hello,

I have connected my Cintiq 13 HD (without touch) to my Raspberry Pi 2B running Raspbian Buster. However, I seem to have a peculiar issue: The cursor is stuck and does not react to the pen, it will only jump to the correct position if I approximate or lift the pen quickly. I think the Wacom kernel module is loaded, at least it is listed when issuing "lsmod".
I have also installed this package from the official Raspbian repository: xserver-xorg-input-wacom/stable,now 0.34.99.1-1+b6 armhf
I think it might be because "usb-devices" says that driver "usbhid" is used for my tablet? I don't know how to change it, though.

```
T:  Bus=01 Lev=03 Prnt=09 Port=00 Cnt=01 Dev#= 11 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=16 #Cfgs=  1
P:  Vendor=056a ProdID=0304 Rev=01.00
S:  Manufacturer=Wacom Co.,Ltd.
S:  Product=Wacom Cintiq 13HD
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
```

Here is the file generated by sysinfo.sh: 
[sysinfo.cs7rYIC1ML.tar.gz](https://github.com/linuxwacom/input-wacom/files/4636652/sysinfo.cs7rYIC1ML.tar.gz)

I would be very thankful if somebody could help me, I've spent hours but don't know what to try anymore.

Best regards,
neutr0nis

neutr0nis, 2020-05-16:
Okay, I think I just found the solution here: https://www.raspberrypi.org/forums/viewtopic.php?t=78805#p1113411
I added usbhid.mousepoll=0 to the line in /boot/cmdline.txt and rebooted my Raspberry. And voilà, the cursor followed my pen as if nothing had been wrong at all.

I'll close my issue now, but hope this hint may help others with the same problem.

closed 2020-05-16 by neutr0nis

jigpu, 2020-05-19:
How strange. It looks like the default "should" be 0, but its possible that something else in Raspbian overrides it. If you don't mind satisfying my curiosity, could you temporarily remove the `usbhid.mousepoll=0` command line option, reboot, and then run `systool -m usbhid -A mousepoll` to check Rasbpian's default value?

neutr0nis, 2020-05-20:
Hi jigpu, I'm glad to contribute, this is my output:
```
root@raspberrypi:/home/pi# systool -m usbhid -A mousepoll
Module = "usbhid"

    mousepoll           = "4294967295"
```
What a strange value...

jigpu, 2020-05-20:
Interesting... Looks like that value indicates that the device should have its polling interval capped at 16 ms (60 Hz). It was introduced in this commit: https://github.com/raspberrypi/linux/commit/b801e0d0f3bda39128cd3d6a755b5a2a01191604 and is specific to the Raspberry Pi's kernel; a similar change isn't in the upstream kernels...

I suspect the problem is a result of the device not being able to handle a polling interval greater than the rate at which it would normally generate events (~7ms for your Cintiq). I doubt the firmware team ever tested to see if the device works properly with a different polling interval since the OS should typically poll the device at the rate specified by the hardware.