linuxwacom/input-wacom #186: Jumping / erratic cursor after turning off/on 27QHD ref #104

n3xxt, 2020-05-22:
Dear Supporter,

ref
https://github.com/linuxwacom/input-wacom/issues/104

In order to make Wacom recognize Pen Stylus/Eraser movements - **switched OFF** Wacom Cintiq 27QHD , waited 5-7 seconds, **then switched ON** (due to issue https://github.com/linuxwacom/input-wacom/issues/104).

**After that the cursor is randomly jumping / erratically moving** when using or not using mouse, typing on USB keyboard, or idle.

Happens with and/or without touching the Wacom surface with Pro pen.

[Which makes it quite annoying experience.]

To exclude possible bluetooth interference - all bluetooth devices are off.

Currently stuck with this cursor jumping issue as the system renders unusable.

Thank you for paying attention!

n3xxt, 2020-05-26:
Cannot replicate the issue on Debian 10.4 [after monitor off/on reset]

[sysinfo.CSUk49KvBs.tar.gz](https://github.com/linuxwacom/input-wacom/files/4683034/sysinfo.CSUk49KvBs.tar.gz)

$ `uname -a`
```
Linux x 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2 (2020-04-29) x86_64 GNU/Linux
```
$ `cat /etc/os-release`
```
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

$ `cat /etc/debian_version `
`10.4`

jigpu, 2020-06-02:
How "often" do these jumps occur? For example, does the cursor jump continuously or only every few seconds/minutes? Also, does the issue only occur immediately after turning the Cintiq back on, or does it usually have a delay before starting (obviously that might be hard to say in some cases)? Finally, are the cursor jumps one-off events (e.g. cursor randomly jumps once to another part of the screen), or does it jump around several times rapidly before settling down until another round of jumps?

I'm interested in getting logs of the hardware and X events at the time the jump occurs but that could be tricky to capture depending on exactly when and how the jumps manifest themselves...

n3xxt, 2020-06-04:
>How "often" do these jumps occur? For example, does the cursor jump continuously or only every few seconds/minutes?

It happens seemingly random - no specific time / action pattern; it might happen while typing on an USB keyboard, moving mouse pointer with USB mouse device and/or Pro Pen, and/or while idle.

>Also, does the issue only occur immediately after turning the Cintiq back on, or does it usually have a delay before starting (obviously that might be hard to say in some cases)?

It happens immediately after turning the Wacom Cintiq 27QHD device back on both physically and/or using a bash USB auth script, as soon as the attention span is over the working area, it does indeed immediately become apparent that the mouse pointer has sporadic patternless undesired movements.

>Finally, are the cursor jumps one-off events (e.g. cursor randomly jumps once to another part of the screen), or does it jump around several times rapidly before settling down until another round of jumps?

It is random: it can jump to the middle of the screens (in the middle of combined area of the both monitors), it might then jump to an Y delta, an X delta or both relative to the current position of the cursor, always random.

After suspicion that mouse cursors jumps are caused by input events, I have left the machine idle from any human input and was able to observe random pointer moves as well.

>I'm interested in getting logs of the hardware and X events at the time the jump occurs but that could be tricky to capture depending on exactly when and how the jumps manifest themselves...

Would you need all input events binary feeds or which particular logs? Please elaborate, I will try to reproduce this issue with a bootable live version of the MX Linux 19.1 xfce and get back to you with required data.

Thank you for paying attention.

Pinglinux, 2022-07-25:
Do you have any other monitors or electronic devices close to the Citniq? It sounds like some kind electronic interference. Please move any of those devices away from the Cintiq or test it with only the minimal accessories attached to the system to see if that makes a difference

Pinglinux, 2022-08-15:
Closing the issue due to no follow-up feedbacks.

closed 2022-08-15 by Pinglinux