linuxwacom/input-wacom #187: xsetwacom get .. shows unknown errors ref #104

n3xxt, 2020-05-22:
Dear Supporter,

When checking properties per Wacom devices

$ `xsetwacom list`

```
Wacom Express Key Remote Pad pad	id: 14	type: PAD       
Wacom Cintiq 27QHD Pen stylus   	id: 15	type: STYLUS    
Wacom Cintiq 27QHD Pen eraser   	id: 21	type: ERASER    
al@x:~/sh
```

errors like these (never seen such errors on Jessie / Stretch) appear:

```
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
RelWheelUp: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Property 'Wacom Wheel Buttons' does not exist on device.
```

# PAD

$ `xsetwacom get 14 all`

```
Property 'Wacom Tablet Area' does not exist on device.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Option "DebugLevel" "0"
Option "CommonDBG" "0"
Option "Suppress" "2"
Option "RawSample" "4"
Property 'Wacom Pressurecurve' does not exist on device.
Option "Mode" "Absolute"
Property 'Wacom Hover Click' does not exist on device.
Option "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
Option "Gesture" "off"
Option "ZoomDistance" "0"
Option "ScrollDistance" "0"
Option "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
Property 'Wacom Rotation' does not exist on device.
RelWheelUp: Actions are not supported by xorg.conf. Try shell format (-s) instead.
RelWheelDown: Actions are not supported by xorg.conf. Try shell format (-s) instead.
AbsWheelUp: Actions are not supported by xorg.conf. Try shell format (-s) instead.
AbsWheelDown: Actions are not supported by xorg.conf. Try shell format (-s) instead.
AbsWheel2Up: Actions are not supported by xorg.conf. Try shell format (-s) instead.
AbsWheel2Down: Actions are not supported by xorg.conf. Try shell format (-s) instead.
StripLeftUp: Actions are not supported by xorg.conf. Try shell format (-s) instead.
StripLeftDown: Actions are not supported by xorg.conf. Try shell format (-s) instead.
StripRightUp: Actions are not supported by xorg.conf. Try shell format (-s) instead.
StripRightDown: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Option "Threshold" "0"
Option "Serial" "0"
Property 'Wacom Pressure Recalibration' does not exist on device.
```

# STYLUS

$ `xsetwacom get 15 all` 

```
Option "Area" "400 400 119740 67520"
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Option "DebugLevel" "0"
Option "CommonDBG" "0"
Option "Suppress" "2"
Option "RawSample" "4"
Option "PressCurve" "0 0 100 100"
Option "Mode" "Absolute"
Option "TPCButton" "off"
Option "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
Option "Gesture" "off"
Option "ZoomDistance" "0"
Option "ScrollDistance" "0"
Option "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
Option "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Option "Threshold" "26"
Option "Serial" "0"
Option "PressureRecalibration" "on"
```

# ERASER

$ `xsetwacom get 21 all`

```
Option "Area" "400 400 119740 67520"
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Option "DebugLevel" "0"
Option "CommonDBG" "0"
Option "Suppress" "2"
Option "RawSample" "4"
Option "PressCurve" "0 0 100 100"
Option "Mode" "Absolute"
Property 'Wacom Hover Click' does not exist on device.
Option "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
Option "Gesture" "off"
Option "ZoomDistance" "0"
Option "ScrollDistance" "0"
Option "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
Option "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Option "Threshold" "26"
Option "Serial" "0"
Option "PressureRecalibration" "on"
```

Thank you for paying attention!

n3xxt, 2020-05-26:
# Same issue on Debian 10.4

[sysinfo.CSUk49KvBs.tar.gz](https://github.com/linuxwacom/input-wacom/files/4683034/sysinfo.CSUk49KvBs.tar.gz)

Hack01 - doesn't work;
Cursor doesn't jump after switching Off/On the Wacom device

$ `uname -a`
```
Linux x 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2 (2020-04-29) x86_64 GNU/Linux
```
$ `cat /etc/os-release`
```
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

$ `cat /etc/debian_version `
`10.4`

n3xxt, 2020-05-26:
Exactly the same output on Debian 10.4 "Buster"

[sysinfo.CSUk49KvBs.tar.gz](https://github.com/linuxwacom/input-wacom/files/4683034/sysinfo.CSUk49KvBs.tar.gz)

$ `uname -a`
```
Linux x 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2 (2020-04-29) x86_64 GNU/Linux
```
$ `cat /etc/os-release`
```
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

$ `cat /etc/debian_version `
`10.4`

jigpu, 2020-06-01:
This are all expected warnings. You can try using the shell format instead to resolve most (but not all) of them. Some of these 

## Warning 1
~~~
XXXXX: Actions are not supported by xorg.conf. Try shell format (-s) instead.
~~~
Button, wheel, and strip properties store what are called "actions". Actions allow you to define a sequence of key presses, mouse clicks, etc. that the driver should send whenever the button/wheel/strip is triggered. For example, you might want to define an action like CTRL+MousewheelUp to cause an application to zoom in. You would do this running a command like `xsetwacom set <id> button <n> key +control button 4 key -control`. In response, xsetwacom would program the property for button N to be a list of the following actions:
* Press the CTRL key
* Press mouse button 4
* Release mouse button 4
* Release the CTRL key
This ability to set actions is quite powerful, but is only possible through xsetwacom. The xorg.conf syntax does have something similar to change the button number of a button/wheel/strip, but it works quite differently under the hood.

When you request xsetwacom list "all" properties for the device, it defaults to printing them in xorg.conf format. Since you can't set actions inside xorg.conf, when it comes across a property that uses actions, it instead prints the warning message and suggests you use shell format instead. Doing so will print out xsetwacom commands, and since those support actions you won't see this error.

## Warning 2
~~~
Property 'XXXXXX' does not exist on device.
~~~
The driver does not set up every device with every possible property. Some properties don't make sense (e.g. pressure recalibration on the pad) and so are never initialized. Xsetwacom doesn't know if any particular property "should" exist on a given device, so it will blindly make a request and print out this property if its missing. When you request xstwacom list "all" the properties, some of these warnings will inevitably printed as it goes through the complete list and stumbles across the ones which aren't relevant.

Note that for historical reasons some nonsencical properties _are_ initialized on every device (e.g. the touch property on the stylus), but its an exception rather than the rule.

closed 2020-06-01 by jigpu