linuxwacom/input-wacom #188: Set MapToOutput when Wacom is switched on?

eezacque, 2020-05-26:
Currently, I need to run xsetwacom to properly configure my Wacom whenever I switch it on.
I would like to do this automatically. However, I cannot use a udev rule for this, as it is triggered by linux discovering the device, before X sees it.

How do I solve this?

jigpu, 2020-06-01:
I've quickly thrown together a small script that should do what you want. See https://gist.github.com/jigpu/1fad061de7e84c2a1ee3208e1a971342 for the source. When added as one of your window manager's startup scripts, it monitors the Xorg log file for hotplug events and then executes a second copy of itself any time one is found to give you the chance to react.

closed 2020-06-01 by jigpu

eezacque, 2020-06-02:
This seems to work. Thanks for sorting this out!