linuxwacom/input-wacom #189: Centos 8.1 Install.

Harpwind, 2020-05-29:
I have a fresh install for Centos 8.1.   I've installed the dependencies successfully.  When I try to build the driver, I get the following:   

 checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking if build path '/home/harpist/Downloads/input-wacom-0.46.0' has spaces... no
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/4.18.0-147.8.1.el8_1.x86_64/source
checking kernel version... 4.18.0-147.8.1.el8_1.x86_64
checking RHEL6 minor release... 0
checking RHEL7 minor release... 0
checking power supply version... pre-v4.1
checking abs accessor version... pre-v2.6.36
checking CONFIG_MODULE_SIG... y
checking CONFIG_MODULE_SIG_FORCE... 
checking CONFIG_LOCK_DOWN_KERNEL... y
checking CONFIG_LOCK_DOWN_IN_EFI_SECURE_BOOT... 
checking secure boot state... off
checking if modules must be signed... recommended
checking CONFIG_MODULE_SIG_HASH... sha256
checking for module signing hash algorithm... sha256
checking for kernel autogenerated key at /lib/modules/4.18.0-147.8.1.el8_1.x86_64/source/certs/signing_key.pem... no
checking for kernel autogenerated cert at /lib/modules/4.18.0-147.8.1.el8_1.x86_64/source/certs/signing_key.x509... no
checking for shim MOK key at /var/lib/shim-signed/mok/MOK.priv... no
checking for shim MOK cert at /var/lib/shim-signed/mok/MOK.der... no
checking for rEFInd MOK key at /etc/refind.d/keys/refind_local.key... no
checking for rEFInd MOK cert at /etc/refind.d/keys/refind_local.cer... no
checking for module signing key... 
checking for module signing certificate... 
configure: WARNING: Kernel module signing is recommended but we are unable to find signing keys. Consider re-running configure with the options '--with-signing-key=<key>' and '--with-signing-cert=<cert>' set.

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.32/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/4.18.0-147.8.1.el8_1.x86_64/source

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/harpist/Downloads/input-wacom-0.46.0'
Making all in 4.5
make[2]: Entering directory '/home/harpist/Downloads/input-wacom-0.46.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/4.18.0-147.8.1.el8_1.x86_64/source M=/home/harpist/Downloads/input-wacom-0.46.0/4.5
make[3]: Entering directory '/usr/src/kernels/4.18.0-147.8.1.el8_1.x86_64'
Makefile:977: *** "Cannot generate ORC metadata for CONFIG_UNWINDER_ORC=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel".  Stop.
make[3]: Leaving directory '/usr/src/kernels/4.18.0-147.8.1.el8_1.x86_64'
make[2]: *** [Makefile:40: all] Error 2
make[2]: Leaving directory '/home/harpist/Downloads/input-wacom-0.46.0/4.5'
make[1]: *** [Makefile:376: all-recursive] Error 1
make[1]: Leaving directory '/home/harpist/Downloads/input-wacom-0.46.0'
make: *** [Makefile:307: all] Error 2
Build Failed

Harpwind, 2020-05-31:
Also:

The touch screen works at login but stops working once GNOME loads.   

And in case it is relevant, I previously had ubuntu 18.04 and 20.04 working without issues.   So I assume it is something with RHEL/Centos causing my issue.

jigpu, 2020-06-01:
Try installing the `elfutils-libelf-devel` package as mentioned in the error message. I believe that will fix the issue.

Harpwind, 2020-06-02:
Sorry,

I should learn to read better.  I missed that in the error message.

I installed elfutils as suggested, I then installed input-wacom, which went without errors, but it still can't find my tablet.   

I did find the the touchscreen actually does work as GNOME is loading, I am able to move the mouse pointer as the desktop appears, but approx 5 seconds later, it cease to work.

Not sure where to go from here.

jigpu, 2020-06-02:
Can you please run the [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the results here? It will contain information about your hardware and drivers that help us better understand the current situation.

Harpwind, 2020-06-02:
[sysinfo.UkJ1SgaCsH.tar.gz](https://github.com/linuxwacom/input-wacom/files/4717632/sysinfo.UkJ1SgaCsH.tar.gz)

Also, just figured out that touching my screen changes the brightness up if it is less than 100%

Harpwind, 2020-06-02:
[sysinfo.cMUUsK7PMx.tar.gz](https://github.com/linuxwacom/input-wacom/files/4717675/sysinfo.cMUUsK7PMx.tar.gz)

That should work better

Harpwind, 2020-06-02:
Also, in case it helps, the hardware is a Mobile Studio Pro 16 (2016)

jigpu, 2020-06-02:
Thanks. There doesn't seem to be an obvious cause for the disconnect so I wonder if something is interfering with the X driver's ability to communicate with the kernel device. Is it possible to boot into a live environment of another distribution (Fedora 32 would be ideal) and check if the touchscreen works or has the same issue there?

<details><summary>Additional details</summary>

The touchscreen initially connects and a kernel device is created:
~~~
[    2.694011] usb 1-6.3: new full-speed USB device number 5 using xhci_hcd
[    2.783864] usb 1-6.3: New USB device found, idVendor=056a, idProduct=034b, bcdDevice= 1.01
[    2.783866] usb 1-6.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.783867] usb 1-6.3: Product: Wacom MobileStudio Pro 16 Touch
[    2.783868] usb 1-6.3: Manufacturer: Wacom Co.,Ltd.      
[    2.789487] input: Wacom Co.,Ltd.       Wacom MobileStudio Pro 16 Touch Touchscreen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3:1.0/0003:056A:034B.0001/input/input5
[    2.789578] hid-generic 0003:056A:034B.0001: input,hiddev96,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       Wacom MobileStudio Pro 16 Touch] on usb-0000:00:14.0-6.3/input0
[    4.138977] input: Wacom MobileStudio Pro 16 Touch Finger as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3:1.0/0003:056A:034B.0001/input/input8
[    4.139235] wacom 0003:056A:034B.0001: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       Wacom MobileStudio Pro 16 Touch] on usb-0000:00:14.0-6.3/input0
~~~

Later on the login screen's X server creates a device:
~~~
[    17.677] (II) config/udev: Adding input device Wacom MobileStudio Pro 16 Touch Finger (/dev/input/event5)
[    17.677] (II) Using input driver 'wacom' for 'Wacom MobileStudio Pro 16 Touch Finger'
[    17.742] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3:1.0/0003:056A:034B.0001/input/input8/event5"
[    17.742] (II) XINPUT: Adding extended input device "Wacom MobileStudio Pro 16 Touch Finger touch" (type: TOUCH, id 14)
~~~

Followed by the user's X server:
~~~
[    36.007] (II) config/udev: Adding input device Wacom MobileStudio Pro 16 Touch Finger (/dev/input/event5)
[    36.007] (II) Using input driver 'wacom' for 'Wacom MobileStudio Pro 16 Touch Finger'
[    36.007] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3:1.0/0003:056A:034B.0001/input/input8/event5"
[    36.008] (II) XINPUT: Adding extended input device "Wacom MobileStudio Pro 16 Touch Finger touch" (type: TOUCH, id 14)
~~~

A short time later the user's X server seems to have trouble reading from the device and so destroys the X device:
~~~
[    45.135] (EE) Wacom MobileStudio Pro 16 Touch Finger touch: Error reading wacom device : No such device
[    45.136] (II) config/udev: removing device Wacom MobileStudio Pro 16 Touch Finger touch
[    45.137] (II) UnloadModule: "wacom"
[    45.137] (II) systemd-logind: releasing fd for 13:69
[    45.162] (II) config/udev: removing device Wacom MobileStudio Pro 16 Pad pad
[    45.164] (II) UnloadModule: "wacom"
[    45.164] (II) systemd-logind: releasing fd for 13:71
[    45.178] (II) config/udev: removing device Wacom MobileStudio Pro 16 Pen stylus
[    45.179] (II) UnloadModule: "wacom"
[    45.179] (II) systemd-logind: not releasing fd for 13:70, still in use
[    45.179] (II) config/udev: removing device Wacom MobileStudio Pro 16 Pen eraser
[    45.180] (II) UnloadModule: "wacom"
[    45.180] (II) systemd-logind: releasing fd for 13:70
~~~

The USB device does eventually disconnect a few minutes later, but it immediately reconnects and I see no evidence of this disconnect/reconnect in the Xorg logs:

~~~
[  392.437857] usb 1-6.3: USB disconnect, device number 5
[  392.622576] usb 1-6.3: new full-speed USB device number 10 using xhci_hcd
[  392.712591] usb 1-6.3: New USB device found, idVendor=056a, idProduct=034b, bcdDevice= 1.01
[  392.712592] usb 1-6.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  392.712593] usb 1-6.3: Product: Wacom MobileStudio Pro 16 Touch
[  392.712594] usb 1-6.3: Manufacturer: Wacom Co.,Ltd.      
[  392.719215] input: Wacom MobileStudio Pro 16 Touch Finger as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3:1.0/0003:056A:034B.000E/input/input37
[  392.719440] wacom 0003:056A:034B.000E: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd.       Wacom MobileStudio Pro 16 Touch] on usb-0000:00:14.0-6.3/input0
~~~
</p>
</details>

As for the brightness issue, I've noticed that GNOME is sometimes a bit too eager to change the backlight level in response to environmental changes. Even just waving your hand over the device is sufficient to cause the display to brighten and dim in some cases. You can disable GNOME's automatic backlight control by opening the Control Center, navigating to the "Power" section, and turning off the automatic backlight control.

Harpwind, 2020-06-02:
So I have booted from the Fedora 32 USB and everything works great, just as it did in Ubuntu 20.04, sees both my styluses.    Also, if it helps, when Using Centos I used the Server with GUI install and with Fedora I did the workstation install.

jigpu, 2020-06-02:
Thanks. That tells me the hardware isn't a problem, so there's something specific to CentOS 8.1 that is causing issues for the hardware. Could you reboot back into CentOS, run `sudo libinput debug-events | tee debug-events.log` and then touch the screen a few times? Afterwards, hit CTRL+C and attach the log file here for review. I want to see if the kernel device still exists and works since the kernel logs didn't show any issues...

Harpwind, 2020-06-02:
And for the record, I started from a fresh install again (thats what took so long)

jigpu, 2020-06-02:
Very interesting. Either the kernel device is missing, or it has gone silent. The command should have printed out a list of devices immediately after starting; can you please include that here?

jigpu, 2020-06-02:
There should be "DEVICE_ADDED" lines before the "POINTER_MOTION" lines, one for each device detected. E.g., I see the following on my system when moving the mouse:
~~~
$ sudo libinput debug-events
-event1   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k
-event0   DEVICE_ADDED     Power Button                      seat0 default group2  cap:k
-event10  DEVICE_ADDED     HDA NVidia HDMI/DP,pcm=9          seat0 default group3  cap:
-event7   DEVICE_ADDED     HDA NVidia HDMI/DP,pcm=3          seat0 default group3  cap:
-event8   DEVICE_ADDED     HDA NVidia HDMI/DP,pcm=7          seat0 default group3  cap:
-event9   DEVICE_ADDED     HDA NVidia HDMI/DP,pcm=8          seat0 default group3  cap:
-event3   DEVICE_ADDED     HDA NVidia HDMI/DP,pcm=3          seat0 default group3  cap:
-event4   DEVICE_ADDED     HDA NVidia HDMI/DP,pcm=7          seat0 default group3  cap:
-event5   DEVICE_ADDED     HDA NVidia HDMI/DP,pcm=8          seat0 default group3  cap:
-event6   DEVICE_ADDED     HDA NVidia HDMI/DP,pcm=9          seat0 default group3  cap:
-event11  DEVICE_ADDED     HDA Intel PCH Front Mic           seat0 default group3  cap:
-event12  DEVICE_ADDED     HDA Intel PCH Rear Mic            seat0 default group3  cap:
-event13  DEVICE_ADDED     HDA Intel PCH Line                seat0 default group3  cap:
-event14  DEVICE_ADDED     HDA Intel PCH Line Out Front      seat0 default group3  cap:
-event15  DEVICE_ADDED     HDA Intel PCH Line Out Surround   seat0 default group3  cap:
-event16  DEVICE_ADDED     HDA Intel PCH Line Out CLFE       seat0 default group3  cap:
-event17  DEVICE_ADDED     HDA Intel PCH Front Headphone     seat0 default group3  cap:
-event18  DEVICE_ADDED     Logitech USB-PS/2 Optical Mouse   seat0 default group4  cap:p left scroll-nat scroll-button
-event20  DEVICE_ADDED     Wacom Cintiq 24HD touch Finger    seat0 default group5  cap:t  size 518x324mm ntouches 10 calib
-event21  DEVICE_ADDED     Wacom Cintiq 24HD touch Pen       seat0 default group5  cap:T  size 518x324mm calib
-event23  DEVICE_ADDED     Wacom Cintiq 24HD touch Pad       seat0 default group5  cap:P buttons:16 strips:0 rings:2 mode groups:2
-event22  DEVICE_ADDED     Microsoft Natural® Ergonomic Keyboard 4000 seat0 default group6  cap:k
-event19  DEVICE_ADDED     Microsoft Natural® Ergonomic Keyboard 4000 seat0 default group6  cap:kp scroll-nat
-event18  POINTER_MOTION   +0.514s	 1.30/ -0.97 ( +4.00/ -3.00)
 event18  POINTER_MOTION   +0.522s	 7.57/ -4.33 ( +7.00/ -4.00)
 event18  POINTER_MOTION   +0.530s	21.00/ -9.55 (+11.00/ -5.00)
 event18  POINTER_MOTION   +0.538s	17.71/ -1.97 ( +9.00/ -1.00)
 event18  POINTER_MOTION   +0.546s	18.70/  0.00 (+10.00/ +0.00)
~~~

Harpwind, 2020-06-02:
-event3   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k
-event6   DEVICE_ADDED     Video Bus                         seat0 default group2  cap:k
-event7   DEVICE_ADDED     Video Bus                         seat0 default group2  cap:k
-event1   DEVICE_ADDED     Power Button                      seat0 default group3  cap:k
-event2   DEVICE_ADDED     Lid Switch                        seat0 default group4  cap:S
-event0   DEVICE_ADDED     Sleep Button                      seat0 default group5  cap:k
-event14  DEVICE_ADDED     USB Camera: USB Camera            seat0 default group6  cap:k
-event12  DEVICE_ADDED     Intel RealSense 3D Camera R200:   seat0 default group7  cap:k
-event13  DEVICE_ADDED     Intel RealSense 3D Camera R200:   seat0 default group7  cap:k
-event16  DEVICE_ADDED     HDA Intel PCH Mic                 seat0 default group8  cap:
-event17  DEVICE_ADDED     HDA Intel PCH Front Headphone     seat0 default group8  cap:
-event18  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=3       seat0 default group8  cap:
-event19  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=7       seat0 default group8  cap:
-event20  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=8       seat0 default group8  cap:
-event21  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=9       seat0 default group8  cap:
-event22  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=10      seat0 default group8  cap:
-event10  DEVICE_ADDED     Intel HID events                  seat0 default group9  cap:k
-event11  DEVICE_ADDED     Intel HID 5 button array          seat0 default group10 cap:k
-event4   DEVICE_ADDED     AT Translated Set 2 keyboard      seat0 default group11 cap:k
-event26  DEVICE_ADDED     G603 Keyboard                     seat0 default group12 cap:k
-event27  DEVICE_ADDED     G603 Mouse                        seat0 default group12 cap:p left scroll-nat scroll-button
-event28  DEVICE_ADDED     G603 Consumer Control             seat0 default group12 cap:kp scroll-nat
-event23  DEVICE_ADDED     G613 Keyboard                     seat0 default group13 cap:k
-event24  DEVICE_ADDED     G613 Consumer Control             seat0 default group13 cap:kp scroll-nat
-event25  DEVICE_ADDED     G613 System Control               seat0 default group13 cap:k
-event27  POINTER_MOTION    +6.46s	 14.86/-15.88 (+29.00/-31.00)
 event27  POINTER_MOTION    +6.46s	  1.79/  0.00 ( +1.00/ +0.00)
 event27  POINTER_MOTION    +6.46s	 34.00/-26.00 (+17.00/-13.00)
 event27  POINTER_MOTION    +6.46s	 24.00/-12.00 (+12.00/ -6.00)
 event27  POINTER_MOTION    +6.46s	 32.00/-14.00 (+16.00/ -7.00)
 event27  POINTER_MOTION    +6.46s	 36.00/-12.00 (+18.00/ -6.00)
 event27  POINTER_MOTION    +6.47s	 40.00/-10.00 (+20.00/ -5.00)
 event27  POINTER_MOTION    +6.48s	 40.00/-10.00 (+20.00/ -5.00)
 event27  POINTER_MOTION    +6.49s	 42.00/ -4.00 (+21.00/ -2.00)
 event27  POINTER_MOTION    +6.50s	 36.00/-10.00 (+18.00/ -5.00)
 event27  POINTER_MOTION    +6.51s	 27.86/ -3.98 (+14.00/ -2.00)
 event27  POINTER_MOTION    +6.53s	  9.28/ -1.55 ( +6.00/ -1.00)
 event27  POINTER_MOTION    +6.54s	  5.53/ -1.11 ( +5.00/ -1.00)
 event27  POINTER_MOTION    +6.56s	  4.04/  0.00 ( +4.00/ +0.00)
 event27  POINTER_MOTION    +6.57s	  1.00/ -1.00 ( +1.00/ -1.00)
 event27  POINTER_MOTION    +6.58s	 13.63/ -3.72 (+11.00/ -3.00)
 event27  POINTER_MOTION    +6.59s	 16.69/ -8.34 (+10.00/ -5.00)
 event27  POINTER_MOTION    +6.61s	 16.32/ -6.53 (+10.00/ -4.00)
 event27  POINTER_MOTION    +6.62s	 14.08/ -6.26 ( +9.00/ -4.00)
 event27  POINTER_MOTION    +6.63s	 29.16/-10.94 (+16.00/ -6.00)
 event27  POINTER_MOTION    +6.64s	 36.00/-14.00 (+18.00/ -7.00)
 event27  POINTER_MOTION    +6.65s	 30.00/ -8.00 (+15.00/ -4.00)
 event27  POINTER_MOTION    +6.66s	 34.00/-10.00 (+17.00/ -5.00)
 event27  POINTER_MOTION    +6.67s	 23.54/ -5.89 (+12.00/ -3.00)
 event27  POINTER_MOTION    +6.68s	 18.87/ -3.43 (+11.00/ -2.00)
 event27  POINTER_MOTION    +6.70s	 10.25/ -2.93 ( +7.00/ -2.00)
 event27  POINTER_MOTION    +6.71s	  3.31/ -2.21 ( +3.00/ -2.00)
 event27  POINTER_MOTION    +6.89s	  0.00/  0.89 ( +0.00/ +1.00)
 event27  POINTER_MOTION    +6.90s	  0.00/  0.74 ( +0.00/ +1.00)
 event27  POINTER_MOTION    +6.91s	 -1.00/  0.00 ( -1.00/ +0.00)
 event27  POINTER_MOTION    +6.92s	  0.00/  1.00 ( +0.00/ +1.00)
 event27  POINTER_MOTION    +6.93s	 -1.00/  2.00 ( -1.00/ +2.00)
 event27  POINTER_MOTION    +6.94s	 -2.00/  3.00 ( -2.00/ +3.00)
 event27  POINTER_MOTION    +6.95s	 -3.11/  5.19 ( -3.00/ +5.00)
 event27  POINTER_MOTION    +6.96s	 -4.49/  4.49 ( -4.00/ +4.00)
 event27  POINTER_MOTION    +6.98s	 -8.89/  7.62 ( -7.00/ +6.00)
 event27  POINTER_MOTION    +6.99s	 -4.99/  3.74 ( -4.00/ +3.00)
 event27  POINTER_MOTION    +7.00s	 -5.79/  5.79 ( -5.00/ +5.00)
 event27  POINTER_MOTION    +7.01s	 -6.09/  4.87 ( -5.00/ +4.00)
 event27  POINTER_MOTION    +7.02s	 -3.25/  3.25 ( -3.00/ +3.00)
 event27  POINTER_MOTION    +7.03s	 -4.20/  4.20 ( -4.00/ +4.00)
 event27  POINTER_MOTION    +7.04s	 -4.32/  3.24 ( -4.00/ +3.00)
 event27  POINTER_MOTION    +7.05s	 -2.02/  2.02 ( -2.00/ +2.00)
 event27  POINTER_MOTION    +7.07s	 -4.04/  3.03 ( -4.00/ +3.00)
 event27  POINTER_MOTION    +7.08s	 -3.05/  3.05 ( -3.00/ +3.00)
[... 1 lines cut ...]
 event27  POINTER_MOTION    +7.10s	 -3.00/  3.00 ( -3.00/ +3.00)
 event27  POINTER_MOTION    +7.11s	 -2.00/  1.00 ( -2.00/ +1.00)
 event27  POINTER_MOTION    +7.12s	 -2.00/  2.00 ( -2.00/ +2.00)
 event27  POINTER_MOTION    +7.13s	 -2.00/  2.00 ( -2.00/ +2.00)
[... cut ...]
 event27  POINTER_MOTION    +7.27s	 -2.00/  1.00 ( -2.00/ +1.00)
 event27  POINTER_MOTION    +7.28s	 -2.00/  2.00 ( -2.00/ +2.00)
 event27  POINTER_MOTION    +7.29s	 -2.00/  2.00 ( -2.00/ +2.00)
 event27  POINTER_MOTION    +7.30s	 -1.00/  1.00 ( -1.00/ +1.00)
 event27  POINTER_MOTION    +7.31s	 -1.00/  1.00 ( -1.00/ +1.00)
 event27  POINTER_MOTION    +7.33s	 -1.00/  0.00 ( -1.00/ +0.00)
-event23  KEYBOARD_KEY      +9.29s	*** (-1) pressed
 event23  KEYBOARD_KEY      +9.42s	*** (-1) pressed
^C
[harpist@moria ~]$

jigpu, 2020-06-02:
Thanks. That confirms that all of the Wacom kernel devices are missing...

Its possible that the sysinfo logs missed something that might explain this since it tries to snip out any irrelevant data. Would you mind sharing the complete output of `sudo journalctl -b0` after a fresh reboot? Given its length I would recommend compressing it before attaching.

Harpwind, 2020-06-02:
[j.zip](https://github.com/linuxwacom/input-wacom/files/4718856/j.zip)

jigpu, 2020-06-02:
Thanks. There's a particularly suspicious log entry from fwupd that occurs near the same time as the failure:
~~~
fwupd[3959]: failed to add USB device 056a:034b: Failed to get feature report: endpoint stalled or request not supported
~~~
fwupd used to have a bug (https://github.com/fwupd/fwupd/issues/882) that caused this exact problem on the MobileStudio Pro 16. It was fixed in version 1.2.2 but looks like CentOS 8.1 is currently shipping version 1.1.4. I believe you can disable fwupd by running `sudo systemctl disable fwupd` which should fix the issue after a reboot.

If this is the case, you'll want to file an issue with Red Hat describing the bug and requesting an updated version of fwupd. It may be possible to temporarily work around the issue while leaving fwupd enabled as well by editing the configuration file (under `/usr/share/fwupd` I believe?) as described in https://github.com/fwupd/fwupd/commit/aac786fcb905db2719c4a1f1bcf125536336cb6c

Harpwind, 2020-06-02:
I've disabled it several times, it turns itself back on with each reboot.    Currently trying to figure out how to turn it off.

Harpwind, 2020-06-02:
ATM I can't get fwupd to be disabled after reboot and I don't have the configuration file that needs to be edited for a fix...

jigpu, 2020-06-02:
Try editing the `/etc/fwupd/daemon.conf` file and setting `BlacklistPlugins=test;wacomhid`

Harpwind, 2020-06-03:
So I was able to get it not to load by using systemctl mask.   My touch screen now works, it can't tell the differences between the pro pen and the pro pen slim.  But everything else seems operational.

Thank you very much, I would never have found that on my own.    I don't have the file that needs to be edited to make it work, so I'll have to see what file covers wacom, or perhaps I need to create the file.   But at least there is a direction to go in. 

Again thank you for your time it is much appreciated.

Harpwind, 2020-06-03:
Guess my page hadn't updated.  I didn't see your suggestion until after.   I edited the daemon.conf file as directed and that did the trick.   I'm good to go!

jigpu, 2020-06-03:
Glad to hear that it works now. Again, please do file a bug at https://bugs.centos.org/main_page.php to request that CentOS either upgrade fwupd or patch it with the commit I linked to. Looking at the RPM file, it looks like the file is at `/usr/share/fwupd/quirks.d/wacomhid.quirk`

closed 2020-06-03 by jigpu