linuxwacom/input-wacom #190: status of One Pen Display (DTC133)

kharris, 2020-06-05:
I'm unclear as to whether the DTC133 models are compatible with this driver. From this commit, it appears so:
https://github.com/linuxwacom/input-wacom/commit/848e44319ce5358ebc75d0ad158ae57fe0111774
The following commit has a comment showing that the device id 0x3A6 corresponds to DTC133
https://github.com/linuxwacom/xf86-input-wacom/commit/095ba9c48e8cf8a96970e211ed734ba5883a0025

However, this model isn't listed in the device id page: https://github.com/linuxwacom/input-wacom/wiki/Device-IDs

Please help me understand whether it is supported (I don't have a device on hand to test but am considering buying one). Thanks

jigpu, 2020-06-05:
Looks like we forgot to update the table. Yes -- the DTC133 is compatible with the driver. Just as a heads up, you may not need to compile and install anything from source since most distributions will have a driver that is 99.9% compatible out of the box. The changes you reference add support for very old kernels (first commit) and fix some minor driver issues that may not be relevant to you (second commit).

kharris, 2020-06-05:
Great, that helps!
Looks like the table could use an update. I'll close the issue.

closed 2020-06-05 by kharris

jigpu, 2020-06-05:
Actually, one quick caveat is that you may need to drop in a tablet definition file from libwacom to configure the device through the GNOME control panel. Aside from that though...

kharris, 2020-06-05:
Cool. I think I will buy it and work on getting it up and running. If I run into issues I'll try and commit any fixes.