linuxwacom/input-wacom #191: Mismatch on physical/logical buttons for Bamboo (CTH-470)

vitaminace33, 2020-06-10:
Hi, there!
I own an old Wacom Baboom (CHT-470, USB ID 056a:00de). The tablet works fine on Windows ([driver v5.3.5-3](http://cdn.wacom.com/u/productsupport/drivers/win/consumer/pentablet_5.3.5-3.exe)), however on Linux there is a mismatch between physical and logical buttons (tried with stock and compiled drivers and libraries on Ubuntu 18.04 and 20.04). In the following description I consider a right handed configuration (buttons and USB cable on the left).

- Physical top button is paired with logical top button (correct).
- Physical upper mid button is paired with nothing.
- Physical lower mid button is paired with logical bottom button.
- Physical bottom button is paired with logical lower mid button.

Below you'll find a physical representation of the tablet where letters represent the logical button to which the corresponding physical one is linked.

```
╔════╦══════════════════════════════╗
║ T  ║ (correct)                    ║
╟────╢                              ║
║    ║ (nothing)                    ║
╠════╣                              ║
║ B  ║                              ║
╟────╢ (reverse)                    ║
║ LM ║                              ║
╚════╩══════════════════════════════╝
```

vitaminace33, 2020-06-13:
I finally solved the problem thanks to your script. Although it did not fully solve the problem, it helped me understand it. After running it, it maps button 3 (bottom) to event button +4 and it should be mapped to event button +8.

Also, you mentioned that it was a Gnome problem , but I've checked and the default Xorg button events are in 1-to-1 correspondence with the driver assignation. So I slightly disagree with you, but I would be happy if you could clarify this in case I am mistaken. Besides, wouldn't the problem affect any Xorg environment?

Below a table with how things are mapped in different cases. [Here an Xorg conf file.](https://gist.github.com/vitaminace33/cbf26a065858dc613e8a0e04211b2441)

Physical | Driver | Default | Script | Needed
---------|--------|---------|--------|-------
 T  | 1 |  +1 | +1 | +1
 NA | 2 |  +2 | key +2 -2 | *
 B  | 3 |  +3 | +4 | +8
 LM | 8 | +8 | +3 | +3
 UM | 9 |  +9 | +2 | +2

Thank you again!

jigpu, 2020-06-15:
> Also, you mentioned that it was a Gnome problem , but I've checked and the default Xorg button events are in 1-to-1 correspondence with the driver assignation. So I slightly disagree with you, but I [...]
The issue is caused by GNOME's incorrect assumptions of what events our driver sends. Other desktops aren't affected because they generally don't care about the events we send. (GNOME wants to show a fancy UI which highlights the button you pressed and wants to be in control of the events that applications receive. To do this, they need to know that our driver will e.g. send a button 1 event in response to pressing the "top" button. Other desktops don't care -- their configuration UIs are much less fancy and the desktops don't act as middle-men between the driver and applications).

GNOME makes two assumptions that don't hold with the Bamboo:
* Tablets use a linear sequence of button numbers (skipping 4-7 for historic reasons)
* The mapping between the button named in libwacom and the button number emitted by our driver is trivial

The first assumption is broken because the Bamboo has a hole in its sequence. The set of buttons is not [1, 2, 3, 8] like GNOME expects, but instead [1, 3, 8, 9]. The sequence is intentional, so its not a driver bug ;)

The second assumption is broken because the mapping is non-trivial. Libwacom uses letters to name each button, and the four buttons are named (from top to bottom) [A, B, C, D]. GNOME assumes that its safe to trivially translate A=1, B=2, etc. In reality, however, the order jumps around on the Bamboo tablet, causing [A=1, B=9, C=8, D=3]. The correct mapping is actually specified in libwacom if GNOME cared to look, so again -- not a driver bug :D

The workaround works by configuring the driver to emit events that match up with GNOME's assumptions. Your top (A) button now sends 1, the upper-mid (B) button now sends 2, lower-mid (C) sends 3, and bottom (D) sends 8. The mapping is linear and trivial, just like GNOME wants :)

vitaminace33, 2020-06-16:
This is a wonderful technical answer. Thank you, @jigpu, I appreciate very much the time you spent replying. This should be pasted into wiki/documentation. Many thanks!

jigpu, 2020-06-16:
Updated our "Known Issues" page with a short description of the problem and link here for reference: https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#gnome-bamboo-buttons