linuxwacom/input-wacom #192: Pen stops working (Intuos Pro S, Ubuntu 18.04, wacom driver 0.46.0)

rm144, 2020-06-14:
I'm using an Intuos Pro S tablet on a Dell Precision 7730 laptop with Ubuntu 18.04. Sporadically, the pen stops working (I saw several other reports on a similar effect). When that happens, the cursor still moves, but I can't draw any longer (probably no motion events are sent to my program). When the pen stops working, immediately messages like this appear in /var/log/Xorg.0.log:

[    87.902] (EE) Wacom Intuos Pro S Pad pad: Error reading wacom device : No such device
[    87.987] (EE) Wacom Intuos Pro S Pen cursor: Error reading wacom device : No such device

Restarting the driver (rmmod, modprobe) restores the function. I have the impression that the problem typically appears shortly after the machine was booted (in my case with the tablet plugged in).

In an attempt to fix this issue, I installed the latest wacom kernel driver:

% modinfo wacom | grep version
version:        v2.00-0.46.0
srcversion:     62537F6171CFA60B57FF9D7

but that didn't change the behavior (all things I reported above are produced with the new driver).

Some more information: The touch switch on the tablet is off. I didn't have problems with an older Intuos 3 tablet on Ubuntu 18.04. I used an Intuos Pro M on Ubuntu 16.04 over a long period without any problems, so apparently the problems started to appear with the wacom driver shipped with Ubuntu 18.04 and are not resolved until now.

Can someone please look into this? I'm using the tablet for online lectures and kind of depend on it. Please let me know if I can help with more information. Thanks a lot!

jigpu, 2020-06-15:
Can you please reboot and then run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) after the issue occurs? Attach the generated tarball to this issue. It will contain various system logs and hardware information that will help us to understand what may be going wrong.

rm144, 2020-06-16:
Thanks for your response! Here is the output of the sysinfo.sh script after the pen stopped working:

[sysinfo.25ECmLQ3zd.tar.gz](https://github.com/linuxwacom/input-wacom/files/4785107/sysinfo.25ECmLQ3zd.tar.gz)

rm144, 2020-06-16:
Here comes another one. I restarted the drivers (rmmod/modprobe) and doodled on the tablet. This time the _eraser_ stopped working after some time, but the stylus still worked. Please let me know if I should test something else.

[sysinfo.GhJvFJWeXL.tar.gz](https://github.com/linuxwacom/input-wacom/files/4785161/sysinfo.GhJvFJWeXL.tar.gz)

jigpu, 2020-06-25:
The "Error reading wacom device" message is typically caused by a device disconnection, but I don't see any evidence of that in the kernel logs. Even stranger, right after the message is printed out the kernel lists a *new* device which the X driver promptly connects to. Its almost like the kernel driver has a good device handle (which is why we don't see any evidence of disconnect) but a bad communication pathway (which triggers the errors and re-probe). I've never seen an issue quite like it...

Do you have the ability to test your tablet under Windows? It would be good to rule out any potential hardware issues. Failing that, it would be interesting to know how the tablet behaves in a Fedora 32 live USB environment (to see if the issue is specific to Ubuntu) or how it behaves on a different computer.

rm144, 2020-06-25:
Thanks for your response. Unfortunately, I currently have no access to a Windows machine. One thing I managed so far was to test the tablet on a Lenovo Thinkpad T530 running Ubuntu 18.04 with the Wacom driver shipped with 18.04. There the tablet seems to work, but I will test it for longer time later.

Just to rule that out: Switching USB ports on the Dell (18.04, newest driver) didn't solve the problem.

Meanwhile I had to use my old Intuos 3 tablet on the Dell, which works without any problems.

rm144, 2020-06-28:
I tried again with the other Laptop (Thinkpad T530), there the tablet seems to work fine (the Wacom driver is the one shipped with 18.04). I attach a sysinfo tarball from the Thinkpad, maybe it's helpful for a comparison: 

[sysinfo.OX79AzoBdQ.tar.gz](https://github.com/linuxwacom/input-wacom/files/4842619/sysinfo.OX79AzoBdQ.tar.gz)

I don't think the tablet itself is broken. I don't think the Dell Laptop has a problem with the USB port, since the old Wacom Intuous 3 tablet works fine on the Dell. Could there be a problem with the communication between the Wacom driver and different USB drivers (just guessing).

Thanks again for you help!

jigpu, 2020-06-29:
Thanks -- that does seem to rule out an issue with the tablet itself.

It looks like your other laptop is running an older kernel version (4.15 instead of 5.3), which might explain the difference in behavior. Probably the closest input-wacom release would be [version 0.38.0](https://github.com/linuxwacom/input-wacom/releases/tag/input-wacom-0.38.0) -- could you [follow the instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) to build that particular version and see if it works any better? Be sure to run `grep "" /sys/module/wacom*/version` after rebooting to be sure version 0.38.0 is in use.

If version 0.38.0 doesn't work, you might also try [0.37.1](https://github.com/linuxwacom/input-wacom/releases/tag/input-wacom-0.37.1) or [0.39.0](https://github.com/linuxwacom/input-wacom/releases/tag/input-wacom-0.39.0). If you find a version which works, let me know and we can try to figure out where exactly things broke so we can fix them.

rm144, 2020-07-04:
I tried 0.38.0, still the same problem:
[sysinfo.J1ML7PDm25.tar.gz](https://github.com/linuxwacom/input-wacom/files/4873121/sysinfo.J1ML7PDm25.tar.gz)
% cat /sys/module/wacom/version 
v2.00-0.38.0

rm144, 2020-07-04:
Just a question: When I run configure in the wacom driver directory (0.37.1), I get this output which confuses me:
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/5.3.0-62-generic/build
Version 4.5 vs. version 5.3.0... Is that ok?

rm144, 2020-07-04:
I now tried 0.37.1, without success:
[sysinfo.3kVU5cnVIz.tar.gz](https://github.com/linuxwacom/input-wacom/files/4873167/sysinfo.3kVU5cnVIz.tar.gz)
% cat /sys/module/wacom/version 
v2.00-0.37.1

rm144, 2020-07-04:
Finally, 0.39.0, still the same problem:
[sysinfo.vrFFO0FUOT.tar.gz](https://github.com/linuxwacom/input-wacom/files/4873199/sysinfo.vrFFO0FUOT.tar.gz)
% cat /sys/module/wacom/version 
v2.00-0.39.0

Anything else I can test? Thanks for your help!

rm144, 2020-07-04:
So, last test for today: I just got a new Intuos Pro M tablet and tested this with the newest driver (0.46.0), but still the same effect:
[sysinfo.kRjaixrTIZ.tar.gz](https://github.com/linuxwacom/input-wacom/files/4873256/sysinfo.kRjaixrTIZ.tar.gz)
% cat /sys/module/wacom/version 
v2.00-0.46.0

So we can exclude a problem with the tablet itself (if it's not a problem with the entire Intuos Pro series).

rm144, 2020-07-05:
Here's the output of dmesg after the pen stopped working, but I'm not sure if that helps. At time 88.75... the tablet seems to be re-initialized, but I can't see a related error message before that. There's a message "wacom: loading out-of-tree module taints kernel" in the boot process, but that's probably because I use the newest driver (0.46.0).
[dmesg.log](https://github.com/linuxwacom/input-wacom/files/4874970/dmesg.log)
% cat /sys/module/wacom/version 
v2.00-0.46.0

rm144, 2020-07-06:
Just an observation: Sometimes just the tablet buttons don't respond any longer, while the pen and the eraser are still working.

rm144, 2020-07-08:
Getting desperate, I bought a USB-C to USB-C cable and connected the Intuos Pro S to a Thunderbolt-3 port of my Dell Precision 7730. Same effect. Stops working after a while, both if plugged in during booting and if plugged in later. Didn't crash so far when plugged out after booting and plugged in again, but I'm not optimistic, that may be a coincidence.

I asked three people from my group to test it with their machines (they also have Intuos Pro M tablets on different versions of Ubuntu on laptops and PCs), but nobody else encountered problems.

Is anyone out there who sees the same effects that I have?

Any ideas?

jigpu, 2020-07-09:
Been studying your results and scratching my head for the past few days but still don't have much of an idea what could be wrong. It still sounds like it could be some kind of issue with your 7730 or its OS since 1) rolling back the Wacom driver doesn't seem have any effect, 2) multiple tablets are affected, and 3) you and your group haven't been able to reproduce the issue on another machine. The only confounding bit of data is your Intuos3 results, but its possible that it just doesn't end up triggering the same code paths for some reason.

I don't see any mention of you trying another OS on the 7730; is it possible to throw Fedora 32 or Ubuntu 20.04 onto a USB stick and boot into the live environment? The tablet should work out of the box with both of those OSes, so you shouldn't have to worry about trying to install input-wacom from source.

rm144, 2020-07-10:
Thanks, I'll try booting a newer Ubuntu version from a USB stick later.

What I tried today was to update the kernel from 5.3 to 5.7.8 (using the "builtin" wacom driver), but it didn't help:

[sysinfo.3ofpOV2fpX.tar.gz](https://github.com/linuxwacom/input-wacom/files/4903193/sysinfo.3ofpOV2fpX.tar.gz)

rm144, 2020-07-10:
I booted Ubuntu 20.04 from a USB stick, then tested the tablet by drawing in "xournal++". Pressure sensitivity etc. was working. The typical error (tablet stops to respond) did not occur. I had "tail -f /var/log/Xorg.0.log'" running on the side and saw now error messages. (From time to time there was a huge time lag when using the eraser, though.)

I'll try upgrading my laptop to 20.04, I hope that there won't be too many things that break.

rm144, 2020-07-11:
I'm afraid the tablet support is not stable under Ubuntu 20.04 (from USB stick) either. I switched from the Intuos Pro S to the Intuos Pro M tablet. First, the behavior with that tablet is somewhat strange. When connected, a window "Wacom Start" appears. Whenever I press the middle button, a little window appears with a tablet/pen symbol and four dots which change in sync with the leds surrounding the middle button. When I do this, strange messages appear in the journalctl log (something about leds that can't be accessed and about a buffer which is filled with "AAAA...AA=="). Second, after a while the eraser stopped working, and then the system crashed completely (blank screen, ended up in the graphical login). So the crash described above also may have been related to the tablet.

I don't know what to do. I'm not very optimistic whether upgrading to 20.04 will solve the problems and not just introduce different ones.

rm144, 2020-07-11:
With _very_ cautious optimism: I may have found the problem. On 
[https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect](url)
I saw a comment on an undesired interaction between the fwupd daemon and some tablets. It didn't fit the description of my problem, but I regardless followed the advice of editing /etc/fwupd/daemon.conf and adding wacomhid;wacom_raw;wacom_usb to the BlacklistPlugins:
`BlacklistPlugins=test;wacomhid;wacom_raw;wacom_usb`
After rebooting I so far didn't encounter the old problem anymore (Ubuntu 18.04, now kernel 5.7.8, default wacom driver).

rm144, 2020-08-25:
So far the problem didn't reappear, so I suppose the fwupd problem was the cause.

closed 2020-08-25 by rm144